arendst/Tasmota #16204: Influxdb v2 cloud

erdinguma, 2022-08-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Current Influxdb API implementation fails with Influxdb Cloud V2 because it needs an update to the new API. The original API implemented in Tasmota (https://github.com/tobiasschuerg/InfluxDB-Client-for-Arduino) is not current and must be updated by following the most current version of Tobias' repository for InfluxDB Cloud V2 to work.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): ESP8266
- [ x] Tasmota binary firmware version number used: 12.0.2
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [x ] Flashing tools used: Platformio
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
13:15:04.924 IFX: Validating connection to https://us-east-1-1.aws.cloud2.influxdata.com/health
13:15:05.046 IFX: Error <html>
<head><title>400 The plain HTTP request was sent to HTTPS port</title></head>
<body>
<center><h1>400 Bad Request</h1></center>
<center>The plain HTTP request was sent to HTTPS port</center>
<hr><center>nginx</center>
</body>
</html>

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-11:
I think I never supported https for influxdb hence the massive amount og code space saved.

So this is as designed.

Jason2866, 2022-08-11:
Closing, since answered

closed 2022-08-11 by Jason2866