arendst/Tasmota #16231: Sonoff ifan04-H maybe not operating as expected?

sotiris-bos, 2022-08-16:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have a Sonoff ifan04-H model. I have replaced the two capacitors with two smaller ones, a 1uF and a 1.5uF capacitor because the included caps had too high of a capacitance and span my fan too fast. However, the difference between fanspeed 1 and fanspeed 2 is very large, the fan consumes 7w when running at fanspeed 1 with cap 1uF, and 30w+ when running at fanspeed 2 with cap 1.5uF.

I am not sure if this is related to the caps or tasmota but I want to raise this issue: The original Sonoff firmware activates both caps when fanspeed 2 is enabled. The tasmota firmware is supposed to only activate one of the caps when it is in either fanspeed 1 or fanspeed 2.

Could this be a case where the firmware does not work as expected and activates both caps at the same time on ifan04-H? I also own a Sonoff ifan03, installed on a second identical fan, with standard caps (2.5uF for fanspeed 1 and 3uF for fanspeed 2) and the rotation speed of ifan04 with fanspeed 2 (1.5uF supposedly) and ifan03 at fanspeed 1 (2.5uF) is quite similar, but I don't think it is identical.

Is there anything I can do to make sure that only one of the relays and hence only one of the caps are providing power to the fan? Any logs or files I can share with you to diagnose?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:26:41.001 CMD: Status 0
14:26:41.008 MQT: stat/tasmota_77E276/STATUS = {"Status":{"Module":71,"DeviceName":"Sotiris Fan","FriendlyName":["Sotiris Fan"],"Topic":"tasmota_77E276","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:26:41.015 MQT: stat/tasmota_77E276/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:00:32","StartupUTC":"2022-08-16T13:26:09","Sleep":50,"CfgHolder":4617,"BootCount":29,"BCResetTime":"2022-08-09T18:46:30","SaveCount":217,"SaveAddress":"FB000"}}
14:26:41.024 MQT: stat/tasmota_77E276/STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"337/699"}}
14:26:41.029 MQT: stat/tasmota_77E276/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000","00000000"]}}
14:26:41.042 MQT: stat/tasmota_77E276/STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
14:26:41.050 MQT: stat/tasmota_77E276/STATUS5 = {"StatusNET":{"Hostname":"tasmota-77E276-0630","IPAddress":"10.0.3.43","Gateway":"10.0.3.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.3.1","DNSServer2":"0.0.0.0","Mac":"70:03:9F:77:E2:76","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
14:26:41.056 MQT: stat/tasmota_77E276/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.3.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_77E276","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:26:41.064 MQT: stat/tasmota_77E276/STATUS7 = {"StatusTIM":{"UTC":"2022-08-16T13:26:41","Local":"2022-08-16T14:26:41","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:44","Sunset":"20:04"}}
14:26:41.071 MQT: stat/tasmota_77E276/STATUS10 = {"StatusSNS":{"Time":"2022-08-16T14:26:41"}}
14:26:41.079 MQT: stat/tasmota_77E276/STATUS11 = {"StatusSTS":{"Time":"2022-08-16T14:26:41","Uptime":"0T00:00:32","UptimeSec":32,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","FanSpeed":0,"Wifi":{"AP":1,"SSId":"iot","BSSId":"8A:8A:20:D7:7C:E9","Channel":6,"Mode":"11n","RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect fanspeed 1 and fanspeed 2 to be more similar to each other. My old controller was running a 1.5uF cap for its fanspeed 2 and the fan was rotating much more slowly than the Sonoff ifan04 with tasmota and 1.5uF capacitor for its fanspeed 2.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-20:
Tasmota controls the same relays as the original firmware.

See source code regarding fanspeed control:
```
uint8_t GetFanspeed(void)
{
  if (ifan_fanspeed_timer) {
    return ifan_fanspeed_goal;                     // Do not show sequence fanspeed
  } else {
    /* Fanspeed is controlled by relay 2, 3 and 4 as in Sonoff 4CH.
      000x = 0
      001x = 1
      011x = 2
      101x = 3 (ifan02) or 100x = 3 (ifan03)
    */
    uint8_t fanspeed = (uint8_t)(TasmotaGlobal.power &0xF) >> 1;
    if (fanspeed) { fanspeed = (fanspeed >> 1) +1; }  // 0, 1, 2, 3
    return fanspeed;
  }
}
```

sotiris-bos, 2022-08-20:
> Tasmota controls the same relays as the original firmware.
>
> See source code regarding fanspeed control:
>
> ```
> uint8_t GetFanspeed(void)
> {
>   if (ifan_fanspeed_timer) {
>     return [...]

Huh. I found this post that let me believe that the author made a Pull Request that got accepted: https://www.breznet.com/sonoff-ifan03-home-assistant/

The post claims that the original handling (original firmware) is wrong and that the relays should not be powered on concurrently.

barbudor, 2022-08-21:
@sotiris-bos He submitted and merged the PR on it's own fork. Can't any clue that he ever been submitted to arendst/asmota
Nevertheless I have no idea if his claim is correct or not

arendst, 2022-08-21:
ifan02 and ifan03 have been around for years. I have them both and they control the fan exactly as advertised using tasmota so I see no reason why to change stable code other than the fact that ifan04 may work differently then the previous ones.

Let other ifan04 users speak if they find fan control wlrks different on ifan04 compared to the previous ones.