arendst/Tasmota #16235: When powering on 1 relay on Shelly 2.5, the other relay also turns on when shutter support is enabled

stehof2, 2022-08-16:
### PROBLEM DESCRIPTION
I'm using a Shelly 2.5 with the latest Tasmota firmware to control my Somfy RTS roller shutter. After flashing Tasmota to the Shelly, connect it to the Wi-Fi and imported the Shelly 2.5 template with the buttons config as described in here: https://gist.github.com/markuspeter/228cdf9083a922a262f7cdc03f38bdaa like this: {"NAME":"Shelly 2.5","GPIO":[56,0,19,0,21,127,0,0,6,126,5,22,156],"FLAG":2,"BASE":18}.
It seems to work fine, but as soon as I enable the shutter support by commanding: SetOption80 1, I have an issue.

First one is, when I press the up button, it goed down, I guess this can be fixed by swapping the GPIO buttons, but the biggest problem is, when I press the down button, both relays turns on what in this case means, it want to go up and down at the same time. So when relay1 turns on, it should go down, when relay to turns on, it should go up.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Shelly 2.5
- [ X] Tasmota binary firmware version number used: 12.0.2
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: OTA - https://github.com/yaourdt/mgos-to-tasmota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

17:58:20.132 RSL: RESULT = {"NAME":"Shelly 2.5","GPIO":[320,0,32,0,224,193,0,0,640,192,608,225,3456,4736],"FLAG":0,"BASE":18}
17:58:20.351 RSL: RESULT = {"Module":{"0":"Shelly 2.5"}}
17:58:20.606 RSL: RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"32":"Button1"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"193":"Switch_n2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"192":"Switch_n1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"225":"Relay2"},"GPIO16":{"3456":"ADE7953 IRQ1"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:59:31.837 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_E62790","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:59:31.843 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:26:44","StartupUTC":"2022-08-16T14:32:47","Sleep":50,"CfgHolder":4617,"BootCount":14,"BCResetTime":"2022-08-16T15:11:17","SaveCount":52,"SaveAddress":"F6000"}}
17:59:31.850 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2(tasmota)","BuildDateTime":"2022-06-20T12:41:03","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"370/699"}}
17:59:31.854 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AIVD Bus #2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000","00000000"]}}
17:59:31.864 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":619,"Free":384,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","14000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
17:59:31.869 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-E62790-1936","IPAddress":"192.168.0.163","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.7","DNSServer2":"0.0.0.0","Mac":"84:F3:EB:E6:27:90","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:59:31.872 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E62790","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:59:31.877 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-08-16T16:59:31","Local":"2022-08-16T17:59:31","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:44","Sunset":"20:04"}}
17:59:31.880 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:59:31.890 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-08-16T17:59:31","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":44.5},"ENERGY":{"TotalStartTime":"2022-08-16T15:11:17","Total":0.005,"Yesterday":0.000,"Today":0.005,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Frequency": 0,"Voltage": 0,"Current":[0.000,0.000]},"TempUnit":"C"}}
17:59:31.896 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-08-16T17:59:31","Uptime":"0T02:26:44","UptimeSec":8804,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"AIVD Bus #2","BSSId":"74:AC:B9:C7:7C:F5","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
18:02:48.192 SHT: Shtr1 Real 26000, Start 27000, Stop 0, Dir -1, Delay 0, Rtc 0.10 [s], Freq 0, PWM 0, Tilt 0
18:02:48.195 RSL: RESULT = {"Shutter1":{"Position":13,"Direction":-1,"Target":0,"Tilt":0}}
18:02:48.209 ADE: 6083783, 4469, [94860, -27748, 34392, -14764], [94018, 30134, 34122, 13889]
18:02:48.212 ADE: U 6083783, C 4469, I 94860 + 94018 = 188878, P 27748 + 30134
18:02:48.239 SHT: Time 3(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 25000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.289 SHT: Time 4(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 24000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.341 SHT: Time 5(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 23000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.390 SHT: Time 6(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 22000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.441 SHT: Time 7(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 21000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.490 SHT: Time 8(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 20000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.541 SHT: Time 9(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 19000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.591 SHT: Time 10(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 18000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.641 SHT: Time 11(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 17000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.690 SHT: Time 12(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 16000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
18:02:48.742 SHT: Time 13(0), cStop 0, cVelo 0, mVelo 0, aVelo 1000, mRun 0, aPos 0, aPos2 15000, nStop 0, Trgt 0, mVelo 0, Dir -1, Tilt 0, TrgtTilt: 0, Tiltmove: 0
[... cut ...]
18:02:51.211 ADE: U 6097522, C 4470, I 0 + 0 = 0, P 0 + 0
18:02:51.450 HTP: Console

```

### TO REPRODUCE
Use a Somfy RTS shutter on a Shelly 2.5 with Tasmota and enable the shutter support: SetOption80 1

### EXPECTED BEHAVIOUR
When powering on 1 relay, the other should stay off

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stehof2, 2022-08-16:
[image]

Jason2866, 2022-08-17:
Please use latest development version of Tasmota. There where changes in shutter code since the release of v12.0.2

stefanbode, 2022-08-17:
There are several shuttermodes psoosible. Please see documentation and set shuttermode to the correct values. I assume "Interlock 1,2" and "interlock on" is missing

stefanbode, 2022-08-22:
@stehof2 : Please can you close the case if the answer helped and now everything is fine. THX

Jason2866, 2022-08-22:
Closing since no feedback, and the issue is fixed. Feel free to reopen if the issue is still there for you

closed 2022-08-22 by Jason2866

stehof2, 2022-08-30:
Sorry for the late response, wasn't able to get back to this earlier.
But, good news, it is working now. By default the Interlock 1,2 were on place, but indeed the Interlock On was missing, after I commanded it to On, it worked right away.

So if you have the same issue, check in the console by typing interlock, what the value is set to, if off, command interlock on :)