arendst/Tasmota #16242: RTC (DS3231) not detected when RGB LED (WS2812) is present

martinren87, 2022-08-17:
### PROBLEM DESCRIPTION
Setting a I2C RTC DS3231 and a RGB LED WS2812 will make the first to stop working.
Using exactly the same configuration without setting WS2812, makes RTC start working again

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP32 - NODEMCU
- [ ] Tasmota binary firmware version number used: 12.0.2.4
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
17:11:56.538 CMD: Backlog Template; Module; GPIO 255
17:11:56.572 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
17:11:56.771 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
17:11:56.989 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"1376":"WS28121"},"GPIO22":{"640":"I2C SDA1"},"GPIO23":{"608":"I2C SCL1"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:09:19.634 CMD: Status 0
17:09:19.641 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_64F02C","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:09:19.679 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:02:14","StartupUTC":"2022-08-17T16:07:05","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-08-17T16:47:02","SaveCount":34}}
17:09:19.706 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.2.4(tasmota)","BuildDateTime":"2022-08-17T12:30:31","Core":"2_0_5_dev","SDK":"v4.4.3-8-g13eca437d1","CpuFrequency":80,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"410/699"}}
17:09:19.732 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Havana Club",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
17:09:19.774 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1354,"Free":1856,"Heap":143,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","1F9A95CE","0001A001","B7F7BFCF","05DA9F00","60360C47","400840D2","20200000","8434480D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,14,16,17,24,29,34,38,50,52,56,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
17:09:19.824 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-64F02C-4140","IPAddress":"192.168.1.206","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"84:CC:A8:64:F0:2C","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:09:19.861 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_64F02C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:09:19.888 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-08-17T16:09:19","Local":"2022-08-17T17:09:19","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:46","Sunset":"20:02"}}
17:09:19.914 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-08-17T17:09:19","ESP32":{"Temperature":68.9},"TempUnit":"C"}}
17:09:19.930 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-08-17T17:09:19","Uptime":"0T00:02:14","UptimeSec":134,"Heap":145,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":36},"POWER":"ON","Dimmer":87,"Color":"58DD8D","HSBColor":"144,60,87","Channel":[34,87,55],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Havana Club","BSSId":"D8:07:B6:81:8A:09","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Set RTC and RGB Led and the first one won´t start.
Set NONE instead of the RGB LED in the configuration, and the RTC will start working. 
Console output of both situations:

```lua
00:00:00.003 HDW: ESP32-D0WDQ6 
00:00:00.086 UFS: FlashFS mounted with 304 kB free
00:00:00.178 CFG: Loaded from File, Count 30
00:00:00.194 QPC: Count 1
00:00:00.205 I2C: DS3231 found at 0x68
17:00:16.207 BRY: Berry initialized, RAM used=3544 bytes
17:00:16.237 Project tasmota - Tasmota Version 12.0.2.4(tasmota)-2_0_5_dev(2022-08-17T12:30:31)
17:00:17.235 WIF: Connecting to AP1 Havana Club Channel 6 BSSId D8:07:B6:81:8A:09 in mode 11n as tasmota-64F02C-4140...
17:00:21.873 WIF: Connected
17:00:22.167 HTP: Web server active on tasmota-64F02C-4140 with IP address 192.168.1.206
17:00:22.802 QPC: Reset
17:00:25.197 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"12.0.2.4(tasmota)","FallbackTopic":"cmnd/DVES_64F02C_fb/","GroupTopic":"todev/tasmotas/"}}
17:00:25.213 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-64F02C-4140","IPAddress":"192.168.1.206"}}
17:00:25.217 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":6}}

00:00:00.003 HDW: ESP32-D0WDQ6 
00:00:00.115 UFS: FlashFS mounted with 304 kB free
00:00:00.217 CFG: Loaded from File, Count 32
00:00:00.233 QPC: Count 1
00:00:00.510 BRY: Berry initialized, RAM used=3544 bytes
00:00:00.555 Project tasmota - Tasmota Version 12.0.2.4(tasmota)-2_0_5_dev(2022-08-17T12:30:31)
00:00:01.308 WIF: Connecting to AP1 Havana Club Channel 6 BSSId D8:07:B6:81:8A:09 in mode 11n as tasmota-64F02C-4140...
00:00:03.327 WIF: Connected
00:00:03.588 HTP: Web server active on tasmota-64F02C-4140 with IP address 192.168.1.206
17:00:56.827 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"12.0.2.4(tasmota)","FallbackTopic":"cmnd/DVES_64F02C_fb/","GroupTopic":"todev/tasmotas/"}}
17:00:56.842 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-64F02C-4140","IPAddress":"192.168.1.206"}}
17:00:56.847 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":7}}

```

### EXPECTED BEHAVIOUR
Both accesories should work simultaneously

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-08-17:
Your configuration above show neither i2c nor ws2812 gpios configured

Also the sensor list in the status do not list xsns_33 which is the ds3221 driver which is normal is you are using a standard tasmota binary as this not built-in by default

martinren87, 2022-08-17:
Sorry but I don´t understand what you mean. 
My configuration shows both i2c and ws2812: 
"GPIO21":{"1376":"**WS28121**"},"GPIO22":{"640":"**I2C SDA1**"},"GPIO23":{"608":"**I2C SCL1**"}

Also as you can see in the console output, when i delete the ws2812, the ds3221 starts working ok:

00:00:00.194 QPC: Count 1
**00:00:00.205 I2C: DS3231 found at 0x68**
17:00:16.207 BRY: Berry initialized, RAM used=3544 bytes 

Please, let me know if there is something i am misunderstanding

Jason2866, 2022-08-17:
How is the ws2812 connected (driver ic?) and powered?

barbudor, 2022-08-17:
Sorry, I missed the GPIOs as the ilst is so loooong on ESP32

I would guess you have an electrical problem
Can you try to move the WS2812 to another GPIO further away from the I2C ?
I believe there my be some short cut or something between GPIO21 and GPIO22/23

sfromis, 2022-08-17:
A small test seems to confirm that there is some issue. Connected a DS3231 to an ESP32 with a couple of other I2C sensors, and it worked as expected. `I2C: DS3231 found at 0x68`
Next configured a WS2812 on a free gpio, and after the restart, the DS3231 was no more detected, even if still visible on `I2CScan`. Persistent over a couple of restarts flipping WS2812 configured or not.

barbudor, 2022-08-18:
Interresting
I don't have a ds3231 to test it myself
Is that the only i2c sensor impacted by the ws2812?
Could you check on an esp8266 too?

sfromis, 2022-08-18:
Yes, the DS3231 not registering was the only oddity, other I2C sensors still working.

arendst, 2022-08-18:
I did the rewrite to speed up rtc detection asap after restart. I'll chk what went wrong. Nice issue.

arendst, 2022-08-18:
Verified to fail. I know where it fails. Need to find a workaround....

arendst, 2022-08-18:
Fixed in v12.1.0.1

martinren87, 2022-08-18:
Thank you all guys!

closed 2022-08-19 by arendst