arendst/Tasmota #16251: ESP32 CAM Not Working anymore with Tasmota 12..1.0

helmar74, 2022-08-18:
### PROBLEM DESCRIPTION
I updated my ESP32 CAM from Tasmota 12.0.2 to 12.1.0.
After that the Main Menu isn't working anymore.
I can access Subpages with direct links (e.g. /cs?, /up?) but the Main Menu isn't working... It looks like this:

[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): AI Tinker CAM
- [X] Tasmota binary firmware version number used: 12.1.0 
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
6:01:00.876 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
16:01:00.900 SRC: Backlog
16:01:00.902 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
16:01:00.911 MQT: stat/webcam8/RESULT = {"NAME":"AITHINKER CAM","GPIO":[4992,1,672,1,416,5088,1,1,1,6720,736,704,1,1,5089,5090,0,5091,5184,5152,0,5120,5024,5056,0,0,0,0,4928,576,5094,5095,5092,0,0,5093],"FLAG":0,"BASE":2}
16:01:01.110 SRC: Backlog
16:01:01.112 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
16:01:01.119 MQT: stat/webcam8/RESULT = {"Module":{"0":"AITHINKER CAM"}}
16:01:01.320 SRC: Backlog
16:01:01.322 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 3, Pld 255, Data '255'
16:01:01.341 MQT: stat/webcam8/RESULT = {"GPIO0":{"4992":"CAM_XCLK"},"GPIO1":{"0":"None"},"GPIO2":{"672":"SPI MISO1"},"GPIO3":{"0":"None"},"GPIO4":{"416":"PWM1"},"GPIO5":{"5088":"CAM_DATA1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"6720":"SDCard CS"},"GPIO14":{"736":"SPI CLK1"},"GPIO15":{"704":"SPI MOSI1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"5089":"CAM_DATA2"},"GPIO19":{"5090":"CAM_DATA3"},"GPIO20":{"0":"None"},"GPIO21":{"5091":"CAM_DATA4"},"GPIO22":{"5184":"CAM_PCLK"},"GPIO23":{"5152":"CAM_HREF"},"GPIO24":{"0":"None"},"GPIO25":{"5120":"CAM_VSYNC"},"GPIO26":{"5024":"CAM_SIOD"},"GPIO27":{"5056":"CAM_SIOC"},"GPIO32":{"4928":"CAM_PWDN"},"GPIO33":{"576":"LedLink_i"},"GPIO34":{"5094":"CAM_DATA7"},"GPIO35":{"5095":"CAM_DATA8"},"GPIO36":{"5092":"CAM_DATA5"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"5093":"CAM_DATA6"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:59:48.587 CMD: status 0
15:59:48.596 MQT: stat/webcam8/STATUS = {"Status":{"Module":0,"DeviceName":"webcam8","FriendlyName":["webcam8"],"Topic":"webcam8","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:59:48.635 MQT: stat/webcam8/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:32","StartupUTC":"2022-08-18T14:59:16","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2022-08-18T15:34:20","SaveCount":56}}
15:59:48.675 MQT: stat/webcam8/STATUS2 = {"StatusFWR":{"Version":"12.1.0(webcam)","BuildDateTime":"2022-08-18T10:33:47","Core":"2_0_4_1","SDK":"v4.4.3","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6 rev.1","CR":"378/699"}}
15:59:48.693 MQT: stat/webcam8/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["hsnetioteg24ghz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
15:59:48.725 MQT: stat/webcam8/STATUS4 = {"StatusMEM":{"ProgramSize":1255,"Free":1856,"Heap":149,"StackLowMark":3,"PsrMax":4093,"PsrFree":4061,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":80,"FlashMode":3,"Features":["00000809","0F9005CA","00040001","00000082","00000000","40000981","80000080","00200000","54000008","00000000"],"Drivers":"1,2,4,7,8,9,10,12,24,27,29,38,50,52,62,81","Sensors":"2,6,127"}}
15:59:48.765 MQT: stat/webcam8/STATUS5 = {"StatusNET":{"Hostname":"webcam8","IPAddress":"192.168.1.131","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"F0:08:D1:D2:F2:90","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:59:48.795 MQT: stat/webcam8/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"webcam8","MqttClient":"webcam8","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:59:48.815 MQT: stat/webcam8/STATUS7 = {"StatusTIM":{"UTC":"2022-08-18T14:59:48","Local":"2022-08-18T15:59:48","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:47","Sunset":"20:00"}}
15:59:48.846 MQT: stat/webcam8/STATUS10 = {"StatusSNS":{"Time":"2022-08-18T15:59:48","ESP32":{"Temperature":95.6},"TempUnit":"C"}}
15:59:48.868 MQT: stat/webcam8/STATUS11 = {"StatusSTS":{"Time":"2022-08-18T15:59:48","Uptime":"0T00:00:32","UptimeSec":32,"Heap":149,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"Berry":{"HeapUsed":2,"Objects":30},"POWER":"ON","Dimmer":92,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"hsnetioteg24ghz","BSSId":"F0:B0:14:6F:B5:4F","Channel":1,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:09"}}}
15:59:48.910 MQT: stat/webcam8/STATUS12 = {"StatusSTK":{"Exception":6,"Reason":"IntegerDivideByZero","EPC":"40168840","EXCVADDR":"00000000","CallChain":["4016883d","401617ae","40161415","40105791","40105be9","40103aa1","4011fab8","400ff591","400fa02e","400fbcb5","400fe40c","4011c806","40103aa1","4012e8a0","400e16b9"]}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior: Update ESP32 Cam from 12.0.2 to 12.1.0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Device should work as before

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2022-08-18:
I'M using this ESP Modul:
[https://templates.blakadder.com/ai-thinker_ESP32-CAM.html](https://templates.blakadder.com/ai-thinker_ESP32-CAM.html)

helmar74, 2022-08-18:
Downgrading to 12.0.2 brings back the main menu:
[image]

MaceMoneta, 2022-08-18:
I just completed upgrade of 5 esp32-cam (AI Thinker) from 12.0.2 to 12.1.0, without issue. It would be helpful to see the weblog immediately after the upgrade (since you say you can get to the console). It was omitted from your report.

helmar74, 2022-08-18:
Thanks for your reply !
Here we go:
```
00:00:00.003 HDW: ESP32-D0WDQ6 (PSRAM)
00:00:00.003 HDW: FoundPSRAM=1 CanUsePSRAM=1
00:00:00.038 UFS: FlashFS mounted with 304 kB free
00:00:00.082 CFG: Loaded from File, Count 83
00:00:00.402 CFG: Saved, Count 84, Bytes 4096
00:00:00.409 QPC: Count 1
00:00:00.412 ROT: Mode 1
00:00:00.428 BRY: GC from 3601 to 2842 bytes, objects freed 8/24 (in 1 ms) - slots from 39/61 to 22/61
00:00:00.451 CFG: No '*.autoconf' file found
00:00:00.454 BRY: GC from 3598 to 3089 bytes, objects freed 2/28 (in 1 ms) - slots from 39/61 to 25/61
00:00:00.454 BRY: Berry initialized, RAM used=3089 bytes
00:00:00.462 BRY: No 'preinit.be'
00:00:00.463 SRC: Restart
00:00:00.466 Project tasmota - webcam8 Version 12.1.0(webcam)-2_0_4_1(2022-08-18T10:33:47)
00:00:00.491 BRY: No 'autoexec.be'
00:00:00.769 WIF: Checking connection...
00:00:00.770 WIF: Attempting connection...
00:00:00.208 WIF: Connecting to AP1 hs.net_aussen_24GHz Channel 11 BSSId 34:31:C4:76:5C:90 in mode 11n as webcam8...
00:00:01.363 WIF: Checking connection...
00:00:01.364 WIF: Connected
00:00:01.618 HTP: Blank Device - Initial Configuration
00:00:01.620 HTP: Web server active on webcam8 with IP address 192.168.1.72
00:00:02.582 NTP: Sync time...
00:00:02.645 RTC: UTC 2022-08-18T17:08:40, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
18:08:40.000 RTC: Synced by NTP
18:08:40.207 MQT: Attempting connection...
18:08:40.234 MQT: Connected
18:08:40.240 MQT: tele/webcam8/LWT = Online (retained)
18:08:40.245 MQT: cmnd/webcam8/POWER = 
18:08:40.247 MQT: Subscribe to cmnd/webcam8/#
18:08:40.251 MQT: Subscribe to cmnd/tasmotas/#
18:08:40.255 MQT: Subscribe to cmnd/webcam8_fb/#
18:08:40.260 MQT: tele/webcam8/INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"12.1.0(webcam)","FallbackTopic":"cmnd/webcam8_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:08:40.278 MQT: tele/webcam8/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"webcam8","IPAddress":"192.168.1.72"}}
18:08:40.284 MQT: tele/webcam8/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":44}}
18:08:40.299 TFS: File 'autoexec.bat' not found
18:08:42.451 HTP: Console
18:08:44.588 QPC: Reset
18:08:45.584 BRY: GC from 4704 to 3056 bytes, objects freed 25/30 (in 1 ms) - slots from 67/76 to 31/76
18:08:45.594 MQT: tele/webcam8/STATE = {"Time":"2022-08-18T18:08:45","Uptime":"0T00:00:09","UptimeSec":9,"Heap":152,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":2,"Objects":30},"Wifi":{"AP":1,"SSId":"hs.net_aussen_24GHz","BSSId":"34:31:C4:76:5C:90","Channel":11,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:08:45.637 MQT: tele/webcam8/SENSOR = {"Time":"2022-08-18T18:08:45","ESP32":{"Temperature":69.4},"TempUnit":"C"}
18:08:46.599 APP: Boot Count 44
18:08:47.507 CFG: Saved, Count 85, Bytes 4096
18:08:50.617 MQT: tasmota/discovery/F008D1D2F290/config = {"ip":"192.168.1.72","dn":"webcam8","fn":["webcam8",null,null,null,null,null,null,null],"hn":"webcam8","mac":"F008D1D2F290","md":"ESP32-DevKit","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.1.0","t":"webcam8","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
18:08:50.630 MQT: tasmota/discovery/F008D1D2F290/sensors = {"sn":{"Time":"2022-08-18T18:08:50","ESP32":{"Temperature":68.9},"TempUnit":"C"},"ver":1} (retained)
```

helmar74, 2022-08-18:
I just replaced the HW Module to a new one. Same issue!

helmar74, 2022-08-18:
Downgrading to 12.0.2 brought Main Page of the Webinterface back!

MaceMoneta, 2022-08-18:
I don't see the "CAM: Initialized" message. Can you try running WcInit in the console of 12.1.0 and see if it initializes the camera properly? If so, is the main menu then correct?

sfromis, 2022-08-18:
Testing a bit, I'm aqlso seeing some RestartReason IntegerDivideByZero (which is also in the `Status 0` output reported). Hunting down after which build it started, led me to https://github.com/arendst/Tasmota/commit/e79cd1b34268bb5820603727925b86a6b4e37586 https://github.com/arendst/Tasmota/pull/16182 as the first one it started to fail with, @Jason2866

helmar74, 2022-08-18:
> WcInit

Hi, i tried. Didn't work... Same issue

Jason2866, 2022-08-18:
> Testing a bit, I'm also seeing some RestartReason IntegerDivideByZero (which is also in the `Status 0` output reported). Hunting down after which build it started, led me to [e79cd1b](https://github.com/arendst/Tasmota/commit/e79cd1b34268bb5820603727925b86a6b4e37586) #16182 as the first one it started to fail with, @Jason2866

Weird, since there is no active code

gemu2015, 2022-08-19:
since the camera driver is a precompiled library we need to update the lib and include file in sync.

in this enum the PIXFORMAT_YUV420 was inserted in between which effects PIXFORMAT_JPEG we use.

comment the YUV420 and it works again

typedef enum {
    PIXFORMAT_RGB565,    // 2BPP/RGB565
    PIXFORMAT_YUV422,    // 2BPP/YUV422
    //PIXFORMAT_YUV420,    // 1.5BPP/YUV420
    PIXFORMAT_GRAYSCALE, // 1BPP/GRAYSCALE
    PIXFORMAT_JPEG,      // JPEG/COMPRESSED
    PIXFORMAT_RGB888,    // 3BPP/RGB888
    PIXFORMAT_RAW,       // RAW
    PIXFORMAT_RGB444,    // 3BP2P/RGB444
    PIXFORMAT_RGB555,    // 3BP2P/RGB555
} pixformat_t;

helmar74, 2022-08-19:
> since the camera driver is a precompiled library we need to update the lib and include file in sync.
>
> in this enum the PIXFORMAT_YUV420 was inserted in between which effects PIXFORMAT_JPEG we use. [...]

Could you please give me detailed information, in which file I have to change what, I can compile the firmware and give it a try...

Thanks in advance

helmar74, 2022-08-19:
I commented PIXFORMAT_YUV420 out in file
```
...Tasmota-master_v1210_webcamfix\lib\libesp32\esp32-camera\driver\include\sensor.h
```
[image]

Now it works like charme ;-)

Thank you very much.
Maybe this should be flaged as BUG and be changed in next dev release?
I'll keep this issue open...

Regards

helmar

Jason2866, 2022-08-19:
Please try the development build firmware with the meged PR #16262.
I than will close this issue.

gemu2015, 2022-08-19:
works with latest dev!

sfromis, 2022-08-19:
`"Version":"12.1.0.1(webcam)","BuildDateTime":"2022-08-19T07:37:26"` works for me.

arendst, 2022-08-19:
``"Version":"12.1.0(webcam)","BuildDateTime":"2022-08-19T08:32:24"`` works for  me too ;-)

Updated binaries are now in release v12.1.0 too.

helmar74, 2022-08-19:
> `"Version":"12.1.0(webcam)","BuildDateTime":"2022-08-19T08:32:24"` works for me too ;-)
> 
> Updated binaries are now in release v12.1.0 too.

Updated with current stable binary, that worked fine

PR #16262 (2022-08-19): Revert "Update Webcam lib `*.h` files"
Reverts #16182 to fix #16251
Info this files have been updated since the webcam source code was updated.
Since the new webcam drivers worked unstable the previous known driver commit was used again. In this step i have forgotten to adjust the `*.h` files here to the used previous ones.
closed 2022-08-19 by Jason2866