arendst/Tasmota #16263: DFRobot AS3935 issues

rb2k, 2022-08-19:
### PROBLEM DESCRIPTION
I am trying to get the DFRobot AS3935 ((https://www.dfrobot.com/product-1828.html)) to work on Tasmota.
It's hooked up to an ESP32 (M5Stack Atom Lite).

I can see it in the console show up as an I2C device:

> 22:43:50.772 CMD: i2cscan
> 22:43:50.801 RSL: RESULT = {"I2CScan":"Device(s) found at 0x03"}

I configured the IRQ Pin and can confirm that the i2cdriver48 is enabled. 'i2cdriver' shows it, manually ran "i2cdriver48 1" just to make sure.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): M5Stack Atom Lite + DFRobot AS3935 sensor
- [X] Tasmota binary firmware version number used: 12.1.0.1 
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Web Interface + OTA update afterwards
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:23:27.647 CMD: Backlog Template; Module; GPIO 255
14:23:27.692 RSL: RESULT = {"NAME":"M5Stack Atom Lite","GPIO":[1,1,1,1,1,1,1,1,1056,1,1,1,1,1,1,1,0,1,1,1,0,1,640,1376,0,0,0,0,608,1,1,1,1,0,0,32],"FLAG":0,"BASE":1}
14:23:27.923 RSL: RESULT = {"Module":{"0":"M5Stack Atom Lite"}}
14:23:28.185 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"1056":"IRsend1"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"608":"I2C SCL1"},"GPIO22":{"0":"None"},"GPIO23":{"4672":"AS3935"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"1376":"WS28121"},"GPIO32":{"608":"I2C SCL1"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"32":"Button1"}}
```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- [ ] Provide the output of this command: `Status 0`:
```lua
  14:24:00.325 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"LightningDetector","FriendlyName":["LightningDetector"],"Topic":"tasmota_A6CB28","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:24:00.371 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:07:09","StartupUTC":"2022-08-19T13:16:51","Sleep":50,"CfgHolder":4617,"BootCount":41,"BCResetTime":"2022-08-18T20:36:15","SaveCount":83}}
14:24:00.400 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.0.1(tasmota)","BuildDateTime":"2022-08-19T10:14:16","Core":"2_0_4_1","SDK":"v4.4.3","CpuFrequency":80,"Hardware":"ESP32-PICO-D4 rev.1","CR":"395/699"}}
14:24:00.421 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Lambda-IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
14:24:00.468 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1414,"Free":1792,"Heap":142,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":2,"Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
14:24:00.526 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-A6CB28-2856","IPAddress":"192.168.1.154","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"4C:75:25:A6:CB:28","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:24:00.562 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A6CB28","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:24:00.594 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-08-19T13:24:00","Local":"2022-08-19T14:24:00","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:48","Sunset":"19:58"}}
14:24:00.620 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-08-19T14:24:00","ESP32":{"Temperature":55.0},"TempUnit":"C"}}
14:24:00.633 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-08-19T14:24:00","Uptime":"0T00:07:09","UptimeSec":429,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:00:00.003 HDW: ESP32-PICO-D4 
00:00:00.003 HDW: FoundPSRAM=0 CanUsePSRAM=0
00:00:00.084 UFS: FlashFS mounted with 44 kB free
00:00:00.179 CFG: Loaded from File, Count 63
00:00:00.195 QPC: Count 1
00:00:00.197 CFG: CR 403/699, Busy 0
00:00:00.213 ROT: Mode 1
00:00:00.420 CFG: No '*.autoconf' file found
00:00:00.426 BRY: Berry initialized, RAM used=3735 bytes
00:00:00.447 BRY: No 'preinit.be'
00:00:00.452 SRC: Restart
00:00:00.468 Project tasmota - LightningDetector Version 12.1.0(tasmota)-2_0_4_1(2022-08-18T10:30:21)
00:00:00.470 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.639 BRY: No 'autoexec.be'
00:00:00.951 WIF: Attempting connection...
00:00:01.278 WIF: Connecting to AP1 Lambda-IOT Channel 6 BSSId D0:21:F9:8D:1F:E3 in mode 11n as tasmota-A6CB28-2856...
00:00:01.365 WIF: Attempting connection...
00:00:03.367 WIF: Connected
00:00:03.577 HTP: Blank Device - Initial Configuration
00:00:03.580 HTP: Web server active on tasmota-A6CB28-2856 with IP address 192.168.1.154
00:00:03.787 RTC: UTC 2022-08-18T22:04:05, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
23:04:05.001 RTC: Synced by NTP
23:04:06.604 RSL: INFO1 = {"Info1":{"Module":"M5Stack Atom Lite","Version":"12.1.0(tasmota)","FallbackTopic":"cmnd/DVES_A6CB28_fb/","GroupTopic":"cmnd/tasmotas/"}}
23:04:06.619 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-A6CB28-2856","IPAddress":"192.168.1.154"}}
23:04:06.623 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":32}}
23:04:06.637 RSL: RESULT = {"POWER":"OFF"}
23:04:06.639 RSL: POWER = OFF
23:04:06.650 TFS: File 'autoexec.bat' not found
23:04:08.727 QPC: Reset
23:04:10.716 APP: Boot Count 32
23:04:10.726 RSL: STATE = {"Time":"2022-08-18T23:04:10","Uptime":"0T00:00:10","UptimeSec":10,"Heap":151,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:04:10.778 RSL: SENSOR = {"Time":"2022-08-18T23:04:10","ESP32":{"Temperature":55.6},"TempUnit":"C"}
23:04:11.339 CFG: Saved, Count 64, Bytes 4096
```

### TO REPRODUCE
Connect DFRobot AS3935 to ESP32 and configure I2C and IRQ Pin

### EXPECTED BEHAVIOUR
I would like to see the sensor data show up, but aside from the i2cscan it does not

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
They have a library for it over here: https://github.com/DFRobot/DFRobot_AS3935

sfromis, 2022-08-19:
What barbudor suggested was to open a **discussion**, not creating an "issue" which is for bugs. What you have looks more like needing help making it work. As indicated in what I posted both on Discord, and in the other thread, I had no trouble getting Tasmota to recognize AS3935 and report values. Did nothing special, just the recommended wiring and Tasmota config.

rb2k, 2022-08-19:
Happy to move it to a discussion, but given that the I2C communication seems to work and I can't see any of the usual issues (not using a sensors build, not having the driver enabled, not setting up the IRQ pin) I would suspect some sort of issues during the initialization of the driver for this particular DFRobot build.

barbudor, 2022-08-19:
Can you set `WebLog 3` and restart tasmota (`restart 1`)
The AS driver may put some more details at boot

rb2k, 2022-08-19:
> Can you set `WebLog 3` and restart tasmota (`restart 1`)
> The AS driver may put some more details at boot

That output is already in the original post. Sadly nothing :(

barbudor, 2022-08-19:
Ok, I will build a more talkative version of Tasmota32 with more logs to track line by line

barbudor, 2022-08-19:
Here you are, based on latest dev branch
@sfromis could you please test also with this tasmota32 version so we can compare logs ?
thanks
[tasmota32-as3935.zip](https://github.com/arendst/Tasmota/files/9383991/tasmota32-as3935.zip)

rb2k, 2022-08-19:
Interesting:

I think this part is the important part:

```
00:00:00.575 AS3935Detect: let's try to find out chip
00:00:00.581 AS3935init: reset done
00:00:00.582 AS3935init: false afe_gb=0x2E
```

The whole thing:
```
00:00:00.553 SRC: Restart
00:00:00.569 Project tasmota - LightningDetector Version 12.1.0.1(tasmota)-2_0_4_1(2022-08-19T18:51:28)
00:00:00.571 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.575 AS3935Detect: let's try to find out chip
00:00:00.581 AS3935init: reset done
00:00:00.582 AS3935init: false afe_gb=0x2E
00:00:00.780 BRY: No 'autoexec.be'
00:00:00.059 WIF: Attempting connection...
00:00:01.344 WIF: Connecting to AP1 Lambda-IOT Channel 6 BSSId D0:21:F9:8D:1F:E3 in mode 11n as tasmota-A6CB28-2856...
00:00:01.427 WIF: Attempting connection...
00:00:03.429 WIF: Connected
00:00:03.640 HTP: Blank Device - Initial Configuration
00:00:03.643 HTP: Web server active on tasmota-A6CB28-2856 with IP address 192.168.1.154
00:00:03.841 RTC: UTC 2022-08-19T17:03:56, DST 2022-03-27T02:00:00, STD 2022-10-30T03:00:00
18:03:56.000 RTC: Synced by NTP
18:03:57.659 RSL: INFO1 = {"Info1":{"Module":"M5Stack Atom Lite","Version":"12.1.0.1(tasmota)","FallbackTopic":"cmnd/DVES_A6CB28_fb/","GroupTopic":"cmnd/tasmotas/"}}
18:03:57.674 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-A6CB28-2856","IPAddress":"192.168.1.154"}}
18:03:57.678 RSL: INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":46}}
18:03:57.692 RSL: RESULT = {"POWER":"OFF"}
18:03:57.694 RSL: POWER = OFF
18:03:57.706 TFS: File 'autoexec.bat' not found
18:03:59.689 QPC: Reset
18:04:01.678 APP: Boot Count 46
18:04:01.688 RSL: STATE = {"Time":"2022-08-19T18:04:01","Uptime":"0T00:00:10","UptimeSec":10,"Heap":151,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:04:01.740 RSL: SENSOR = {"Time":"2022-08-19T18:04:01","ESP32":{"Temperature":53.3},"TempUnit":"C"}
18:04:02.879 CFG: Saved, Count 92, Bytes 4096
18:09:01.715 RSL: STATE = {"Time":"2022-08-19T18:09:01","Uptime":"0T00:05:10","UptimeSec":310,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:09:01.767 RSL: SENSOR = {"Time":"2022-08-19T18:09:01","ESP32":{"Temperature":53.3},"TempUnit":"C"}
18:13:51.707 APP: Serial logging disabled
18:14:01.667 RSL: STATE = {"Time":"2022-08-19T18:14:01","Uptime":"0T00:10:10","UptimeSec":610,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:14:01.677 RSL: SENSOR = {"Time":"2022-08-19T18:14:01","ESP32":{"Temperature":53.3},"TempUnit":"C"}
18:19:01.680 RSL: STATE = {"Time":"2022-08-19T18:19:01","Uptime":"0T00:15:10","UptimeSec":910,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:19:01.690 RSL: SENSOR = {"Time":"2022-08-19T18:19:01","ESP32":{"Temperature":53.3},"TempUnit":"C"}
18:24:01.707 RSL: STATE = {"Time":"2022-08-19T18:24:01","Uptime":"0T00:20:10","UptimeSec":1210,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:24:01.718 RSL: SENSOR = {"Time":"2022-08-19T18:24:01","ESP32":{"Temperature":53.3},"TempUnit":"C"}
18:29:01.709 RSL: STATE = {"Time":"2022-08-19T18:29:01","Uptime":"0T00:25:10","UptimeSec":1510,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:29:01.720 RSL: SENSOR = {"Time":"2022-08-19T18:29:01","ESP32":{"Temperature":53.3},"TempUnit":"C"}
18:34:01.703 RSL: STATE = {"Time":"2022-08-19T18:34:01","Uptime":"0T00:30:10","UptimeSec":1810,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"OFF","Dimmer":10,"Color":"1A1A1A","HSBColor":"0,0,10","Channel":[10,10,10],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Lambda-IOT","BSSId":"D0:21:F9:8D:1F:E3","Channel":6,"Mode":"11n","RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:34:01.721 RSL: SENSOR = {"Time":"2022-08-19T18:34:01","ESP32":{"Temperature":55.6},"TempUnit":"C"}
```

sfromis, 2022-08-19:
Here's what I got:
```
00:00:00.553 AS3935Detect: let's try to find out chip
00:00:00.555 AS3935init: reset done
00:00:00.556 AS3935init: true
00:00:00.557 I2C: AS3935 found at 0x3
00:00:00.558 AS3935Detect: GPIO_AS3935 is used
00:00:00.559 AS3935Setup: Settings.as3935_sensor_cfg[0]=0x24 
00:00:00.571 AS3935CalibrateRCO: true
```

barbudor, 2022-08-19:
Thanks a lot
Let me get that datasheet to understand that afe_gb value

rb2k, 2022-08-19:
I wonder if this is what their 'official' library checks against over here:

https://github.com/DFRobot/DFRobot_AS3935/blob/master/DFRobot_AS3935_I2C.cpp#L349

(Just a guess... I'm not that great when it comes to reading these)

barbudor, 2022-08-19:
Ok, so the driver sends a PRESET_DEFAULT and from the datasheet, the AFE_GB (gaib boos) should be 0x24
So 
- either we don't  leave anough time to the chip to set itself to the default
- your chip has different default values for some reason (different version, chinese clone ?)

Here is a test version with 10ms delay after reset, instead of 2 ms (I have not found in the datasheet any details on the expect reset delay so that is just a guess)
If not, we could try a larger window value for that test
[tasmota32-as3935-v2.zip](https://github.com/arendst/Tasmota/files/9384777/tasmota32-as3935-v2.zip)

barbudor, 2022-08-19:
Sorry, I missed your previous message pointing to DFrobot code
I will also check that

rb2k, 2022-08-19:
10 ms delay logs:

```
00:00:00.389 Project tasmota - LightningDetector Version 12.1.0.1(tasmota)-2_0_4_1(2022-08-19T21:10:08)
00:00:00.392 ETH: No ETH MDC and/or ETH MDIO GPIO defined
00:00:00.395 AS3935Detect: let's try to find out chip
00:00:00.409 AS3935init: reset done
00:00:00.410 AS3935init: false afe_gb=0x2E
```

barbudor, 2022-08-19:
ok, let's move further with that version that accept either 0x24 or 0x2E

[tasmota32-as3935-v3.zip](https://github.com/arendst/Tasmota/files/9384819/tasmota32-as3935-v3.zip)

rb2k, 2022-08-19:
Why hello!

```
00:00:00.410 AS3935Detect: let's try to find out chip
00:00:00.424 AS3935init: reset done
00:00:00.425 AS3935init: true afe_gb=0x2E
00:00:00.426 I2C: AS3935 found at 0x3
00:00:00.427 AS3935Detect: GPIO_AS3935 is used
00:00:00.427 AS3935Setup: Settings.as3935_sensor_cfg[0]=0x00 
00:00:00.474 AS3935CalibrateRCO: true
```

Also in the UI now:
[image]

barbudor, 2022-08-19:
Ok, at least we passed through the test
That doen't mean it is working

Any lighning expected today on your side ?
I'd like to have some confirmation before comitting this as permanent

rb2k, 2022-08-19:
None expected today. I wonder if any of the commands would show that it's communicating correctly?

https://tasmota.github.io/docs/AS3935/#commands-and-use

barbudor, 2022-08-19:
`AS3935settings` will read the settings in the chip (and in Tasmota)
You can change some chip settings and see if they are taken into account
`AS3935disturber` with 0 or 1
`AS3935setnf` set noise floor 0 to 7
`AS3935setrej` set spike rejection 0 to 15
etc ...

rb2k, 2022-08-19:
Sadly doesn't seem to stick:

20:43:42.414 CMD: AS3935disturber 1
20:43:42.418 SRC: WebConsole from 192.168.1.155
20:43:42.423 CMD: Grp 0, Cmd 'AS3935DISTURBER', Idx 1, Len 1, Pld 1, Data '1'
20:43:42.445 MQT: stat/lightning_detector_A6CB28/RESULT = {"AS3935":{"disturber":"Off"}}

AS3935SETTINGS remains unchanged

barbudor, 2022-08-19:
So your problem may be a deffective module
Do you have another one ?
May be try another source than DFrobot ?
I understand that's not the cheapest sensor ...

rb2k, 2022-08-19:
Yeah, I'll probably have to take a look at that.

I think my next step is to hook it up to an arduino or raspberry pi and see what it does with their 'native' code :(

barbudor, 2022-08-19:
I will put one in my next Ali order. Probably early next week

sfromis, 2022-08-19:
The breakout board that worked for me is [GY-AS3935](https://tasmota.github.io/docs/AS3935/#gy-as3935-board). Got it a couple of years ago.

device111, 2022-08-20:
ok, i have found a User that have also a issue with the DF-Robot AS3935 and the ESP32:
[https://www.komputer.de/zen/index.php?main_page=product_reviews_info&products_id=509&reviews_id=57&zenid=ai0u7f0r6q1h7b5rka6tum1480](url) (Its' a German comment).

I'm not a Hardware Guru but i found in the Schematic from DFRobot, that the Pullup Resistor from SDA is missing.
Could that be the Issue for the communication Problem?

Here the Schematic from MOD1016:

[image]

and here from the DF-Robot Modul:

[image]

It looks like that the internall Pullup resistors are too weak:
https://esp32.com/viewtopic.php?t=379

Maybe someone can answer with more experience with the ESP32 and I2C pullups...

device111, 2022-08-20:
@rb2k 
Do you have the possibility to switch a 10k resistor between SDA and VCC as a test?

rb2k, 2022-08-20:
Oh neat. Yes, I should be able to make that happen once I’m back home!

device111, 2022-08-20:
ok, great!

there are 4 measuring points at the backside of the module but i can not determine the order and labeling:

[image]

T1-VCC
T2-SDA
T3-SCL
T4-GND

In the best case, it is like the same as at the frontside, but beware, it must not be the same. Please measure out the points.

then do a 10k resistor between T1 and T2.

good luck!

sfromis, 2022-08-20:
The board overview from their docs suggests a different numbering, assuming that the pads match the socket on the other side.
[image]