arendst/Tasmota #16269: DNS lookup fails for *.local domains

rogg, 2022-08-20:
### PROBLEM DESCRIPTION
Resolving *.local doesn't work since "Add user control over DNS timeout reducing blocking"  https://github.com/arendst/Tasmota/commit/c988ba16451186c55873793978e7730619a25074 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): nodeMCU
- [x] Tasmota binary firmware version number used: development / 12.1.0.1(tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
05:38:09.888 CMD: Backlog Template; Module; GPIO 255
05:38:09.910 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
05:38:10.135 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
05:38:10.389 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
05:38:46.308 CMD: status 0
05:38:46.313 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"node","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
05:38:46.339 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:17","StartupUTC":"2022-08-20T04:36:29","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2022-08-20T05:36:36","SaveCount":7,"SaveAddress":"F5000"}}
05:38:46.372 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.0.1(tasmota)","BuildDateTime":"2022-08-19T21:33:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
05:38:46.392 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXXX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
05:38:46.423 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004084","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,38,45,62","Sensors":"1,2,3,4,5,6"}}
05:38:46.454 RSL: STATUS5 = {"StatusNET":{"Hostname":"node-2316","IPAddress":"10.0.1.37","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.1.1","DNSServer2":"0.0.0.0","Mac":"BC:DD:C2:9D:69:0C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
05:38:46.476 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9D690C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
05:38:46.499 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-08-20T04:38:46","Local":"2022-08-20T05:38:46","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:50","Sunset":"19:56"}}
05:38:46.517 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-08-20T05:38:46"}}
05:38:46.523 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-08-20T05:38:46","Uptime":"0T00:02:17","UptimeSec":137,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"XXXXXXX","BSSId":"XX:XX:XX:XX:XX:XX","Channel":6,"Mode":"11n","RSSI":74,"Signal":-63,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
05:45:22.116 CMD: ping1 pi.local
05:45:22.118 SRC: Serial
05:45:22.120 CMD: Grp 0, Cmd 'PING', Idx 1, Len 12, Pld -99, Data 'pi.local'
05:45:22.148 DNS: Unable to resolve 'pi.local'
05:45:22.151 RSL: RESULT = {"Ping":"Unable to resolve IP address"}
05:45:22.264 RSL: RESULT = {"Ping":{"pi.local":{"Reachable":false,"IP":"","Success":false}}}
```

### TO REPRODUCE
pinging *.local domains fails to resolve.

### EXPECTED BEHAVIOUR
DNS should resolve *.local domains.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
The only change I've made is to build with "USE_PING" to diagnose the issue. Was there a reason for using a homebrewed DNS lookup besides the ability to set a timeout? Because if not, the old WiFi.hostByName(), also has a parameter to set the timeout and works for .local.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-20:
Where the ESP8266 core supports a timeout option the ESP32 does not. The latter uses a fixed timeout of 16 seconds!!

By providing the current code timeouts can be reduced solving user noticed "hang" situations.

If anyone can find a way to fix your ".local" issue while keeping the "hangs" as short a spossible be my guest.

Jason2866, 2022-08-20:
Using `*.local` is a bad idea in general since it is used by mDNS and this conflicts.
For your local devices use a extension which is NOT used from the crappy mDNS.
For example `*.lan`. 
Do a search in your favorite search engine about problems with `.local`. ONLY reliable solution to solve the problems generated from is not to use.

PR #16273 (2022-08-21): mDNS resolution for *.local fixes #16269
"Add user control over DNS timeout reducing blocking"  https://github.com/arendst/Tasmota/commit/c988ba16451186c55873793978e7730619a25074 created DnsClient so that a timeout can be used to prevent blocking on ESP32, but it can't do one-shot mDNS queries. Use WiFi.hostByName () for those.

## Description:

**Related issue (if applicable):** fixes #16269

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles wi [... cut]
closed 2022-08-20 by Jason2866