arendst/Tasmota #16274: sunset/sunraise german translation swapped

tubercel, 2022-08-21:
### PROBLEM DESCRIPTION
under Settings -> Timers > timer   german translation for sunset/sunraise are swapped.

closed 2022-08-21 by close-issue-app[bot]

barbudor, 2022-08-21:
No it isn't
#define D_SUNRISE "Sonnenaufgang"
#define D_SUNSET "Sonnenuntergang"
(@Jason2866 Can you please confirm ?)

Your configuration is somehow or you need to upgrade to latest release
Read https://github.com/arendst/Tasmota/issues/15816
- You must have a proper NTP service providing time
- You must have your timezone settings correctly set
- You must have you Latitude and Longitude properly set

tubercel, 2022-08-21:
Hi

I am using Tasmota 12.1.0

I have setp up ntp server,
longitude / latituce settings,
Timezone and DST Settings.

Time is correct on device.

But have a look at the attached Picture!

Sonnenaufgang (20:13)
Sonnenuntergang (05:47)

Seems wrong to me.

barbudor, 2022-08-21:
Are the defines wrong? 
Google translate say they are correct

Sonnenaufgang and Sonnenuntergang are in the right order on the GUI (sunrise at the top, sunset at the bottom) 
If the values are in the wrong order, then your config is wrong

barbudor, 2022-08-21:
When correctly configured:

[image]

tubercel, 2022-08-21:
I had a closer look.

When I Update OTA from 11.1.0 to 12.1.0 then devicetime is reset to epoch 0 (1.1.1970).

NTP does NOT kick in to set the correct time. (perhaps timeoffset too high?)

When I set the time via "Time" manually everything is fine again an revives reboots.

Verified with 3 devices (Gosund SP111) 7 left for testing.

-- 
Matthias Meyser
38678 Clausthal-Zellerfeld, Marktstrasse 40
Telefon: +49 5323 9839910
Fax:     +49 5323 9839917

barbudor, 2022-08-21:
May be again a DNS issue
Are you using a local NTP server or tasmota default?
Check your IP settings (IPAddress) you should have a gateway and a DNS server

Progaros, 2022-08-23:
This command fixed it for me:
`backlog IPAddress4 8.8.8.8; time 0`
(set dns to google & set time to ntp)

barbudor, 2022-08-24:
If your device have NTP servers specified with a name, you need a DNS.
If your device is configured with static IP, then of course you must set you DNS statically too. In DHCP, the DNS, as well as the gateway, is provided by the DHCP server

Not that you don't need to use an Internet NTP server. Your local network may already have a local NTP server. Some routers may be configured to provide NTP, also if you have a home automation server it can do that. In that case you can also set a local static IP for the NTP server.

Flexible

sfromis, 2022-08-24:
Many NAS boxes also have the feature of providing NTP. You can even have a Tasmota device act as a NTP server, if it has a real time source like RTC or GPS.

tubercel, 2022-08-24:
I finaly found the culprit.

There are 2 nameserver configured via DHCP.

The 2. one works without problems.

The 1. one only resolvs internal zones.
Any request for an external zone results in an "SERVFAIL" reply.
(DNSSEC managed Keys hosed)

When I fix this failure everything works fine.

IMHO the 2. nameserver should be used in this case.

barbudor, 2022-08-24:
AFAIK, Tasmota (or even may be Espressif SDK, not sure who is really in charge here) falls back to the 2nd server only if there is no answer, not for an error that is likely to be assumed as "not known"

barbudor, 2022-08-24:
From what I see, it is Arduino/Espressif who handles that part. Tasmota as no impact on the choice of DNS server.

sfromis, 2022-08-24:
Some refactoring around DNS was made recently in Tasmota itself, for better handling in some situations. Not looked into how this may affect this case.

tubercel, 2022-10-11:
DNS / IP Setting were ok an working.

Default Tasmota NTP Server.

When I set the device time from epoch 0 +-
to real time +-5minute.

e.g. by "Time 1661096000" the devicetime is
immideatly set to the exact real time.

I wonder why

1) Device Time ist reset to epoch 0 during update.

2) Why time is not updated by ntp in this situation.

-- 
Matthias Meyser
38678 Clausthal-Zellerfeld, Marktstrasse 40
Telefon: +49 5323 9839910
Fax:     +49 5323 9839917