arendst/Tasmota #16300: Cannot connect to MQTT over TLS

CodingSpiderFox, 2022-08-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff S202
- [X] Tasmota binary firmware version number used: 12.1.0
  - [X] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: 
21:21:25.013 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
1:26:12.291 SRC: Backlog
21:26:12.294 CMD: Grp 0, Cmd 'RULE', Idx 1, Len 0, Pld -99, Data ''
21:26:12.298 RSL: RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:26:12.506 SRC: Backlog
21:26:12.510 CMD: Grp 0, Cmd 'RULE', Idx 2, Len 0, Pld -99, Data ''
21:26:12.514 RSL: RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
21:26:12.755 SRC: Backlog
21:26:12.758 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 1, Pld -99, Data '#'
21:26:12.761 RUL: Stored uncompressed, would compress from 1 to 2 (-0%)
21:26:12.765 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":1,"Free":510,"Rules":"#"}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
1:27:10.463 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_115273","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:27:10.491 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:57","StartupUTC":"2022-08-22T20:19:13","Sleep":50,"CfgHolder":4617,"BootCount":149,"BCResetTime":"2021-08-09T22:03:56","SaveCount":273,"SaveAddress":"FA000"}}
21:27:10.523 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.0(tasmota)","BuildDateTime":"2022-08-19T08:30:16","Boot":6,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"419/699"}}
21:27:10.541 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<HIDDEN>",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00044000","00000000"]}}
21:27:10.575 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
21:27:10.606 RSL: STATUS5 = {"StatusNET":{"Hostname":"<HIDDEN>","IPAddress":"<HIDDEN>","Gateway":"<HIDDEN>","Subnetmask":"<HIDDEN>","DNSServer1":"<HIDDEN>","DNSServer2":"0.0.0.0","Mac":"<HIDDEN>","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
21:27:10.627 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"<HIDDEN>","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_<HIDDEN>","MqttUser":"user","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:27:10.650 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-08-22T20:27:10","Local":"2022-08-22T21:27:10","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:53","Sunset":"19:52"}}
21:27:10.668 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-08-22T21:27:10"}}
21:27:10.675 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-08-22T21:27:10","Uptime":"0T00:07:57","UptimeSec":477,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"<HIDDEN>","BSSId":"<HIDDEN>","Channel":1,"Mode":"11n","RSSI":100,"Signal":-35,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:30:31.846 MQT: Attempting connection...
21:30:31.868 MQT: Connect failed to <HIDDEN>:8883, rc -1. Retry in 120 sec
21:31:15.376 WIF: Sending Gratuitous ARP

21:33:32.772 CMD: Grp 0, Cmd 'MQTTFINGERPRINT', Idx 1, Len 0, Pld -99, Data ''
21:33:32.777 RSL: RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure MQTT Broker with TLS on port 8883 and unencrypted on port 1883
Configure WIFI on Sonoff S20
Configure MQTT Connection on Sonoff S20 without encryption -> works
Configure MQTT Port to 8883 (TLS) -> does not connect, seet weblog 4 output
Try to set MqttFingerprint command with web console. Error occurs, see second part of weblog 4 output.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect that a MqttFingerprint can be configured

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-08-22:
TLS not included in regular precompiled ESP8266 binaries.

closed 2022-08-23 by Jason2866

CodingSpiderFox, 2022-08-25:
thanks for the info :) would it be possible to provide prebuilt binaries with TLS support in the releases section of this repo?

barbudor, 2022-08-25:
The unoffical `tasmota-scripting` binary includes TLS
But as it has scripting, it doesn't have rules
Whatever, it takes only a matter of minutes to make a custom build using [TasmotaCompiler](https://github.com/benzino77/tasmocompiler/)