arendst/Tasmota #16308: Tasmota (POW) not creating ```state_class: total_increasing``` field in Home Assistant for Energy

sticilface, 2022-08-23:
### PROBLEM DESCRIPTION
Home Assistant requires TWO fields to be created for Energy (kWh) so that the Energy Dashboard can count it correctly.  
Below is screenshot of the tasmota created device.  
[image]
- ```state_class: total_increasing```  <----  Missing this entry
- ```unit_of_measurement: kWh```
- ```device_class: energy```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
There are issues relating to the ```device_class``` but none relating to ```state_class```
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4) 
Can't log in. went round in circles with the captcha 
- [X] Device used (e.g., Sonoff Basic): Snodd POW r1
- [X] Tasmota binary firmware version number used: 12.1.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: Buspirate
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
09:14:41.776 CMD: Backlog Template; Module; GPIO 255
09:14:41.798 MQT: stat/tasmota_20DC18/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:14:42.029 MQT: stat/tasmota_20DC18/RESULT = {"Module":{"6":"Sonoff Pow"}}
09:14:42.287 MQT: stat/tasmota_20DC18/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2592":"HLWBL SEL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"2656":"HLWBL CF1"},"GPIO14":{"2688":"HLW8012 CF"},"GPIO15":{"288":"Led1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:15:27.232 CMD: Status 0
09:15:27.239 MQT: stat/tasmota_20DC18/STATUS = {"Status":{"Module":6,"DeviceName":"TasmotaPow1","FriendlyName":["TasmotaPow1"],"Topic":"tasmota_20DC18","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
09:15:27.246 MQT: stat/tasmota_20DC18/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"3T22:05:44","StartupUTC":"2022-08-19T10:09:43","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-08-19T10:44:37","SaveCount":26,"SaveAddress":"FB000"}}
09:15:27.253 MQT: stat/tasmota_20DC18/STATUS2 = {"StatusFWR":{"Version":"12.1.0(tasmota)","BuildDateTime":"2022-08-18T10:30:01","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"350/699"}}
09:15:27.260 MQT: stat/tasmota_20DC18/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BoomBox",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
09:15:27.273 MQT: stat/tasmota_20DC18/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
09:15:27.281 MQT: stat/tasmota_20DC18/STATUS5 = {"StatusNET":{"Hostname":"tasmota-20DC18-7192","IPAddress":"192.168.1.121","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.20","DNSServer2":"0.0.0.0","Mac":"5C:CF:7F:20:DC:18","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:15:27.288 MQT: stat/tasmota_20DC18/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_20DC18","MqttUser":"andrew","MqttCount":7,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:15:27.298 MQT: stat/tasmota_20DC18/STATUS7 = {"StatusTIM":{"UTC":"2022-08-23T08:15:27","Local":"2022-08-23T09:15:27","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:54","Sunset":"19:50"}}
09:15:27.304 MQT: stat/tasmota_20DC18/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:15:27.316 MQT: stat/tasmota_20DC18/STATUS10 = {"StatusSNS":{"Time":"2022-08-23T09:15:27","ENERGY":{"TotalStartTime":"2022-08-19T10:44:37","Total":4.763,"Yesterday":1.368,"Today":0.536,"Power":52,"ApparentPower":79,"ReactivePower":59,"Factor":0.66,"Voltage":244,"Current":0.322}}}
09:15:27.325 MQT: stat/tasmota_20DC18/STATUS11 = {"StatusSTS":{"Time":"2022-08-23T09:15:27","Uptime":"3T22:05:44","UptimeSec":338744,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":7,"POWER":"ON","Wifi":{"AP":1,"SSId":"BoomBox","BSSId":"80:2A:A8:11:9D:EF","Channel":1,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Connect Sonoff POW to Homeassistant

### EXPECTED BEHAVIOUR
The field ```state_class: total_increasing``` should be created for the total powers, including power today

### SCREENSHOTS
see above

### ADDITIONAL CONTEXT
Great work, much appreciated.  I ran my own ESP8266 Manager software and switch stuff 
https://github.com/sticilface/ESPmanager
but tasmota was just far too tempting so i've switched all my devices to it.  Amazing work! Thank you.

**(Please, remember to close the issue when the problem has been addressed)**

sticilface, 2022-08-23:
It would appear so :( 
My apologies. 

I had in fact read that issue, and forgot. it stated that it was fixed, and the PR was 9m ago... then it was changed to a discussion. 

Should that issue be reopened and I close this one? Or are there plans to add it. Should be fairly straightforward....

barbudor, 2022-08-23:
If the PR didn't cover everything that is needed then it  needs someone with the skills and the interest in Home Assistant integration to make new PR
Personally I don't have a HA setup so that won't be me

arendst, 2022-08-24:
The PR noticed here is for the abandonend HASS discovery driver.

The new Tasmota discovery in HASS should handle this. The solution is likely to be found in https://github.com/emontnemery/hatasmota so pls ask here.

closed 2022-08-31 by Jason2866