arendst/Tasmota #16319: Delay with TL SS01S Swtch and v12.1.0

merdely, 2022-08-23:
### PROBLEM DESCRIPTION

I have several Treat Life SS01S switches in my house.

I upgraded from v10.1.0 to v12.1.0 and ALL of my SS01S switches started behaving badly. There is now ALWAYS a delay (maybe 2 seconds) when pressing the switch and the light (or fan) turning on. And sometimes (maybe 1 out of 3 times) the button press is not registered at all.

Note: I am not talking about the SetOption13 delay while detecting whether I double/triple/long press. This is a LONG delay.

As a test, I downgraded from v12.1.0 to v11.1.0 and the delay was gone. So I upgraded to v12.0.2 and everything was still working as expected. So I tried going back to v12.1.0 and the problem was immediately there again.

As a test, I had the fan in my bathroom on v12.0.2 and the light on v12.1.0.  I pressed both buttons at the same time.  The fan came on over a full second before the light.

So now I'm back on v12.0.2 and all of my switches are behaving properly.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Treatlife SS01S Switch
- [x] Tasmota binary firmware version number used: 12.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

18:03:08.151 MQT: stat/hall_bath_light/RESULT = {"NAME":"TL SS01S Swtch","GPIO":[0,0,0,0,288,576,0,0,224,32,0,0,0,0],"FLAG":0,"BASE":18}
18:03:08.377 MQT: stat/hall_bath_light/RESULT = {"Module":{"0":"TL SS01S Swtch"}}
18:03:08.631 MQT: stat/hall_bath_light/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"288":"Led1"},"GPIO5":{"576":"LedLink_i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

18:03:28.887 MQT: stat/hall_bath_light/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
18:03:29.114 MQT: stat/hall_bath_light/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
18:03:29.365 MQT: stat/hall_bath_light/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

18:04:06.638 MQT: stat/hall_bath_light/STATUS = {"Status":{"Module":0,"DeviceName":"Hall Bath Light","FriendlyName":["Hall Bath Light"],"Topic":"hall_bath_light","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:04:06.669 MQT: stat/hall_bath_light/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://repo.erdely.in/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:01:42","StartupUTC":"2022-08-23T22:02:24","Sleep":50,"CfgHolder":4617,"BootCount":67,"BCResetTime":"2021-01-10T21:49:36","SaveCount":13349,"SaveAddress":"FA000"}}
18:04:06.709 MQT: stat/hall_bath_light/STATUS2 = {"StatusFWR":{"Version":"12.1.0(tasmota)","BuildDateTime":"2022-08-18T10:30:01","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"429/699"}}
18:04:06.729 MQT: stat/hall_bath_light/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":2,"LogHost":"192.168.25.44","LogPort":514,"SSId":["Ross's Apartment",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["40008009","2805C80001000600003C5A0A002800000000","00008280","00006000","00004000","00000000"]}}
18:04:06.764 MQT: stat/hall_bath_light/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
18:04:06.801 MQT: stat/hall_bath_light/STATUS5 = {"StatusNET":{"Hostname":"hall-bath-light-0758","IPAddress":"192.168.26.136","Gateway":"192.168.25.1","Subnetmask":"255.255.252.0","DNSServer1":"192.168.25.1","DNSServer2":"0.0.0.0","Mac":"F4:CF:A2:E0:C2:F6","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:04:06.825 MQT: stat/hall_bath_light/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.erdely.in","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E0C2F6","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:04:06.849 MQT: stat/hall_bath_light/STATUS7 = {"StatusTIM":{"UTC":"2022-08-23T22:04:06","Local":"2022-08-23T18:04:06","StartDST":"2022-03-13T02:00:00","EndDST":"2022-11-06T02:00:00","Timezone":99,"Sunrise":"00:54","Sunset":"14:50"}}
18:04:06.869 MQT: stat/hall_bath_light/STATUS10 = {"StatusSNS":{"Time":"2022-08-23T18:04:06"}}
18:04:06.878 MQT: stat/hall_bath_light/STATUS11 = {"StatusSTS":{"Time":"2022-08-23T18:04:06","Uptime":"0T00:01:42","UptimeSec":102,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ross's Apartment","BSSId":"F6:92:BF:A1:F6:D1","Channel":6,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

18:05:31.467 CMD: weblog 4
18:05:31.475 MQT: stat/hall_bath_light/RESULT = {"WebLog":4}
18:05:31.580 CFG: Saved to flash at F9, Count 13350, Bytes 4096
18:05:38.453 APP: Button1 multi-press 1
18:05:39.457 SRC: Button
18:05:39.466 MQT: tele/hall_bath_light/STATE = {"Time":"2022-08-23T18:05:39","Uptime":"0T00:03:15","UptimeSec":195,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Ross's Apartment","BSSId":"F6:92:BF:A1:F6:D1","Channel":6,"Mode":"11n","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:05:39.493 MQT: stat/hall_bath_light/RESULT = {"POWER":"ON"}
18:05:39.498 MQT: stat/hall_bath_light/POWER = ON
18:05:39.574 CFG: Saved to flash at F8, Count 13351, Bytes 4096
18:05:43.237 APP: Button1 multi-press 1
18:05:44.240 SRC: Button
18:05:44.248 MQT: tele/hall_bath_light/STATE = {"Time":"2022-08-23T18:05:44","Uptime":"0T00:03:20","UptimeSec":200,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ross's Apartment","BSSId":"F6:92:BF:A1:F6:D1","Channel":6,"Mode":"11n","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
18:05:44.276 MQT: stat/hall_bath_light/RESULT = {"POWER":"OFF"}
18:05:44.281 MQT: stat/hall_bath_light/POWER = OFF
18:05:44.614 CFG: Saved to flash at F7, Count 13352, Bytes 4096
```

### TO REPRODUCE
1. Upgrade to v12.1.0
2. Press switch button to toggle light

### EXPECTED BEHAVIOUR
Light toggles in reasonable delay period while switch detects single, double, triple, hold press (less than a second?).

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
repo.erdely.in is a local mirror of the Tasmota releases from Github, unmodified.

**(Please, remember to close the issue when the problem has been addressed)**

merdely, 2022-08-23:
One thing I just noticed: In v12.1.0, the main page (http://192.168.26.136/?) does not display the template name at the top of the page above the friendly name. In v12.0.2, the main page DOES display "TL SS01S Swtch" above "Hall Bath Light".

TimDowker, 2022-08-23:
Can confirm on multiple different versions of light switches.

arendst, 2022-08-24:
See #15867. About to revert

tobiwashere, 2022-08-24:
I can confirm this behaviour on three Gosund SP111 smart sockets.

With this version, if you don't press the button long enough, the action (turn the socket on or off) won't be executed.

sfromis, 2022-08-24:
Actually, holding the button "too long" may make it slower, due to the need for detecting long or multiple presses. Normal latency is 0.5 seconds due to this, but you can use SetOption13 for immediate action, if you neither need long nor multiple presses.

tobiwashere, 2022-08-24:
> Actually, holding the button "too long" may make it slower, due to the need for detecting long or multiple presses. Normal latency is 0.5 seconds due to this, but you can use SetOption13 for immediate action, if you neither need long nor multiple presses.

Good to know, I wasn't aware of this option!
So with SetOption13 enabled the delay doesn't exist anymore for my socket, but with the previous Version 12.0.2 there was no delay even if SetOption13 was disabled. So probably just a workaround for people, who don't use the long press function.

sfromis, 2022-08-24:
The usual delay without SetOption13 is half a second (required for multipress detection), which to me feels a bit sluggish,but not like a WTF! delay of > 1 second.

tobiwashere, 2022-08-26:
Just upgraded my sockets to 12.1.1 and with SetOption13 disabled the behaviour is as expected and as it was in version 12.0.x.
Thank you @arendst 😊

closed 2022-08-26 by arendst