arendst/Tasmota #16334: Template name no longer shown on main page since v12.1.0

steinthomas, 2022-08-24:
### PROBLEM DESCRIPTION
The template name is no longer displayed on the main page since new tasmota version 12.1.0.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP111
- [x] Tasmota binary firmware version number used: 12.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
19:39:15.365 CMD: Status 0
19:39:15.372 MQT: stat/tasmota_8C29ED/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_8C29ED","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
19:39:15.378 MQT: stat/tasmota_8C29ED/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"5T21:47:21","StartupUTC":"2022-08-18T20:51:54","Sleep":50,"CfgHolder":4617,"BootCount":107,"BCResetTime":"2020-12-29T23:44:07","SaveCount":3301,"SaveAddress":"F7000"}}
19:39:15.382 MQT: stat/tasmota_8C29ED/STATUS2 = {"StatusFWR":{"Version":"12.1.0(tasmota)","BuildDateTime":"2022-08-18T10:30:01","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"442/699"}}
19:39:15.386 MQT: stat/tasmota_8C29ED/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wlan1",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
19:39:15.398 MQT: stat/tasmota_8C29ED/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
19:39:15.404 MQT: stat/tasmota_8C29ED/STATUS5 = {"StatusNET":{"Hostname":"tasmota-8C29ED-2541","IPAddress":"192.168.1.124","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"84:0D:8E:8C:29:ED","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
19:39:15.412 MQT: stat/tasmota_8C29ED/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8C29ED","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:39:15.419 MQT: stat/tasmota_8C29ED/STATUS7 = {"StatusTIM":{"UTC":"2022-08-24T18:39:15","Local":"2022-08-24T19:39:15","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"05:55","Sunset":"19:48"}}
19:39:15.423 MQT: stat/tasmota_8C29ED/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:39:15.431 MQT: stat/tasmota_8C29ED/STATUS10 = {"StatusSNS":{"Time":"2022-08-24T19:39:15","ENERGY":{"TotalStartTime":"2020-12-29T23:44:07","Total":23.817,"Yesterday":0.022,"Today":0.010,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,"Voltage": 0,"Current":0.000}}}
19:39:15.437 MQT: stat/tasmota_8C29ED/STATUS11 = {"StatusSTS":{"Time":"2022-08-24T19:39:15","Uptime":"5T21:47:21","UptimeSec":510441,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Wlan1","BSSId":"68:D7:9A:5F:6F:94","Channel":1,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Before Update (with template name): v12.0.2
[image]

After Update (without template name): v12.1.0
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Is there a way to show the name again on main page?

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-08-24:
Dupplicate of https://github.com/arendst/Tasmota/discussions/16324
Already fixed
Please close

steinthomas, 2022-08-24:
Thanks, I'll wait for the new version then. I will close the ticket.

closed 2022-08-24 by steinthomas

barbudor, 2022-08-24:
It is already available in the development branch http://ota.tasmota.com/tasmota/
A workaround is also described in the above thread