arendst/Tasmota #16335: 12.1.0.1
Sonoff zbbridge reboots with system/restart reason. 

While testing new feature with .zb file to match tuya sensors (it is ok work), after several file uploads and restart with button from webui - my zbbridge starts reboot continuously.  
Working about 30 sec and reboot again with system restart reason.

I have turn off zbbridge power and turn on it again.
After that it working fine about 2 hours till now.

I didn't find any errors in logs.

midnight1979, 2022-08-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2022-09-04:
I can confirm the reboots with my recent flashed devices. Please ensure that you disable MQTT first or change the topic. Then the device is stable. It looks like it reboots, when getting zbsend commands on non existing channels. After fixing all zbsend commands i reconnect the MQTT and now everything is fine.

stefanbode, 2022-09-04:
There is no difference using 6.7.8 or 6.7.9 firmware of the NCP. same behavior of reboots after about 30seconds

s-hadinger, 2022-09-04:
It boots with any `.zb` file?
I suppose your talking about the eqp8266 based bridge

s-hadinger, 2022-09-04:
> It looks like it reboots, when getting zbsend commands on non existing channels

What do you mean with non existing channels?

stefanbode, 2022-09-04:
I have an openHAB integration where i send zbsend commands through mqtt. This is part of my automation. If the target 0xabdef does not exist looks like a reboot get initiated.

s-hadinger, 2022-09-04:
Oh this is bad. I will look at it

s-hadinger, 2022-09-23:
I can't reproduce. Can you check with latest version from development? v12.1.1.2

closed 2022-10-02 by s-hadinger

TimNe0, 2022-12-27:
I have the same issue on v12.3.1 - reboots constantly.  Haven't managed to keep it online for longer than about 1 minute.

@s-hadinger if this isn't something you can replicate do you want me to send you the hardware?