arendst/Tasmota #16339: Tracker for new button debounce issues

arendst, 2022-08-25:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
This issue can be used to track possible issues caused by button debounce rewrite

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
  None.

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
  None.

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
  None.

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
  None.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
None.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
None.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
None.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-25:
Latest dev v12.1.1.1 contains a new button driver based on the switch debounce code. It is default enabled within the driver file `` support_button_v3.ino`` as define BUTTON_V3.

In case of immediate fallback needed you can disable the previous define and enable define BUTTON_V2 in file ``support_button_v2.ino`` which holds the original button driver as used until today.

Any comments are welcome.

lugerun, 2022-08-28:
I've compiled it and installed it and so far no ghost switching occured.

Kedryn, 2022-10-13:
I have a Slamper that i upgraded from 10.1 to 12.1.1 and started with ghost switching on several times a day.  It turns on randomly, but it never turns off.
Nobody touched the small button on the Slamper since years. I've now went back to 10.1 to see if it happens again.

arendst, 2022-10-13:
@Kedryn the change was implemented in v12.1.1.1 and NOT in v12.1.1 so the ghost switching is probably caused by RF. If so change the template and remove the button from it. RF is NOT under control of Tasmota.

Kedryn, 2022-10-13:
ah, i missed a .1 then. Ok, i'll test the newer one
If for "RF" you mean the 433MHz receiver, the slamper 1 does not have it.

arendst, 2022-11-08:
As now released in v12.2 and no new issues reported I'll close this.

Thx for your support.

closed 2022-11-08 by arendst