arendst/Tasmota #16350: Devgroupshare command : input as decimal result as hex  is misleading

ouinouin, 2022-08-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
the Devgroupshare command result is showing result as hex value, but we input the command as decimal value 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): M5stack
- [x] Tasmota binary firmware version number used: 12.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: usbweb installer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
follow the steps to include our tasmota in a devgroup then issue the command devgroupshare 64,64 for example (sharing and receiving events only , the result of the command is displaying : RESULT = {"DevGroupShare":{"In":"40","Out":"40"}}
(40 si 64 in hex, but this result can be misleading).

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
expecting either to input the command as hex (no a good solution to my mind , or to get the result as decimal.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-08-27:
I mainly think it shouldn't be a string
Either `"In":0x40` or `"In":64` would be correct

While the command accept both decimal and hexadecimal inputs (as documented) , one format as to be chosen for the output but it have to be correct.

ouinouin, 2022-08-27:
exact i didnt even noticed this is a string , i would set this as decimal, maye easing the reuse of this somewhere in some rules or something like that.
however, in the documentation, i noticed it states : 
64 = Event , but I could also send some commands (devgroupsend1 193=power5 \1)  while having masked to whatever value, so maybe its worth to mention it in the documentation in : https://tasmota.github.io/docs/Device-Groups/#commands section DevGroupShare
and in 
https://tasmota.github.io/docs/Device-Groups/#commands containing the same table as in the commands section.