arendst/Tasmota #16353: RestartReason": {"Exception": 0, "Reason": "Exception","EPC"

EDSTOBI, 2022-08-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have problems with data not being recorded cleanly and daily values not being correct at one measuring point. After observation, I noticed that the Shelly 1PM boots uncontrollably. I have updated the Shelly to the latest firmware, no change. I removed the Shelly and replaced it with another one. The problem remained at the measurement source. The replaced Shelly runs without problems on another measurement source.
Now I have read out this information and need some information on what it means in order to continue troubleshooting.

It is a normal, non-electronically controlled solar pump which is controlled and measured by this.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): __Shelly 1pm
- [x] Tasmota binary firmware version number used: _12.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: __Tasmota web___
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
13:06:39.372 CMD: Backlog Template; Module; GPIO 255
13:06:39.501 MQT: Shelly_7/stat/RESULT = {"NAME":"SolarPumpe","GPIO":[320,0,0,0,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}
13:06:39.727 MQT: Shelly_7/stat/RESULT = {"Module":{"0":"SolarPumpe"}}
13:06:39.982 MQT: Shelly_7/stat/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}
13:06:41.431 MQT: Shelly_7/tele/STATE = {"Time":"2022-08-27T13:06:41","Uptime":"0T00:04:49","UptimeSec":289,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Sophos","BSSId":"00:1A:8C:A3:37:23","Channel":1,"Mode":"11n","RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:06:41.462 MQT: Shelly_7/tele/SENSOR = {"Time":"2022-08-27T13:06:41","Switch1":"ON","ANALOG":{"Temperature":67.2},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.332,"Yesterday":0.000,"Today":0.332,"Period": 0,"Power":33},"TempUnit":"C"}
13:06:41.742 MQT: Shelly_7/stat/RESULT = {"POWER":"ON"}
13:06:41.747 MQT: Shelly_7/stat/POWER = ON
13:06:51.408 MQT: Shelly_7/tele/STATE = {"Time":"2022-08-27T13:06:51","Uptime":"0T00:04:59","UptimeSec":299,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Sophos","BSSId":"00:1A:8C:A3:37:23","Channel":1,"Mode":"11n","RSSI":26,"Signal":-87,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:06:51.439 MQT: Shelly_7/tele/SENSOR = {"Time":"2022-08-27T13:06:51","Switch1":"ON","ANALOG":{"Temperature":67.1},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.332,"Yesterday":0.000,"Today":0.332,"Period": 0,"Power":33},"TempUnit":"C"}
13:06:51.666 MQT: Shelly_7/stat/RESULT = {"POWER":"ON"}
13:06:51.672 MQT: Shelly_7/stat/POWER = ON

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
13:07:46.459 MQT: Shelly_7/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":152,"Free":359,"Rules":"on Switch1#state=ON do Publish Shelly_7/stat/statua/ {\"schalter ist an\"} on Switch1#state=OFF do Publish Shelly_7/stat/statua/ {\"schalter ist aus\"}ENDON"}}
13:07:46.700 MQT: Shelly_7/stat/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
13:07:46.949 MQT: Shelly_7/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
13:08:37.965 MQT: Shelly_7/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Shelly_7","FriendlyName":["Shelly_7"],"Topic":"Shelly_7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
13:08:37.997 MQT: Shelly_7/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:06:45","StartupUTC":"2022-08-27T12:01:52","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2022-08-15T10:44:05","SaveCount":33,"SaveAddress":"FB000"}}
13:08:38.002 MQT: Shelly_7/stat/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:33:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"358/699"}}
13:08:38.022 MQT: Shelly_7/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Sophos",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
13:08:38.055 MQT: Shelly_7/stat/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15400B","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
13:08:38.090 MQT: Shelly_7/stat/STATUS5 = {"StatusNET":{"Hostname":"Shelly-7","IPAddress":"10.xxx.xxx.xxx","Gateway":"10.xxx.xxx.xxx","Subnetmask":"255.255.255.0","DNSServer1":"10.xxx.xxx.xxx","DNSServer2":"10.xxx.xxx.xxx","Mac":"48:65:19:CE:F2:60","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
13:08:38.115 MQT: Shelly_7/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.11.12.160","MqttPort":1883,"MqttClientMask":"Solar","MqttClient":"Solar","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
13:08:38.138 MQT: Shelly_7/stat/STATUS7 = {"StatusTIM":{"UTC":"2022-08-27T12:08:38","Local":"2022-08-27T13:08:38","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:00","Sunset":"19:42"}}
13:08:38.157 MQT: Shelly_7/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
13:08:38.174 MQT: Shelly_7/stat/STATUS10 = {"StatusSNS":{"Time":"2022-08-27T13:08:38","Switch1":"OFF","ANALOG":{"Temperature":67.1},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.333,"Yesterday":0.000,"Today":0.333,"Power": 0},"TempUnit":"C"}}
13:08:38.195 MQT: Shelly_7/stat/STATUS11 = {"StatusSTS":{"Time":"2022-08-27T13:08:38","Uptime":"0T00:06:46","UptimeSec":406,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Sophos","BSSId":"00:1A:8C:A3:37:23","Channel":1,"Mode":"11n","RSSI":32,"Signal":-84,"LinkCount":1,"Downtime":"0T00:00:03"}}}
13:08:38.233 MQT: Shelly_7/stat/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["4027a35e","00000000","00000000"],"EXCVADDR":"0000000a","DEPC":"00000000","CallChain":["402625e1","40101206","4027a2e8","402624f8","40266c54","4027027b","4026fb3b","40000f49","40000f49","40000e19","40105aa9","40105aaf","4010000d","4026ea18","4026e9c9","40100a7c","40102b21","401026f0","401059a3","40100a7c","40103287","401026f0","40100a7c","40100a7c","4000050c","40106249","4000050c","40253d26","40253d29","401013fd","401013ec"]}}
[... cut ...]
I removed the Shelly and replaced it with another one. The problem remained at the measurement source. The replaced Shelly runs without problems on another measurement source.
Now I have read out this information and need some information on what it means in order to continue troubleshooting.
Info3": {"RestartReason": {"Exception": 0, "Reason": "Exception","EPC": [ "4027a35e","00000000","00000000"
Please explain what this error message signifies.
**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-27:
Looks like network connection retries to me considering the network code addresses and low connection info :``"RSSI":34,"Signal":-83``

EDSTOBI, 2022-08-27:
But that would not mean losing data and restarting the meters.
There are 6 Shelly sitting next to each other on the DIN rail and only this measuring source has the problem, all others run constantly.
Over 60 days of uptime until I updated them all to a new firmware version.
All Shelly are connected to the same phase and fuse.
Has RestartReason": {"Exception": 0, "Reason": "Exception","EPC" something to do with connection problems?

arendst, 2022-08-27:
> Has RestartReason": {"Exception": 0, "Reason": "Exception","EPC" something to do with connection problems?

It's the result of the problem. The exception occurs because strange things happen in the code. The numbers represent code addresses. Yours point to code servicing the network out of the hands of Tasmota and within the SDK.

For further debugging I suggest you start monitoring the heap (enable heap logging with command ``SO130 1``) and monitor the situation before an exception 0 restart occurs (I noticed you have mqttlog enabled. Set it to 3 and prepare for backtracing by storing the logs somewhere).
Also have a look at the other shellies network status (RSSI and Signal).
Finally having several device close to each other also will have a negative impact on wifi (connection).

EDSTOBI, 2022-08-27:
SO130 1 set, syslog activated again (data are also forgotten) and Shelly rebooted to better track uptime. 

0:00:00.001 HDW: ESP8266EX
00:00:00.049-031 CFG: Loaded from flash at FA, Count 34
00:00:00.055-031 QPC: Count 1
00:00:00.065-031 Project tasmota - Shelly_7 Version 12.1.1(tasmota)-2_7_4_9(2022-08-25T11:33:55)
00:00:00.555-032 WIF: Connecting to AP1 Sophos Channel 1 BSSId 00:1A:8C:A3:37:23 in mode 11n as Shelly-7...
00:00:01.754-031 WIF: Connected
00:00:02.005-029 HTP: Web server active on Shelly-7 with IP address 1xx
15:06:05.018-028 MQT: Attempting connection...
15:06:05.036-027 MQT: Connected
15:06:05.039-024 MQT: Shelly_7/tele/LWT = Online (retained)
15:06:05.041-024 MQT: Shelly_7/cmnd/POWER = 
15:06:05.050-026 MQT: Shelly_7/tele/INFO1 = {"Info1":{"Module":"SolarPumpe","Version":"12.1.1(tasmota)","FallbackTopic":"cmnd/Solar_fb/","GroupTopic":"Shelly_7/cmnd/"}}
15:06:05.058-024 MQT: Shelly_7/tele/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"Shelly-7","IPAddress":"1xx"}}
15:06:05.077-024 MQT: Shelly_7/tele/INFO3 = {"Info3":{"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["4027093f","00000000","00000000"],"EXCVADDR":"4026fabf","DEPC":"00000000","CallChain":["401026f0","40270c97","402701e9","4026fb3b","4024dad4","40000f49","40000f49","40000e19","40105aa9","40105aaf","4010000d","4026ea18","4026e9c9","40100a7c","40102b21","401026f0","4010598b","40104b66","40100a7c","40104fd7","401026f0","4010598b","40104a9f","401026f0","40103964","40104a9f","40100a7c","40102b21","401026f0","401059a3","40100a7c"]},"BootCount":19}}
15:06:05.120-025 MQT: Shelly_7/stat/RESULT = {"POWER":"OFF"}
15:06:05.124-025 MQT: Shelly_7/stat/POWER = OFF
15:06:08.459-027 QPC: Reset

STATUS = {"Status":{"Module":0,"DeviceName":"Shelly_7","FriendlyName":["Shelly_7"],"Topic":"Shelly_7","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:06:45","StartupUTC":"2022-08-27T12:01:52","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2022-08-15T10:44:05","SaveCount":33,"SaveAddress":"FB000"}}
STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:33:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"358/699"}}
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Sophos",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15400B","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
STATUS5 = {"StatusNET":{"Hostname":"Shelly-7","IPAddress":"1x","Gateway":"1x","Subnetmask":"255.255.255.0","DNSServer1":"1x","DNSServer2":"10x","Mac":"48:x:63","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
STATUS6 = {"StatusMQT":{"MqttHost":"1x","MqttPort":1883,"MqttClientMask":"Solar","MqttClient":"Solar","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
STATUS7 = {"StatusTIM":{"UTC":"2022-08-27T12:08:38","Local":"2022-08-27T13:08:38","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:00","Sunset":"19:42"}}
STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
STATUS10 = {"StatusSNS":{"Time":"2022-08-27T13:08:38","Switch1":"OFF","ANALOG":{"Temperature":67.1},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.333,"Yesterday":0.000,"Today":0.333,"Power": 0},"TempUnit":"C"}}
STATUS11 = {"StatusSTS":{"Time":"2022-08-27T13:08:38","Uptime":"0T00:06:46","UptimeSec":406,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Sophos","BSSId":"0x7:23","Channel":1,"Mode":"11n","RSSI":32,"Signal":-84,"LinkCount":1,"Downtime":"0T00:00:03"}}}
STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["4027a35e","00000000","00000000"],"EXCVADDR":"0000000a","DEPC":"00000000","CallChain":["402625e1","40101206","4027a2e8","402624f8","40266c54","4027027b","4026fb3b","40000f49","40000f49","40000e19","40105aa9","40105aaf","4010000d","4026ea18","4026e9c9","40100a7c","40102b21","401026f0","401059a3","40100a7c","40103287","401026f0","40100a7c","401…
c

EDSTOBI, 2022-08-27:
and from syslog
SSId":"00:1A:8C:A3:37:23","Channel":1,"Mode":"11n","RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:03"}}
Aug 27 16:04:08 Shelly-7 ESP-MQT: Shelly_7/tele/SENSOR = {"Time":"2022-08-27T15:04:07","Switch1":"OFF","ANALOG":{"Temperature":67.1},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.336,"Yesterday":0.000,"Today":0.336,"Period": 0,"Power": 0},"TempUnit":"C"}
Aug 27 16:04:08 Shelly-7 ESP-NRG: Energy reset by invalid data
Aug 27 16:04:08 Shelly-7 ESP-APP: Boot Count 19
Aug 27 16:04:09 Shelly-7 ESP-SRC: MQTT
Aug 27 16:04:09 Shelly-7 ESP-CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'OFF'
Aug 27 16:04:09 Shelly-7 ESP-MQT: Shelly_7/stat/RESULT = {"POWER":"OFF"}
Aug 27 16:04:09 Shelly-7 ESP-MQT: Shelly_7/stat/POWER = OFF
Aug 27 16:04:13 Shelly-7 ESP-MQT: tasmota/discovery/485519CBF263/config = {"ip":"10.11.12.213","dn":"Shelly_7","fn":["Shelly_7",null,null,null,null,null,null,null],"hn":"Shelly-7","mac":"485519CBF263","md":"SolarPumpe","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.1.1","t":"Shelly_7","ft":"Shelly_7/%prefix%","tp":["cmnd","stat","tele"],"rl":[1,0,0,0,0,0,0,0],"swc":[-1,-1,-1,-1,-1,-1,-1,-1],"swn":[null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":0,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1} (retained)
Aug 27 16:04:13 Shelly-7 ESP-MQT: tasmota/discovery/485519CBF263/sensors = {"sn":{"Time":"2022-08-27T15:04:12","Switch1":"OFF","ANALOG":{"Temperature":67.2},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.336,"Yesterday":0.000,"Today":0.336,"Power": 0},"TempUnit":"C"},"ver":1} (retained)
Aug 27 16:04:18 Shelly-7 ESP-MQT: Shelly_7/tele/STATE = {"Time":"2022-08-27T15:04:17","Uptime":"0T00:00:19","UptimeSec":19,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Sx","BSSId":"00:1x:23","Channel":1,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:03"}}
Aug 27 16:04:18 Shelly-7 ESP-MQT: Shelly_7/tele/SENSOR = {"Time":"2022-08-27T15:04:17","Switch1":"OFF","ANALOG":{"Temperature":67.4},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.336,"Yesterday":0.000,"Today":0.336,"Period": 0,"Power": 0},"TempUnit":"C"}
Aug 27 16:04:18 Shelly-7 ESP-SRC: MQTT
Aug 27 16:04:18 Shelly-7 ESP-CMD: Grp 0, Cmd 'POWER', Idx 1, Len 3, Pld 0, Data 'OFF'
Aug 27 16:04:18 Shelly-7 ESP-MQT: Shelly_7/stat/RESULT = {"POWER":"OFF"}
Aug 27 16:04:18 Shelly-7 ESP-MQT: Shelly_7/stat/POWER = OFF
Aug 27 16:04:21 Shelly-7 ESP-SRC: Switch
Aug 27 16:04:21 Shelly-7 ESP-MQT: Shelly_7/stat/RESULT = {"POWER":"ON"}
Aug 27 16:04:21 Shelly-7 ESP-MQT: Shelly_7/stat/POWER = ON
Aug 27 16:04:22 Shelly-7 ESP-WIF: Checking connection...
Aug 27 16:04:28 Shelly-7 ESP-MQT: Shelly_7/tele/STATE = {"Time":"2022-08-27T15:04:27","Uptime":"0T00:00:29","UptimeSec":29,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Sx","BSSId":"00:x7:23","Channel":1,"Mode":"11n","RSSI":30,"Signal":-85,"LinkCount":1,"Downtime":"0T00:00:03"}}
Aug 27 16:04:28 Shelly-7 ESP-MQT: Shelly_7/tele/SENSOR = {"Time":"2022-08-27T15:04:27","Switch1":"ON","ANALOG":{"Temperature":67.4},"ENERGY":{"TotalStartTime":"2022-08-15T10:44:05","Total":0.336,"Yesterday":0.000,"Today":0.336,"Period": 0,"Power":32},"TempUnit":"C"}

EDSTOBI, 2022-08-29:
Update:
Shelly plus 1 PM 24h without a single error ore reboot.
Flash with tasmota32solo1

closed 2022-09-28 by github-actions[bot]