arendst/Tasmota #16354: Please clarify where TLS settings must be added within the overrides file when compiling for ESP8266

CodingSpiderFox, 2022-08-27:
### PROBLEM DESCRIPTION
I don't know where to add the required lines when I want to complie with TLS support for Sonoff S20

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff S20
- [ ] Tasmota binary firmware version number used: 12.1.1
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
The documentation says "https://tasmota.github.io/docs/TLS/#compiling-tls-for-esp8266"

"To use it you must [compile your build](https://tasmota.github.io/docs/Compile-your-build/). Add the following to user_config_override.h:"

```
#ifndef USE_MQTT_TLS 
#define USE_MQTT_TLS                             // Use TLS for MQTT connection (+34.5k code, +7.0k mem and +4.8k additional during connection handshake)
#define MQTT_TLS_ENABLED       true              // [SetOption103] Enable TLS mode (requires TLS version)
//  #define USE_MQTT_TLS_CA_CERT                   // Force full CA validation instead of fingerprints, slower, but simpler to use.  (+2.2k code, +1.9k mem during connection handshake)
                                                   // This includes the LetsEncrypt CA in tasmota_ca.ino for verifying server certificates
//  #define USE_MQTT_TLS_FORCE_EC_CIPHER           // Force Elliptic Curve cipher (higher security) required by some servers (automatically enabled with USE_MQTT_AWS_IOT) (+11.4k code, +0.4k mem)
#endif
```
**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-08-27:
It still writen clearly in the text you copied
The blue text are links to the documention which is a little more than optional to read when you want to compile your own binary.

Remember that Issues are only to report qualified bugs. ASking for help should be done in Discussions or on Discord server

CodingSpiderFox, 2022-08-27:
The text does not say _where_ within that file :(

barbudor, 2022-08-27:
Considering that file is by default EMPTY that should be quite easy to figure out.
The doc and the file itself are very clear for anyone expecting to compile C code.
If you don't understand it, you are probably not the audience for compiling with VisualCode.

TasmoCompiler is probably be more suited for you.