arendst/Tasmota #16361: Sonoff Dual R3 with Tasmota Switch SW1 is not working

Mario-1977, 2022-08-27:
PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I flashed 20 Sonoff Dual R3 using Tasmota32-bluetooth 12.1.1.1 and 4 of them are not working properly using the switch inputs.
SW2 is working but SW1 is not making any action. On the other hand the relays 1 and 2 work properly when inputs came via WiFi.
I don't know if the problem was present using the original firmware because i didn't tested them before flashing. 
I decided to ask support because 2% of 20 devices are defective and always on SW1. It seems to me too strange.
I tried also an older Tasmota32 release (8.5.0) but I didn't see any improvement using the hardware inputs.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Sonoff Dual R3
- [x ] Tasmota binary firmware version number used: Tasmota32-bluetooth 12.1.1.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [x ] Flashing tools used: Esp-flasher Windows-x64
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`: used template {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
```lua
  Configuration output here:
<html>
<body>
<!--StartFragment-->

Program Version | 12.1.0.1(bluetooth)
-- | --
2022-08-24T15:11:13
2_0_4_1/v4.4.3
0T00:30:10
28
5
Vbat power on reset
Persiana_cameretta_pt
 
 
AndroidAP9830 (82%, -59 dBm) 11n
Persiana-cameretta-pt-2156
08:3A:F2:24:E8:6C
192.168.191.172
192.168.191.176
255.255.255.0
192.168.191.176
0.0.0.0
 
Enabled
 
192.168.43.42
1883
Disabled
Persiana_cameretta_pt
DVES_24E86C
Persiana_cameretta_pt
cmnd/tasmotas/
cmnd/Persiana_cameretta_pt/
cmnd/DVES_24E86C_fb/
Disabled
 
None
 
2418796 (ESP32-D0WD-V3 rev.3)
0x16405E (DIO)
4096 KB
4096 KB
1538 KB
1856 KB
103.4 KB (frag. 13%)
 
1856 KB (used 82%)
1856 KB
320 KB

<!--EndFragment-->
</body>
</html>

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:40:56.951 CMD: Status 0
20:40:56.959 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Persiana_cameretta_pt","FriendlyName":["Persiana_cameretta_pt",""],"Topic":"Persiana_cameretta_pt","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:40:56.998 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:32:39","StartupUTC":"2022-08-27T19:08:17","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2022-08-26T19:21:32","SaveCount":28}}
20:40:57.039 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.0.1(bluetooth)","BuildDateTime":"2022-08-24T15:11:13","Core":"2_0_4_1","SDK":"v4.4.3","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"466/699"}}
20:40:57.059 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["AndroidAP9830","AndroidCristina"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
20:40:57.095 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1538,"Free":1856,"Heap":102,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["00000809","8F9AC7C7","00148001","000000CF","010013C0","C0000981","00004080","00200000","5400082C","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,12,16,20,21,24,26,27,29,35,38,50,52,59,62,79,85","Sensors":"1,2,3,5,6,52,62,127"}}
20:40:57.134 RSL: STATUS5 = {"StatusNET":{"Hostname":"Persiana-cameretta-pt-2156","IPAddress":"192.168.191.172","Gateway":"192.168.191.176","Subnetmask":"255.255.255.0","DNSServer1":"192.168.191.176","DNSServer2":"0.0.0.0","Mac":"08:3A:F2:24:E8:6C","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:40:57.166 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.43.42","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_24E86C","MqttUser":"Persiana_cameretta_pt","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:40:57.198 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-08-27T19:40:57","Local":"2022-08-27T20:40:57","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:00","Sunset":"19:42"}}
20:40:57.216 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
20:40:57.241 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-08-27T20:40:57","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-08-26T19:21:32","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Frequency": 0,"Voltage": 0,"Current":[0.000,0.000]},"ESP32":{"Temperature":53.3},"TempUnit":"C"}}
20:40:57.284 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-08-27T20:40:57","Uptime":"0T00:32:40","UptimeSec":1960,"Heap":104,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"AndroidAP9830","BSSId":"76:C3:61:3E:5E:1B","Channel":1,"Mode":"11n","RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I connect the Sonoff Dual R3 to main power (neutral and line) then I connect the L1 phase with SW1

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expect the relay action but nothing happen

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2022-09-27 by github-actions[bot]

petermicuch, 2023-01-31:
Well this sound very familiar @Mario-1977. I bought 10 pieces of Sonoff Dual R3 and on one of them I observed exactly the same behavior as you described here (SW2 working fine, SW1 not). Have you been able to discover the root cause?

Mitch1675, 2023-01-31:
I have the same problem

Mario-1977, 2023-02-01:
I never discovered the root cause. No one help me to understand and solve. Actually I have 5 Sonoff R3 with the same issue of SW1 not working.
I'm convinced that is related to some firmware issue su I'm waiting for an update to test it

petermicuch, 2023-02-02:
@Mario-1977 - I would definitely like to try with original firmware, but I was not clever enough to back it up before flashing with tasmota. Getting sonoffs firmware for download somewhere seems to be impossible. Would you mind checking difference of CPUs and other components on the board between working and not working ones? I will do the same. And perhaps could you reopen the issue again? It is just strange that multiple people have issues always with switch #1. In my case failrate is 10% in your 20%, which is huge number for Chinese production.

BTW - I like shelly switches much more. They do fit into European wall switches, sonoffs are impossible to fit. I guess in USA wall switches are much bigger than here.

Mitch1675, 2023-02-02:
I tested before flashing. I used eWelink to do the orders as well as with sw1 and sw2 and everything worked. After flashing with Tasmota the sw1 does not work.

arendst, 2023-02-02:
Just to verify my working dual r3 uses this template

```{"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}```

and switchmodes ```"SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]```

petermicuch, 2023-02-02:
@arendst 
I do have newer version of sonoff dual r3, bought just recently:
`
{"NAME":"Sonoff Dual R3 v2","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
`
But as said, it works on 9 switches out of 10 with no issue. I will try to check if I see any HW difference between those modules later today and will post back here.

petermicuch, 2023-02-02:
@Mitch1675 - I totally overlook your comment and it is very valuable that you have tested input before flashing - at least someone is not so in hurry :-)

Could you perhaps post the details of you HW + templates that you used for tasmota?

petermicuch, 2023-02-02:
@arendst - I observed working and not working sonoff modules. From the HW point of view, they look the same, same bluetooth chip, same CPU, all part that I was able to read with magnifier glass look the same. Same print out on the board.

Is there anything I could run to gather more data for analysis? Any commands that would help to analyze the problem?

arendst, 2023-02-02:
Nope. As it's a haredware issue just get out your multimeter and measure the signal path from the SW connector to the esp32.

petermicuch, 2023-02-02:
And is it HW issue even if @Mitch1675 confirmed that eWeLink worked fine?

arendst, 2023-02-02:
Yes. If some devices use different groups we don't know. Therefore it's your task to measure the hardware difference.

arendst, 2023-02-02:
Or play with the template and select different type switch for switch1. Be creative. There are switches without pull-ups.

petermicuch, 2023-02-03:
Thank you @arendst. I will try to play with it a bit, will be my first time trying to trace something like that. Seem I have to read more about templates in tasmota.

Hoedi1030, 2024-07-04:
My Sonoff DualR3 is suffering from the same problem. Has anyone discovered the cause of Switch1 not responding?

Since I have a second unit which still has the original Sonoff-firmware I am wondering if the original firmware can be backed up and flashed to the "faulty" device?

petermicuch, 2024-07-04:
@Hoedi1030 I have not been able to tackle this problem. I even wrote to Sonoff, if they could provide me with the original firmware binary to flash it back. I wnated to check if the switch is working before flashing Tasmota. They replied, that they do not provide firmware binaries 👎.

But since you still have one device without update, could you perhaps share the binary? Try to follow steps [here](https://hobbytronics.pk/sonoff-original-firmware-backup-restore/) to backup original firmware with esptool.

Hoedi1030, 2024-07-05:
Sure. I will share it as soon as I have made the backup. I will do so probably this weekend.
However, I do not know if flashing the firmware from another device would work. I read somewhere, that the firmware (at least for certain Sonoff devices) is tied to the MAC Address, so we'll see...

If someone else could provide an additional backup we would be able to do a file comparison of the binaries and rule out this uncertainty...