arendst/Tasmota #16364: Tasmota 12.1.1 unable to flash EFM8BB1 on Sonoff RfBridge

meyergru, 2022-08-28:
### PROBLEM DESCRIPTION
I tried to flash the RF chip on a Sonoff RF Bridge R2 V1.0 with two jumper wires attached. This always ends up in an error 211, regardless of switch position and jumpers crossing or not.

I knew for sure that this device once worked, so I downgraded to 12.0.2 and voila - flashing works again!

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff RF Bridge R2 V1.0
- [x] Tasmota binary firmware version number used: 12.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Upgrade from older version via web UI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Flash RF firmware via web UI -> error 211 

Afterwards, the RF units seems to permamently send a signal, interfering with operation of RF sender.

### EXPECTED BEHAVIOUR
successful flash

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-29:
Testing.....

arendst, 2022-08-29:
Verified to fail.

My personal v12.1.1.1 having a smaller code base works fine so I expect an out-of-memory error.

Investigating....

arendst, 2022-08-29:
It's not a memory issue. The error code 211 points to an erase issue of the EFM8BB1. I'm unable to receive better errorcodes as when I try to add debug code it works as expected....

WORKAROUND:
For now I suggest you use v12.0.2 or v12.1.1.1 (latest development) for flashing the EFM8BB1 while v12.1.1 is perfectly fine for normal use.

meyergru, 2022-08-29:
I know, that is what I did. Maybe flashing is not really a thing any more, because you cannot flash Portisch firmware anymore on the now-available R2 V2 models. But at least there should be a note in the Wiki as to using an older or a minimal version to flash.

The process itself is rather error-prone using floating wiring and such. I was just before assuming I had destroyed my device before trying to flash it under 12.0.2 and reviving it. I at least did not expect a memory shortage, but luckily I tried because my second device had been flashed successfully under 12.0.2.

arendst, 2022-08-29:
I didn't flash for years so I thought I had to add the wires too. After opening the box it appeared I soldered fixed wires and made the cuts to seperate from USB as documented in the docs.

So flashing was easy for me (except on v12.1.1 :-)) in the end.

meyergru, 2022-08-29:
I had the device open anyway because I just installed Tasmota, it was on Espurna before. But now I switched to Tasmota / Home Assistant.

closed 2022-09-28 by github-actions[bot]