arendst/Tasmota #16365: Sonoff Dual R3 no Voltage or power shown (Tasmota 12.1.1)

jheisel01, 2022-08-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The device does not show Voltage or Power values if an output is powered on. -> Works with version 12.0.2.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Dual R3
- [X ] Tasmota binary firmware version number used: 12.1.1
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: esp-flasher
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:09:50.462 CMD: Backlog Template; Module; GPIO 255
11:09:50.518 MQT: sonoff/sonoff_009/stat/RESULT = {"NAME":"Sonoff Dual R3","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,7296,7328,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
11:09:50.753 MQT: sonoff/sonoff_009/stat/RESULT = {"Module":{"0":"Sonoff Dual R3"}}
11:09:51.015 MQT: sonoff/sonoff_009/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"7296":"CSE7761 Tx"},"GPIO26":{"7328":"CSE7761 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
11:22:41.454 CMD: Backlog Rule1; Rule2; Rule3
11:22:41.507 MQT: sonoff/sonoff_009/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:22:41.746 MQT: sonoff/sonoff_009/stat/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
11:22:41.992 MQT: sonoff/sonoff_009/stat/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:23:19.348 CMD: status 0:
11:23:19.368 MQT: sonoff/sonoff_009/stat/STATUS = {"Status":{"Module":0,"DeviceName":"Sonoff_009","FriendlyName":["Sonoff_009","Sonoff_009"],"Topic":"sonoff_009","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
11:23:19.381 MQT: sonoff/sonoff_009/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:24:25","StartupUTC":"2022-08-28T09:58:54","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2022-08-27T16:25:05","SaveCount":61}}
11:23:19.390 MQT: sonoff/sonoff_009/stat/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:34:17","Core":"2_0_4_1","SDK":"v4.4.3","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"420/699"}}
11:23:19.400 MQT: sonoff/sonoff_009/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["mywlR2400",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
11:23:19.429 MQT: sonoff/sonoff_009/stat/STATUS4 = {"StatusMEM":{"ProgramSize":1414,"Free":1856,"Heap":141,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
11:23:19.443 MQT: sonoff/sonoff_009/stat/STATUS5 = {"StatusNET":{"Hostname":"Sonoff-D-009","IPAddress":"192.168.168.43","Gateway":"192.168.168.112","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"94:3C:C6:C6:59:0C","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:23:19.459 MQT: sonoff/sonoff_009/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.168.12","MqttPort":1883,"MqttClientMask":"Sonoff_009","MqttClient":"Sonoff_009","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:23:19.471 MQT: sonoff/sonoff_009/stat/STATUS7 = {"StatusTIM":{"UTC":"2022-08-28T10:23:19","Local":"2022-08-28T11:23:19","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:01","Sunset":"19:40"}}
11:23:19.489 MQT: sonoff/sonoff_009/stat/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:23:19.514 MQT: sonoff/sonoff_009/stat/STATUS10 = {"StatusSNS":{"Time":"2022-08-28T11:23:19","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2022-08-27T16:25:05","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000]},"ESP32":{"Temperature":41.7},"TempUnit":"C"}}
11:23:19.531 MQT: sonoff/sonoff_009/stat/STATUS11 = {"StatusSTS":{"Time":"2022-08-28T11:23:19","Uptime":"0T00:24:25","UptimeSec":1465,"Heap":140,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":40},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"mywlR2400","BSSId":"EA:C3:2A:9C:04:3C","Channel":1,"Mode":"11n","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
11:26:08.752 MQT: sonoff/sonoff_009/stat/RESULT = {"SerialLog":{"2":{"Active":"2"}}}
11:26:08.771 MQT: sonoff/sonoff_009/stat/RESULT = {"WebLog":4}
11:26:08.783 SRC: Backlog
11:26:08.787 CMD: Grp 0, Cmd 'MQTTLOG', Idx 1, Len 1, Pld 0, Data '0'
11:26:08.793 MQT: sonoff/sonoff_009/stat/RESULT = {"MqttLog":0}
11:26:08.808 SRC: Backlog
11:26:08.813 CMD: Grp 0, Cmd 'SYSLOG', Idx 1, Len 1, Pld 0, Data '0'
11:26:08.826 MQT: sonoff/sonoff_009/stat/RESULT = {"SysLog":{"0":{"Active":"0"}}}
11:26:08.839 SRC: Backlog
11:26:08.842 CMD: Grp 0, Cmd 'LOGHOST', Idx 1, Len 1, Pld 1, Data '1'
11:26:08.845 CFG: CR 420/699, Busy 0
11:26:08.854 MQT: sonoff/sonoff_009/stat/RESULT = {"LogHost":""}
11:26:08.869 SRC: Backlog
11:26:08.874 CMD: Grp 0, Cmd 'LOGPORT', Idx 1, Len 3, Pld 514, Data '514'
[... 19 lines cut ...]
11:27:00.617 BRY: GC from 7259 to 3798 bytes, objects freed 13/40 (in 1 ms) - slots from 44/122 to 41/122

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Toggle output ON

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Show Voltage and power values as in version 12.0.2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
 added two screenshots 
Sonoff_Dual_R3--12-1-1.PNG = Sonoff Dual R3 with Tasmota 12.1.1
Sonoff_Dual_R3--12-0-2.PNG = Sonoff Dual R3 with Tasmota 12.0.2

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-08-28:
Cannot reproduce 

[image]

MatejsLab, 2022-08-28:
Actually, I have reproduced that issue on Athom Plug V2, I upgraded to 12.1.1 and suddenly I noticed that the MQTT SENSOR topic is not consistent with spaces. 

For Voltage, Current, Power, and some more, they have spaces between colon and number. This breaks integration into Openhab as parsing strings into numbers breaks with spaces.

The issue is that the MQTT topic is not consistent over time also. When I freshly loaded the update it was fine, with no spaces in the topic between colons and numbers, but then I turned the relay on, and suddenly SENSOR Topic Changed, restarting the device doesn't revert it.

This is that broken MQTT Sensor Topic, **where bolded items have intrusive spaces**
{"Time":"2022-08-28T13:59:05","ENERGY":{"TotalStartTime":"2022-08-20T11:16:21","Total":0.002,"Yesterday":0.000,"Today":0.002,**"Period": 0,"Power": 0,"ApparentPower": 0,"ReactivePower": 0,"Factor":0.00,**"Voltage": 0,**"Current":0.000}}

I downgraded to 9.5.0, thaat I know is working fine for this, as version 11.0.0 had the same issue.

jheisel01, 2022-08-28:
Hi,
if you look to my screenshot from Sonoff_009, there is also no line with Frequency.

MatejsLab, 2022-08-28:
Yeah my plug also has no frequency shown in Tasmota GUI, I am not sure if it is even measured.
I have robaxo plug https://templates.blakadder.com/robaxo_RSP-025.html , running still 9.5.0 and also has no frequency readout, if not shown in the GUI that may be hardware limitation.

But that MQTT Sensors spacing issue is for sure a bug in software.

arendst, 2022-08-28:
Although JSON allows spaces at those location I agree it's an anomaly likely caused by our use of printing floats.

I'll have a look.

jheisel01, 2022-08-28:
In the meantime I have made a downgrade to version 12.0.2. The error pattern is the same. No power, volt and frequency values. Possibly a change in the Sonoff itself ?

MatejsLab, 2022-08-28:
Oh, thanks that `wattres 1`  workaround did resolve that spacing issue.

Sorry for my misconception, at first glance, it seemed like a similar issue.

Thanks a lot, for the super quick response and a neat workaround.

sfromis, 2022-08-28:
While getting rid of the extra spaces is nice, it is perfectly valid JSON to have them, and any parsing failing due to them is in need of correction.

arendst, 2022-08-28:
@jheisel01 if you (like I) suspect a hardware change pls open both dualR3's and try to find differences (or a solder error) and let us know.

arendst, 2022-08-28:
Just looked at the Sonoff Dual R3 website and noticed this:

[image]

You may have bought the lite version.....

jheisel01, 2022-08-28:
Hi,
This is the item description of the ordered item:
Smart Wlan Schalter Garagentor SONOFF DUAL R3, 2Wege 2Gang Schalter, mit Messfunktion, 3 Arbeitsmodi, Fernbedienung Licht, Haushaltsgeräte, Fensterläden, Rollläden ; Alexa/Google Home/Siri Supported
also on the package DualR3 with power monitoring

arendst, 2022-08-28:
In that case open the device and check for power monitoring chip and/or bad soldering.

jheisel01, 2022-08-28:
Ok,
will open both devices and check. But it will be tomorrow. Here a picture from the case (Sonoff009
[image]
)

arendst, 2022-08-28:
Oh have a look at this: https://templates.blakadder.com/sonoff_DUALR3_v2.html

sfromis, 2022-08-28:
Also, the lite version should be clearly indicated on the plastic cover. 
[image]

jheisel01, 2022-08-28:
After applying the template from blakadder it shows power & Volt values but no Frequency (dont need it)
{"NAME":"Sonoff Dual R3 (ZCD)","GPIO":[32,0,0,0,7552,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
[image]

Thanks for your help and support !

regards,
Jürgen

closed 2022-08-28 by jheisel01

sfromis, 2022-08-28:
Not all energy monitoring chips capture the frequency, and the v2 version has a different chip. My v1 DualR3 devices do provide the frequency.

sfromis, 2022-08-28:
You might want to consider `SetOption129` for split total energy results