arendst/Tasmota #16367: Upgrade from 12.1.0 > 12.1.1 Warns about minimal upgrade

aaronnad, 2022-08-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When upgrading from 12.1.0 > 12.1.1 there is a warning about using a minimal version install on the device. The device then restarts after 10-15 seconds of being unresponsive and comes back as normal.

This is using the default OTA upgrade URL in tasmota.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:10:43.057 MQT: stat/sw_hallway_pc_plug/RESULT = {"NAME":"Avatar UK 10A","GPIO":[0,0,320,0,0,2720,0,0,2624,32,2656,224,0,0],"FLAG":0,"BASE":45}
12:10:43.271 MQT: stat/sw_hallway_pc_plug/RESULT = {"Module":{"0":"Avatar UK 10A"}}
12:10:43.527 MQT: stat/sw_hallway_pc_plug/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:11:14.367 MQT: stat/sw_hallway_pc_plug/STATUS = {"Status":{"Module":0,"DeviceName":"Hallway PC Plug","FriendlyName":["Hallway PC Plug"],"Topic":"sw_hallway_pc_plug","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:11:14.402 MQT: stat/sw_hallway_pc_plug/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:57","StartupUTC":"2022-08-28T11:01:17","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2022-08-23T20:28:56","SaveCount":38,"SaveAddress":"FA000"}}
12:11:14.438 MQT: stat/sw_hallway_pc_plug/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:33:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"396/699"}}
12:11:14.457 MQT: stat/sw_hallway_pc_plug/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT-WiFi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
12:11:14.491 MQT: stat/sw_hallway_pc_plug/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144068","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
12:11:14.526 MQT: stat/sw_hallway_pc_plug/STATUS5 = {"StatusNET":{"Hostname":"sw-hallway-pc-plug-0587","IPAddress":"10.0.50.106","Gateway":"10.0.50.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.45","DNSServer2":"192.168.1.46","Mac":"F4:CF:A2:4C:82:4B","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:11:14.554 MQT: stat/sw_hallway_pc_plug/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.139","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4C824B","MqttUser":"mqtt_user","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:11:14.578 MQT: stat/sw_hallway_pc_plug/STATUS7 = {"StatusTIM":{"UTC":"2022-08-28T11:11:14","Local":"2022-08-28T12:11:14","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:01","Sunset":"19:40"}}
12:11:14.596 MQT: stat/sw_hallway_pc_plug/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:11:14.617 MQT: stat/sw_hallway_pc_plug/STATUS10 = {"StatusSNS":{"Time":"2022-08-28T12:11:14","ENERGY":{"TotalStartTime":"2022-08-23T20:28:56","Total":0.923,"Yesterday":0.546,"Today":0.042,"Power": 3,"ApparentPower":43,"ReactivePower":43,"Factor":0.08,"Voltage":241,"Current":0.177}}}
12:11:14.641 MQT: stat/sw_hallway_pc_plug/STATUS11 = {"StatusSTS":{"Time":"2022-08-28T12:11:14","Uptime":"0T00:09:57","UptimeSec":597,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"IoT-WiFi","BSSId":"A2:DA:C4:C2:00:CE","Channel":1,"Mode":"11n","RSSI":12,"Signal":-94,"LinkCount":1,"Downtime":"0T00:00:17"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Unable to capture issue here with log due to it occuring during an upgrade and web server restarting

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Upgrade from 12.1.0 to 12.1.1 using the default OTA url

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A warning that i am using Minimal install to not appear

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]
[image]
[image]
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-08-28:
You are just describing tasmota standard upgrade procedure using the intermediate temporary tasmota-minimal
No issue here

aaronnad, 2022-08-28:
Intersting, i've not experienced this before on any other upgrade from previous versions, this is the first version to show this issue for me

barbudor, 2022-08-28:
Most of time the duration of the minimal step is short and your browser doesn't have the opportunity to catch it

aaronnad, 2022-08-28:
Ahh ok, must just be a bit longer this time round then. No worries, shall close this issue down

closed 2022-08-28 by aaronnad