arendst/Tasmota #16368: Serial buffer overrun when reading P1

andrasg, 2022-08-28:
### PROBLEM DESCRIPTION

I am using a Wemos D1 mini to read the P1 interface of my electricity meter. Data is arriving nicely, but is getting truncated. Tried increasing the buffer using `SerialBuffer 520`, but still get buffer overruns in each read cycle:

Log excerpt:

```lua
12:33:16.855 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:8.7.0(00.113*kvar)\r"}
12:33:16.918 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"0-0:98.1.0(220801000000S)(011181.085*kWh)(005916.460*kWh)(005264.625*kWh)(000000.000*kWh)(000000.000*kWh)(000000.000*kWh)(001842.123*kvarh)(002155.507*kvarh)(001842.123*kvarh)(000000.000*kvarh)(000000.000*kvarh)(002155.507*kvarh)(011181.087*kWh)(03.332*kW)(03.332*kW)(03.156*kW)(00.000*kW)(00.000*kW)(00.000*kW)\r"}
12:33:17.269 CMD: Serial buffer overrun
12:33:17.274 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":")\r"}
12:33:17.284 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"!46EC\r"}
12:33:26.482 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"/AUX59902711681\r"}
12:33:26.488 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"\r"}
12:33:26.503 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"0-0:1.0.0(220828133330S)\r"}
12:33:26.523 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"0-0:42.0.0(AUX1030302711681)\r"}
12:33:26.538 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"0-0:96.1.0(9902711681)\r"}
12:33:26.548 CMD: Serial buffer overrun
12:33:26.554 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":".0(.5017.0.)\r"}
12:33:26.563 CMD: Serial buffer overrun
12:33:26.567 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"0-0:1.8.4(000*kWh)\r"}
12:33:26.575 CMD: Serial buffer overrun
12:33:26.580 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":".kWh)\r"}
12:33:26.590 CMD: Serial buffer overrun
12:33:26.595 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"k0(000000.00rh)\r"}
12:33:26.600 CMD: Serial buffer overrun
12:33:26.605 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"\r"}
12:33:26.618 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:8.8.0(002230.636*kvarh)\r"}
12:33:26.633 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:15.8.0(011507.837*kWh)\r"}
12:33:26.647 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:32.7.0(241.9*V)\r"}
12:33:26.661 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:52.7.0(236.2*V)\r"}
12:33:26.674 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:72.7.0(241.7*V)\r"}
12:33:26.687 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:31.7.0(000*A)\r"}
12:33:26.699 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:51.7.0(002*A)\r"}
12:33:26.713 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:71.7.0(000*A)\r"}
12:33:26.726 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:13.7.0(0.990)\r"}
12:33:26.739 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:33.7.0(0.912)\r"}
12:33:26.753 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:53.7.0(0.978)\r"}
12:33:26.766 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:73.7.0(0.840)\r"}
12:33:26.779 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:14.7.0(50.02*Hz)\r"}
12:33:26.792 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:1.7.0(00.599*kW)\r"}
12:33:26.806 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:2.7.0(00.000*kW)\r"}
12:33:26.820 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:5.7.0(00.031*kvar)\r"}
12:33:26.833 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:6.7.0(00.000*kvar)\r"}
12:33:26.848 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:7.7.0(00.000*kvar)\r"}
12:33:26.862 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:8.7.0(00.113*kvar)\r"}
12:33:27.309 CMD: Serial buffer overrun
12:33:27.314 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":")\r"}
12:33:27.323 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"!AE83\r"}
```

What can I do to avoid the buffer overrun and read each P1 message properly?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wmos D1 mini
- [X] Tasmota binary firmware version number used: Tasmota 12.1.1
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: browser
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
12:37:44.130 CMD: Backlog Template; Module; GPIO 255
12:37:44.162 MQT: tasmota-szeher/stat/wemos-p1/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
[... 36 lines cut ...]
12:40:06.660 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:52.7.0(237.0*V)\r"}
12:40:06.664 RUL: No valid JSON ({"SERIALRECEIVED":"1-0:52.7.0(237.0*V)\R"})
12:40:06.675 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:72.7.0(242.2*V)\r"}
12:40:06.680 RUL: No valid JSON ({"SERIALRECEIVED":"1-0:72.7.0(242.2*V)\R"})
12:40:06.690 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:31.7.0(000*A)\r"}
12:40:06.695 RUL: No valid JSON ({"SERIALRECEIVED":"1-0:31.7.0(000*A)\R"})
12:40:06.706 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:51.7.0(002*A)\r"}
12:40:06.710 RUL: No valid JSON ({"SERIALRECEIVED":"1-0:51.7.0(002*A)\R"})
12:40:06.721 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:71.7.0(000*A)\r"}
12:40:06.725 RUL: No valid JSON ({"SERIALRECEIVED":"1-0:71.7.0(000*A)\R"})
12:40:06.736 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":"1-0:13.7.0(0.989)\r"}
12:40:06.741 RUL: No valid JSON ({"SERIALRECEIVED":"1-0:13.7.0(0.989)\R"})
12:40:06.747 CMD: Serial buffer overrun
12:40:06.751 MQT: tasmota-szeher/tele/wemos-p1/RESULT = {"SerialReceived":".7.0(0.907)\r"}
12:40:06.756 RUL: No valid JSON ({"SERIALRECEIVED":{"DATA":".7.0(0.907)\R"}})
```

### TO REPRODUCE
Followed guide to create P1 interface: https://tasmota.github.io/docs/P1-Smart-Meter/
Installed Tasmota, configured MQTT, set `SerialDelimiter 10` and `SerialSend 1`.
Connected to meter's P1 port.
Observed data being sent from meter.

### EXPECTED BEHAVIOUR
No buffer overrun is experienced and all messages are sent properly, each send cycle.

barbudor, 2022-08-28:
What is the baudrate used ?

andrasg, 2022-08-28:
> What is the baudrate used ?

The meter, according to its docs uses 115200 baud, 8N1 config.

barbudor, 2022-08-28:
That may be a bit high. Do you know the quantity of data transmitted every second ? (roughly)

andrasg, 2022-08-28:
> Do you know the quantity of data transmitted every second ? (roughly)

Data being sent is unfortunately controlled by the electricity company. Docs say the meter (Sanxing SX631) emits 42 data points, every 10 seconds. Each datapoint is around 20 characters on average, but there are one or two datapoints (eg: monthly summary data) that are significantly longer (200+ chars).

From the log timestamps, I see that one cycles worth of data is arriving in about 70ms.

barbudor, 2022-08-28:
So every 10 seconds you get 40 * 20 + 2 * 250 = 1300
I think this is the problem
If it comes as a burst and you only have a buffer of 520, Tasmota cannot cope with that

I've simulated that (1300 bytes as a burst every 10 seconds) and I have to reduce the Baudrate to 19200 to avoid overrun

Any way you can change the baudrate of the meter ?
1300 Bytes every 10 seconds is around an effective 1300 bps (130 Bps). Too bad then send that as a high speed burst

andrasg, 2022-08-28:
> So every 10 seconds you get 40 * 20 + 2 * 250 = 1300 I think this is the problem If it comes as a burst and you only have a buffer of 520, Tasmota cannot cope with that
>
> I've simulated that (1300 [...]

Is this an ESP8266 limit or Tasmota?

andrasg, 2022-08-28:
I cannot influence the sender, but have an RTS lane. Perhaps Tasmota could clear the RTS while the buffer is full?

barbudor, 2022-08-28:
> Is this an ESP8266 limit or Tasmota?

Hum, Tasmota as far as I can see. It should be possible to allow larger size. I'm looking into it

> I cannot influence the sender, but it does offer an RTS lane. Perhaps Tasmota could clear the RTS while the buffer is full?

Unfortunately Tasmota do not support hardware handshake (software neither I'm afraid)

barbudor, 2022-08-28:
Can you try this version with `Serialbuffer 1400` or `Serialbuffer 1500`
It seems to work with my simulation.
If you confirm that it is working stable, I will push a PR but serial management on Tasmota is a bit complex, not yet 100% sure that it is clean and acceptable

[tasmota-serialbuff2048.zip](https://github.com/arendst/Tasmota/files/9439875/tasmota-serialbuff2048.zip)

sfromis, 2022-08-28:
Depending on what you want to do with the data, I can also recommend ESP32, not only because it has more headroom for speed, but also due to much better local processing options with Berry, where you are not getting the overhead of the logging of the incoming serial data.

andrasg, 2022-08-28:
> Can you try this version with `Serialbuffer 1400` or `Serialbuffer 1500` It seems to work with my simulation.

It works well! I had to go to 1700, even though the actual characters of the message yielded only 1220 bytes.

Thanks for the help, I will keep monitoring it and give a shout if I encounter something unexpected.

andrasg, 2022-08-28:
Ah, thinking about it more, it's probably the CRC that's on top of the payload making it slightly more than the 1220 bytes.

barbudor, 2022-08-28:
In SerialBridge mode, Tasmota won't process any CRC
I don't see anything that could be a CRC in a typical line such as `{"SerialReceived":"1-0:72.7.0(241.7*V)\r"}`
May be the last line `{"SerialReceived":"!AE83\r"}` could be such CRC for the whole frame but that's just a matter of a few bytes. 
Nevertheless, I took some margin as it allows up to 2048 bytes.

closed 2022-08-28 by arendst

deesnow, 2022-10-03:
> It works well! I had to go to 1700, even though the actual characters of the message yielded only 1220 bytes.
>
> Thanks for the help, I will keep monitoring it and give a shout if I encounter [...]

@andrasg I had the same issue, but with this modified version all messages are sent to mttq. Unfortunatelly here on github I cannot send you direct message and I would like to ask some question regardind the parsig of the mttq message on HomaAssistant side. Could you send me your mail address? You can find my address on profile page.

deesnow, 2022-10-04:
> Can you try this version with `Serialbuffer 1400` or `Serialbuffer 1500` It seems to work with my simulation. If you confirm that it is working stable, I will push a PR but serial management on [...]

Hi @barbudor 

In this version did you modify anything else or just the followings?
```
#define SML_MAX_VARS 60
#define SML_BSIZ 1060
#define MAX_METERS 1
#define TMSBSIZ 2048
#define USE_SML_SCRIPT_CMD
#define SML_REPLACE_VARS
```

barbudor, 2022-10-04:
Do not use this version which was temporary
Changes made are now in Tasmota supporting 'SerialBuffer' command up to 2048
Note that if you are using scripting I have no idea if this improvement also applies to scripting

andrasg, 2022-10-04:
> @andrasg I had the same issue, but with this modified version all messages are sent to mttq. Unfortunatelly here on github I cannot send you direct message and I would like to ask some question [...]

Hey János, I am not using HomeAssistant, cannot help you there, unfortunately.

deesnow, 2022-10-04:
> Hey János, I am not using HomeAssistant, cannot help you there, unfortunately.

Thx anyway ;)