arendst/Tasmota #16390: berry json module load HEX error

darcyg, 2022-08-31:
### PROBLEM DESCRIPTION
berry json module load HEX  error
There is no error in cmnd parsing. berry has errors.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): esp32-devkit
- [x] Tasmota binary firmware version number used: 12.1.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: idf.py and platformio

### TO REPRODUCE
run 1
```
import json
var s = "{\"abc\":16}"
var j = json.load(s)
print(j)
```
output:
```
{'abc': 16}
```
run 2
```
import json
var s = "{\"abc\":0x10}"
var j = json.load(s)
print(j)
```
output:
```
nil 
```
run 3
```
import json
var s = "{\"abc\":{\"hex\":16}}"
var j = json.load(s)
print(j)
```
output
```
{'abc': {'hex': 16}}
```
run 4
```
import json
var s = "{\"abc\":{\"hex\":0x10}}"
var j = json.load(s)
print(j)
```
reboot... 

**(Please, remember to close the issue when the problem has been addressed)**

closed 2022-08-31 by close-issue-app[bot]

darcyg, 2022-08-31:
patch
```
diff --git a/lib/libesp32/berry/src/be_strlib.c b/lib/libesp32/berry/src/be_strlib.c
index 23fdf978f..2c5424634 100644
--- a/lib/libesp32/berry/src/be_strlib.c
+++ b/lib/libesp32/berry/src/be_strlib.c
@@ -263,6 +263,9 @@ BERRY_API bint be_str2int(const char *str, const char **endstr)
         while ((c = be_char2hex(*str++)) >= 0) {
             sum = sum * 16 + c;
         }
+        if (endstr) {
+            *endstr = str - 1;
+        }
         return sum;
     } else {
         /* decimal literal */

```

barbudor, 2022-08-31:
@darcyg if you have the solution, you are welcome to provide a PR
Thanks

s-hadinger, 2022-08-31:
Discussion moved to https://github.com/berry-lang/berry/issues/287

s-hadinger, 2022-08-31:
Let me copy here my answer
---

This is as designed. `{"abc":{"hex":0x10}}` is not valid json. Json does not support hex integers. You can double check with jsonlint.com

If you want hex, you need to wrap into a string. Yet Berry can transparently convert to an int using `int()`. If you force conversion to int it will work in both cases.

``` berry
import json
var s = '{"abc":{"hex":"0x10","dec":16}}'
var j = json.load(s)
print(j)
# {'abc': {'hex': '0x10', 'dec': 16}}
print(int(j['abc']['hex']))
# 16
print(int(j['abc']['dec']))
# 16
```