arendst/Tasmota #16419: Reset Problem since a couple of weeks (device was working OKE for years)

MAWMN, 2022-09-02:
### PROBLEM DESCRIPTION
Hello,

since a couple of weeks my device An Arduino D1 R2 is having problems of resetting itself suddenly, so to be sure I did update to the latest minimal version and had to reconfige completely the device and application settings, so I did..(previous was 10.0.0)
Without the new hardware (GPIO setting made, no resets occured, however after having set the GPIO configuration, again the resetting started (Reset every minute...)

the gpio settings I used were :
1 GPIO0 4 (Relay4, can also be set by the Web Interface)
2 GPIO1 0 (None, can also be set by the Web Interface)
3 GPIO2 1 (Relays1, can also be set by the Web Interface)
4 GPIO3 0 (None, can also be set by the Web Interface)
5 GPIO4 7 (Relays7, can also be set by the Web Interface)
6 GPIO5 6 (Relays6, can also be set by the Web Interface)
7 GPIO12 3 (Relays3, can also be set by the Web Interface)
8 GPIO13 7 (Button1, can also be set by the Web Interface)
9 GPIO14 2 (Led1, can also be set by the Web Interface)
10 GPIO15 2 (Relays2, can also be set by the Web Interface)
11 GPIO16 5 (Relays5, can also be set by the Web Interface)

Also ADC was set to Input

Configuration to do by the console (type the following commands one by one !!) :
Rule1 on Mqtt#Disconnected do Power2 ON endon on Mqtt#Connected do Power2 OFF endon
Rule2 on Wifi#Disconnected do Power3 ON endon on Wifi#Connected do Power3 OFF endon
Rule1 1 (Switch Rule1 ON)
Rule1 4 (Disable one-shot detection (Perform commands as long as trigger is met))
Rule2 1 (Switch Rule2 ON)
Rule2 4 (Disable one-shot detection (Perform commands as long as trigger is met))
LedState 7 (Show power state and MQTT messages as a led blink)
WifiConfig 5 (NO Reset of the board when Wifi Signal is lost, just wait for reconnect)

After a Device Reboot, a Reset is performed every minute
00:00:00.002 HDW: ESP8266EX
00:00:00.050 CFG: Loaded from flash at F8, Count 278
00:00:00.056 QPC: Count 1
00:00:00.069 Project tasmota - Device_Main Version 12.1.1(tasmota)-2_7_4_9(2022-08-25T11:33:55)
00:00:00.557 WIF: Connecting to AP1 mna_AP Channel 8 BSSId E0:3F:49:09:D2:D0 in mode 11n as sonoff01-3173...
00:00:03.754 WIF: Connected
00:00:04.006 HTP: Web server active on sonoff01-3173 with IP address 10.0.0.191
15:40:25.022 MQT: Attempting connection...
15:40:25.034 MQT: Connected
15:40:25.037 MQT: tasmota/kamer/sonoff01/tele/LWT = Online (retained)
15:40:25.039 MQT: tasmota/kamer/sonoff01/cmnd/POWER =
15:40:25.044 MQT: tasmota/kamer/sonoff01/tele/INFO1 = {"Info1":{"Module":"Generic","Version":"12.1.1(tasmota)","FallbackTopic":"cmnd/DVES_104C65_fb/","GroupTopic":"tasmota/kamer/tasmotas/cmnd/"}}
15:40:25.046 MQT: tasmota/kamer/sonoff01/tele/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff01-3173","IPAddress":"10.0.0.191"}}
15:40:25.048 MQT: tasmota/kamer/sonoff01/tele/INFO3 = {"Info3":{"RestartReason":"External System","BootCount":220}}
15:40:25.050 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER1":"OFF"}
15:40:25.051 MQT: tasmota/kamer/sonoff01/stat/POWER1 = OFF
15:40:25.053 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER2":"OFF"}
15:40:25.054 MQT: tasmota/kamer/sonoff01/stat/POWER2 = OFF
15:40:25.056 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER3":"OFF"}
15:40:25.057 MQT: tasmota/kamer/sonoff01/stat/POWER3 = OFF
15:40:25.059 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER4":"OFF"}
15:40:25.061 MQT: tasmota/kamer/sonoff01/stat/POWER4 = OFF
15:40:25.064 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER5":"OFF"}
15:40:25.065 MQT: tasmota/kamer/sonoff01/stat/POWER5 = OFF
15:40:25.067 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER6":"OFF"}
15:40:25.069 MQT: tasmota/kamer/sonoff01/stat/POWER6 = OFF
15:40:25.071 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER7":"OFF"}
15:40:25.072 MQT: tasmota/kamer/sonoff01/stat/POWER7 = OFF
15:40:25.419 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER4":"OFF"}
15:40:25.422 MQT: tasmota/kamer/sonoff01/stat/POWER4 = OFF
15:40:25.470 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER5":"OFF"}
15:40:25.471 MQT: tasmota/kamer/sonoff01/stat/POWER5 = OFF
15:40:25.521 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER6":"OFF"}
15:40:25.523 MQT: tasmota/kamer/sonoff01/stat/POWER6 = OFF
15:40:25.572 MQT: tasmota/kamer/sonoff01/stat/RESULT = {"POWER7":"OFF"}
15:40:25.574 MQT: tasmota/kamer/sonoff01/stat/POWER7 = OFF
15:40:26.461 QPC: Reset
15:40:29.387 MQT: tasmota/kamer/sonoff01/tele/STATE = {"Time":"2022-09-01T15:40:29","Uptime":"0T00:00:11","UptimeSec":11,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":33,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","Wifi":{"AP":1,"SSId":"mna_AP","BSSId":"E0:3F:49:09:D2:D0","Channel":8,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:40:29.429 MQT: tasmota/kamer/sonoff01/tele/SENSOR = {"Time":"2022-09-01T15:40:29","ANALOG":{"A0":511}}

### REQUESTED INFORMATION
Question: What is going wrong ???

regards Marcel

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Just power UP...

### EXPECTED BEHAVIOUR
Normal, had been working for several years wit version 6.4.0, 8.3.0 and 10.0.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-09-02:
Hardware can become defect. Try a other device.

barbudor, 2022-09-02:
You are likely triggering the device recovery because Tasmota sees your button_1 continuously pressed
Solve your Button_1 connection

MAWMN, 2022-09-03:
Finally after thorough investigation I found the problem...
When the serial connection was available it also was (also) supplying power, so the problem did not occur anymore, the loggings above were of the console..

The problem was a NON stable powersupply, now just still needs to fix it, but won't be any problem..

thanks for the support guys..

closed 2022-09-03 by MAWMN