arendst/Tasmota #16424: Thermostat fails after 16 hours

wchulze, 2022-09-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
one output. 

The controller is set to run in **predictive ramp-up** mode (ControllerModeSet 2).

The system operate in a closed loop:
=> **controller => solid state relays => heater(resistor) => sensor ( DS18B20)**

The controller operates in a closed loop until the preset time TimeRampupMaxSet = 960 has expired ( default 16 hours,max 24 hours). Then controller **always** stops controlling and sets the output into a fixed OFF state. Only a restart or **ControllerModeSet** reactivate the controller. As a result, however, the controller loses the values it has learned in the predictive ramp-up mode.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 Mini Pro
- [x] Tasmota binary firmware version number used: 11.1.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: Tasmota-PyFlasher-1.0
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
na

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Rule1 ON Power1#boot DO Backlog
Power on;
SensorInputSet 1;
OutputRelaySet 1;
ThermostatModeSet 1;
ControllerModeSet 2;
TempTargetSet 29;
TimeRampupCycleSet 1;
TimePiCycleSet 1;
TimeSensLostSet 1439
ENDON

17:47:44.043 CMD: Rule1 ON Power1#boot DO Backlog Power on; SensorInputSet 1; OutputRelaySet 1; ThermostatModeSet 1; ControllerModeSet 2; TempTargetSet 29; TimeRampupCycleSet 1; TimePiCycleSet 1; TimeSensLostSet 1439 ENDON
17:47:44.053 RUL: Stored uncompressed, would compress from 198 to 129 (-35%)
17:47:44.056 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":198,"Free":313,"Rules":"ON Power1#boot DO Backlog Power on; SensorInputSet 1; OutputRelaySet 1; ThermostatModeSet 1; ControllerModeSet 2; TempTargetSet 29; TimeRampupCycleSet 1; TimePiCycleSet 1; TimeSensLostSet 1439 ENDON"}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

17:52:30.474 CMD: status 0
17:52:30.479 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"NodeMcu2","FriendlyName":["NodeMcu2"],"Topic":"tasmota_193755","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
17:52:30.486 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T01:00:14","StartupUTC":"2022-09-01T14:52:16","Sleep":40,"CfgHolder":4617,"BootCount":73,"BCResetTime":"2022-04-18T08:36:19","SaveCount":93,"SaveAddress":"F8000"}}
17:52:30.491 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0(tasmota)","BuildDateTime":"2022.09.01 16:43:05","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"344/699"}}
17:52:30.495 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wolfgangtespe",""],"TelePeriod":10,"Resolution":"55A180C0","SetOption":["00008000","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
17:52:30.505 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":3,"Features":["00000407","8FDAC787","04368001","000000CF","010013C0","C000F981","00004404","00001000","04000020"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,39,45,56,62","Sensors":"1,2,3,4,5,6"}}
17:52:30.512 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-193755-5973","IPAddress":"192.168.178.77","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"BC:FF:4D:19:37:55","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:52:30.518 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-09-01T15:52:30","Local":"2022-09-01T17:52:30","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":99,"Sunrise":"07:07","Sunset":"20:32"}}
17:52:30.526 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-09-01T17:52:30","DS18B20":{"Id":"0300A27955AB","Temperature":24.9},"Thermostat0":{"ThermostatModeSet":1,"TempTargetSet":29.00,"CtrDutyCycleRead":0},"TempUnit":"C"}}
17:52:30.533 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-09-01T17:52:30","Uptime":"0T01:00:14","UptimeSec":3614,"Heap":26,"SleepMode":"Dynamic","Sleep":40,"LoadAvg":24,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wolfgangtespe","BSSId":"9C:C7:A6:30:9E:65","Channel":1,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

[... 198 lines cut ...]
Line 477 in function **ThermostatHybridCtrPhase** (not executed in predictive ramp-up mode)
Line 1466 in function **CmndControllerModeSet**

How is the time controlled
Line 845: Calculate the running time
  `time_in_rampup = TasmotaGlobal.uptime - Thermostat[ctr_output].timestamp_rampup_start;`

Line 856: Checking if the time is done 
   `if ((time_in_rampup <= (60 * (uint32_t)Thermostat[ctr_output].time_rampup_max))...`

In fact, the value of TimeRampupMaxSet seems to be the total run time of the controller for the **predictive ramp-up** operation.

After line  845 I implemented  the following statement:
```
if (time_in_rampup > (60 * (uint32_t)Thermostat[ctr_output].time_rampup_max))
    time_in_rampup = 60 * (uint32_t)Thermostat[ctr_output].time_rampup_max;
```
This fix the problem, but the local **time_in_rampup variable** is also used to calculate the predictive behaviour for the thermostat. Manipulation of the timing behaviour my be result in strange side effects, overflows, ...

### Further consideration
Assuming the heating system is designed for a minimum outside temperature of -15°C. If the outside temperature falls below this value, the desired room temperature (e.g. 20°C) can no longer be reached, but the heating remains active and keeps the room frost-free down to -35°C outside temperature. But after **TimeRampupMaxSet** minutes, the controller switches the output to a fixed OFF state. Now the room temperature drops to the outside temperature! That is not desirable.

The switch-off behaviour of the controller after reaching **TimeRampupMaxSet** minutes should be adjustable. I'm unsure about the side effects.

**(Please, remember to close the issue when the problem has been addressed)**

hemangjoshi37a, 2022-09-06:
same problem here. Please let me know if solved.
https://hjlabs.in

arendst, 2022-09-07:
@arijav can you take a look pls.

mihailovsboriss, 2022-10-28:
Thank you @wchulze for the detailed report. I can relate to the same issue using Tasmota 12.2.0.1 precompiled binary. Hope for some solution for this.

Gubelat, 2022-11-02:
Same here, i fuckt up my brain why the damn thing stops working after a short amount of time.
thank you guys. now i have found a trace for start investigating.

wchulze, 2022-11-13:
Yes, the author doesn't seem to have neither time or interest in removing it.

My test stops time_in_rampup. 
```
if (time_in_rampup > (60 * (uint32_t)Thermostat[ctr_output].time_rampup_max))
    time_in_rampup = 60 * (uint32_t)Thermostat[ctr_output].time_rampup_max;
```
Manipulation of the timing behaviour my be result in strange side effects, overflows, ..
Please test it very carefully.

Since I saw more problems with the thermostat, I implemented the solution from:
 https://tasmota.github.io/docs/Rules/#simple-thermostat-example
I made some additions and changes. If you are interested, please let me know.
This is not a solution to the problem, but it helps me at the moment.

Wolfgang

Gubelat, 2022-11-17:
@wchulze 
Can you explain wich other problems you run in to?

@arendst 
Maybe somebody should add a Warning in the documentation about this issue. before other people  run in to the same trap as we here.

Maybe i switch off the controller mode and use the Home assistant Thermostat feature until sombody can fix this.

Greatful thanks to the Community.

wchulze, 2022-11-19:
**_Can you explain wich other problems you run in to?_**
When searching, I tentatively set time_in_rampup = 0. Then, there were irregular exceptions over time with subsequent reboots.

Furthermore, the thermostat can be switched off and then not switched on. A restart is necessary for this. It is unclear how timed on/off via the Tasmota timers should be done..

The frost protection limits the possibility of entering a temperature that is too low. I couldn't find another check.

**_Maybe i switch off the controller mode and use the Home assistant Thermostat feature until somebody can fix this._**
-> do not use the thermostat ???

Wolfgang

Gubelat, 2022-11-20:
@wchulze 

I mean, if i run in serius problems in the long term, then i will turn off the thermostat driver in tasmota and use home assistants thermostat feature. Or i use my trusty old Heimeier Radio Control F System that is still wired up to the sw1 input to the shellys.

Frogmore42, 2022-11-24:
I am a little confused what the use case you have is and why you believe you need to use predictive ramp up mode.  I didn't read all the code, but it seems that TimeRampupMax is a safety to keep the heater from being on always.  The whole reason you need PI control is if you have a heater that is significantly greater than the heat loss and thermal inertia of the system.  If your heat is so under powered that it can take more than 24 hours to bring the system up to temperature, a bang-bang control system is more than capable of handling it.

It seems like turning off the heater when it appears that it is not really working might be an okay solution and in the case where the temperature sensor is defective would be a really good thing.

Looking at the code some more, I believe you are correct that the implementation is not correct for mode CTR_RAMP_UP.  When this is set (which you are doing with ``` ControllerModeSet 2```) ```ThermostatWorkAutomaticRampUp()``` always gets called.  This function appears to have been written to work correctly with the hybrid control.  When in that mode, lines 455-463 (in ```ThermostatHybridCtrPhase```) will switch modes to ```CTR_HYBRID_PI```.

I would add these lines at about 953 (in the last else  of ```ThermostatWorkAutomaticRampUp```):
```
    // restart ramp-up if in CTR_RAMP_UP mode
    if (Thermostat[ctr_output].status.controller_mode == CTR_RAMP_UP) {
      Thermostat[ctr_output].timestamp_rampup_start = TasmotaGlobal.uptime;
    }
   // Set to now time to get out of ramp-up
```
This will reset the rampup start time.  You should also set the max time to a reasonable value, based on the thermal dynamics of your system.

I am still unclear why you feel you need to always be in ramp up mode and not hybrid?

Frogmore42, 2022-11-24:
or, in better keeping with the style of the code, make the following change in ```ThermostatCtrState()```
```
    // Ramp-up controller (predictive)
    case CTR_RAMP_UP:
      // If ramp-up off time counter has been initialized
      // AND ramp-up off time counter value reached
      if((Thermostat[ctr_output].time_ctr_checkpoint != 0)
        && (TasmotaGlobal.uptime >= Thermostat[ctr_output].time_ctr_checkpoint)) {
        // Reset times
        Thermostat[ctr_output].time_ctr_checkpoint = 0;
        Thermostat[ctr_output].timestamp_rampup_start = TasmotaGlobal.uptime;
      }
      break;
```
Both versions of the code do the same thing, reset the ramp up start time whenever the temperature is set correctly (under normal conditions) or the total time has expired.  Turning off the heat if ramp up time is exceeded is probably a good thing (it probably indicates a bad condition and tuning off the heat is probably safer than leaving it on forever).  Since CTR_RAMP_UP is always in the same mode, it clearly needs to reset the ramp up start time when things are good (which it currently doesn't).  I have not tested this code, since I use much simpler bang-bang control for all my systems.  But, that was because this module was not available when I started and doing it in Node-RED was easy enough.

wchulze, 2022-11-24:
Now I'm confused.
You wrote
**_I am a little confused what the use case you have is and why you believe you need to use predictive ramp up mode.
...
The whole reason you need PI control is if you have a heater that is significantly greater than the heat loss and thermal inertia of the system._**

I hope that nobody owns such a heater  What does that have to do with Proportional/Integral behavior? I prefer a two point controller Finally, however, you wrote:

_**I am still unclear why you feel you need to always be in ramp up mode and not hybrid?**_

From my understanding hybrid mode is a mix of ramp up follwed by PI control. So , after some time, you run in PI mode.

Thank's for your fix:
```
      // If ramp-up off time counter has been initialized
      // AND ramp-up off time counter value reached
      if((Thermostat[ctr_output].time_ctr_checkpoint != 0)
        && (TasmotaGlobal.uptime >= Thermostat[ctr_output].time_ctr_checkpoint)) {
        // Reset times
        Thermostat[ctr_output].time_ctr_checkpoint = 0;
        Thermostat[ctr_output].timestamp_rampup_start = TasmotaGlobal.uptime;
```
This code will reset the counter in case off timer overflow and start again. 
From my point of view it do not fix the root cause. It may even change the predictive behavior

From the declaration:

` uint32_t time_ctr_checkpoint = 0;    // Time to finalize the control cycle within the PI strategy or to switch to PI from Rampup in seconds`

It looks like **time_ctr_checkpoint** shouldn't have anything to do with ramp up mode

Wolfgang

Frogmore42, 2022-11-25:
Lots of people have heaters that are much greater than the heat loss.  I have at least two.

Since I don't know how much you know about controllers, here is more information on the details of PID control.  I am not sure how the controller in Tasmota could really be considered Proportional.  It does not appear to have the ability to change the amount of heat, just on or off.  Integral control helps prevent the offset error that a proportional only control system would have, but the way it does this tends to lead to overshooting the set point.  Derivate control is there to reduce overshoot.

I didn't look at all of the code to see what it was doing (as there is quite a bit of code and it wasn't all that clear).  I did see that the comments suggested that Ramp Up is for when the set point is changed, so forcing it to always be in ramp up mode doesn't seem like a good idea (and wasn't since it didn't work right).

@wchulze you haven't stated what you are controlling nor why you like a two point controller nor which two points.

I use my controllers for freeze protection, so maintaining a particular set point is not critical at all.  Having said that, I did make a recent change to reduce heat input on one system (and had to increase the heat input significantly on another to deal with very low temperatures). 

As far as ```time_ctr_checkpoint``` goes, the codes uses that to switch states.  There was a bug in the original code.  You are correct that ramp up doesn't need it, but other parts of the code do need it and it is used sort of like a boolean to indicate a state change is needed.  The code I added takes advantage of that to reset the state, since ramp up wasn't really designed to be run continuously.

Have you tried out the new code and does it fix your issue?

wchulze, 2022-11-26:
I got the knowledge about control in the lecture on control technology in the 6th semester of physics, Therefore, the correlations are well known.

I would like to heat a room with 12 m². The maximum heating capacity is 1.4 kWh. It can be scaled down. The radiator is a Yali Digital. External control is planned for convenience with Alexa.

When implementing a PI controller, it takes a lot of time to optimize the parameters, especially for the integral part. Otherwise, it is easy to have significant over shoots and "wind-ups". Using an infinite step function is difficult for my uses case. For this reason I would like to start with the ramp up implementation.

It looks like the controller implements a pulse width modulated control The frequency is `1/TimePiCycleSet` or `1/TimeRampupCycleSet` . The output is switched on/off once within this time window This is typically PWM, in this case at a fairly low frequency. The duty cycle can be determined with **CtrDutyCycleRead**.
I could not identify the controller equation
    `y = Kp * e + Ki * Ta * esum`
or a corresponding equivalent in the time domain, the code is far too complex.

**Now to the actual problem:**

The problem occurs when the` ramp up controller` – unconditionally - run the time
`Thermostat[ctr_output].time_rampup_max`
I have already described it in detail above. 

I'm not sure if the fix solves the problem. To me it looks like the problem persists, but the controller is being forced to live on.
It would be good if the control value temp_rampup_start was reset for a functioning system. But what is a functioning control system? Unfortunately, we don't have a comparative value for the temperature to decide. 

We could include timing. To do this, a resettable timer, like `temp_rampup_start`, might have to be used. The reset condition may be the regulation of the output. Unfortunately I haven't found the appropriate place and condition in the code to retrigger it. The code is quite difficult to understand.

wchulze, 2022-11-26:
The controller equation for the PI part:

`aux_temp_error = (int32_t)(Thermostat[ctr_output].temp_target_level_ctr - Thermostat[ctr_output].temp_measured) * 10;`
...
`Thermostat[ctr_output].time_proportional_pi = ((int32_t)(Thermostat[ctr_output].temp_pi_error * (int16_t)Thermostat[ctr_output].kP_pi) * ((int32_t)Thermostat[ctr_output].time_pi_cycle * 60)) / 10000;`
...
`Thermostat[ctr_output].time_integral_pi = (((int32_t)Thermostat[ctr_output].temp_pi_accum_error * (int32_t)Thermostat[ctr_output].kI_pi) * (int32_t)((uint32_t)Thermostat[ctr_output].time_pi_cycle * 60)) / 1000000;`
...
`Thermostat[ctr_output].time_total_pi = Thermostat[ctr_output].time_proportional_pi + Thermostat[ctr_output].time_integral_pi;`

It is a PI controler with PWM output,

Wolfgang

micaelaman, 2022-11-27:
I have the same issue. Glad I found this thread. I could not understand why my thermostats (7 pcs) suddenly just stop working.

Gubelat, 2022-11-28:
I compiled the version in dev suggestet by Theo. but this version restarts frequntly with a kind of exeption.

wchulze, 2022-11-28:
I can confirm that as well, as I mentioned it before:
**_When searching, I tentatively set time_in_rampup = 0. Then, there were irregular exceptions over time with subsequent reboots._**
The reason is unknown yet.

Wolfgang

serhat-ozkara, 2022-11-28:
Why dont you use https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xdrv_driver/xdrv_49_pid.ino in thermostat driver and have a sing actual "working" logic? From my experience of using Thermostat driver, it always overshoots or dips too low. It is simply not usable. PID holds temps really really well... 
This also simplifies maintaining as more of the same logic is implemented 2X. Just my 2 cents... @wchulze

wchulze, 2022-11-28:
@serhat-ozkara Thanks for the information. I haven't considered the module yet because its usage is not entirely clear.
https://tasmota.github.io/docs/PID-Control/
The documentation doesn't help directly. So I need to analyze the source code. I try it later.

Wolfgang

micaelaman, 2022-11-28:
Because my knowledge in coding and compiling Tasmota is somewhat (very) limited, I will reset the thermostats once every 24 hours. 
However, I'm willing to help with tests of new versions of thermostats.

micaelaman, 2022-11-30:
Looks good! I will try to do a test to night.

wchulze, 2022-11-30:
Looks good! ? Division by zero is a fatal exception !

micaelaman, 2022-11-30:
Told you. My coding competence is limited 😀
I'm trying to figure out how to compile with modified code.

Gubelat, 2022-12-01:
> **Fix** ` Thermostat[ctr_output].timestamp_rampup_start = TasmotaGlobal.uptime;`
> 
> **845** `time_in_rampup = TasmotaGlobal.uptime - Thermostat[ctr_output].timestamp_rampup_start;`
> 
> **879** `Thermostat[ctr_output].temp_rampup_meas_gradient = (int32_t)((360000 * (int32_t)temp_delta_rampup) / (int32_t)time_in_rampup);`
> 
> **Divide by zero**
> 
> Wolfgang

@wchulze thanks for your work to get this running.  Is this an new try to fix the issue?
and where i need to insert your new code?  In stable or development?

I need to copy paste line 845 and 879 right? an what is about the code behind **Fix**

best regards Marco

wchulze, 2022-12-02:
`Thermostat[ctr_output].timestamp_rampup_start = TasmotaGlobal.uptime;`
This is the fix form Frogmore42, It results in divide by zero and is followed by an exception.

In line 845 the zero is generated and line 879 shows the dividing by zero follwed by restart. This both lines ( 845 und 878 ) are from the original code, This is not a solution to the problems I have described!

Wolfgang

Frogmore42, 2022-12-04:
Thanks for finding the logic error.  I did not write the original code, so I missed that.

The quick fix is to change the "fixed" line to:
```
Thermostat[ctr_output].timestamp_rampup_start = TasmotaGlobal.uptime - 1;
```
This will avoid the divide by zero and will be exactly what would happen at the start of the process.

wchulze, 2022-12-04:
Well, this eliminates dividing by zero.

The author of the thermostat described:
**_The "Ramp-Up" controller evaluates the time constant of the system and predicts when to switch off the actuator to reach the desired temperature as fast as possible. This controller offers the best speed to reach the Setpoint. This controller will be improved by a learning process to evaluate how accurate the target value has been reached and therefore minimize gradually the overshoot._**

That was my main reason for choosing this mode. I think the control flow and data flow is mixed up. It is therefore difficult to see the effects.  From my point of view with the code 

    Thermostat[ctr_output].timestamp_rampup_start = TasmotaGlobal.uptime - 1;

the improved learning process is omitted, since the controller then starts with fresh values. That repeats every

`Thermostat[ctr_output].time_rampup_max`
 
set by TimeRampupMaxSet  minutes.

Frogmore42, 2022-12-06:
How frequently are you changing the set point?
My (limited) reading of the comments/code indicated that the mode you selected is not the mode one should normally use.  Use hybrid mode.  I believe the code for that works as designed.

If your heater is so underpowered that it takes many hours to reach the set point, it will never get there faster.

I am still unclear on what type of system you are controlling and what benefit you hope to gain from using the ramp up mode.

I believe this controller is a learning controller so you don't have to figure out the constants to use for a simple PID controller.  But, a bang-bang controller is possibly the most simple.  It typically has issues with how much error it has/allows (both under and overshoot) and how much it cycles the heater.  For my freeze protection systems, none of that really matters.  For my home furnace, both of those could be issues.  Short cycling a big furnace is a bad idea and big differences from the set point can lead to lower comfort.

What does "best" control look like to you?

As an example many home thermostats take into account the outside temperature to determine when to start the furnace to deal with getting the house to the new temperature set point at the time it changes instead of responding to the new set point at the time it actually changes.  For houses that two stage or variable input, some thermostats are smart enough to know which stage/input level to turn on.  For houses that only have a single stage, the best it can do is decide when to turn on the heat and when to turn it off and to know that it can only do it so many times an hour.  There isn't any real magic here.  But, there can be some better response for some events.

wchulze, 2022-12-06:
Frogmore42 In this forum we should discuss problems and especially solutions in the Tasmota software. Tasmota has its own forum for exchanging experiences etc

https://github.com/arendst/Tasmota/discussions

I have described the parameters of my heating, such as room size, power... I also described my reason and expectations for the ramp-up mode.

Wolfgang

Frogmore42, 2022-12-06:
My point is that you are using something to do something it was not really designed to do and clearly doesn't work.  Ramp up mode is not the mode you want for your needs.  As Steve Jobs famously said: https://www.wired.com/2010/06/iphone-4-holding-it-wrong/

wchulze, 2022-12-06:
Good to know, eh

Frogmore42, 2022-12-07:
https://www.futuredesigncontrols.com/DIN%20Special%20Features.htm

See the section on ramping.  Used for when the set point has changed and/or startup.  It is not really for steady state control.

Now, the original author of this code might have had something else in mind, but it looked like to me this was the intention.  Ramp mode is designed to get better results when the set point is changed.  But, it won't help much (if any) if the heater is going to take many hours to make a significant change to the system.  Overshoot is unlikely to be an issue in such a system.

wchulze, 2022-12-07:
Since you obviously do not know the (time) behavior of my heating and its environment, your statements regarding my system are quite questionable. This is also not necessary in the context of this problem report. Please let us concentrate on the problem with the software.

Frogmore42, 2022-12-07:
It is impossible to focus on a problem when you don't know which problem you are trying to solve.  Perhaps I am the only one who is confused here.  If that is the case I will stop talking.  To see what I was missing I took a look for some documentation on the controller and was able to find the page that talks about the various parameters.  I didn't see anything that didn't match what I saw in the code, except that the documentation says the maximum ramp up time is only applicable in hybrid mode.  That was clearly incorrect.  The latest change above should address the controller stopping after that time period.

My understanding, which might be flawed, is that ramp up mode is only for when there is a step change in the system.  It is not for general use.  This is why I asked what you hoped to achieve with using that mode.  Clearly only using that mode is not working the way you expect.  I didn't write the original code and don't use it.  It looked like an interesting problem so I took a closer look.  I don't see this code as being an improvement over what I am currently using, so my interest in it is diminishing.  I am sorry I wasn't able to help you.

wchulze, 2022-12-07:
I described the problem in detail at the top of the PR. Thank's for your interest in finding a solution.

Wolfgang

Jason2866, 2022-12-07:
@wchulze 
What is the status here? It looks to me like an academic discussion now.
If there is no explanation provided (with a link to a source) telling the way it is implemented in Tasmota is faulty, i will move to discussion.

wchulze, 2022-12-07:
Jason2866
I just ended this discussion with Frogmore42 as it didn't lead to a solution!. However, the actual problem still exists, as I described at the beginning of this PR. A number of users have reported the same problem.

[4303968](https://github.com/arendst/Tasmota/commit/430396832f82cde51ff3ee3370ef20d9753e59e8)
This Fix ends in divison by zero and do not fix the problem.

Wolfgang

arijav, 2022-12-07:
Hello everybody,

I've just seen the messages. I do not read this email so frequently
anymore. Unfortunately I do not possess Tasmota based devices any longer,
so I can not replicate the issues described. Some comments on my side:

1- At the time the driver was developed it was working perfectly fine with
the tasmota SW, not just for me, but for many other users I got feedback
from. If it does not work now, look at changes on the interfaces, not the
driver itself (you might however need to adapt the code to those iface
changes if any).

2- I have used, in my case, this driver exclusively with a heating floor
system (described in the documentation). It is a system with a huge delay.
When using exclusively a PI controller, I had either issues at the
setpoint to maintain accuracy or it took ages to reach it. This is the
reason of the creation of an hybrid mode with a simple, so called (you
might like it or not) Ramp-Up mode which basically makes a linear
extrapolation to estimate when to switch off the valves to reach the
setpoint during the phase where it is far away from the measured
temperature. Once the driver gets close to the setpoint it switches to a
classical PI controller. The Ramp-Up mode alone was not really meant to be
used, even if the possibility is given to the user to select it. In my
heating system, tuning the P/I for it, I got an accuracy of <0.2°C.

3- I know for a fact that other people have used the driver with fan based
systems or other kinds of radiators. To optimize the driver for those
systems a reparametrization could be needed. It is an open source driver,
not meant as a commercial solution.

BR,

Javier

Frogmore42, 2022-12-07:
> The Ramp-Up mode alone was not really meant to be
used, even if the possibility is given to the user to select it.

I believe that statement above is the crux of the problem.  Ramp-Up mode has a time limit, which is quite short.  The design intent (for allowing Ramp-Up only mode) was not really clear from the code.  I thought it might have been there to make testing easier.

This problem report is all about when the controller is forced to stay in Ramp-Up mode it stops working after the time limit.  That seemed like a good design choice for the mode.  Allowing that mode to be selected as the only mode did not seem like a good idea to me, nor did using just that mode.  I submitted a fix to allow Ramp-Up mode to work by itself.  That has been included in the dev branch.  That could lead to a divide by zero error, so 4 days ago I submitted a quick fix for that.

I don't believe anybody has said there is a problem with the code running in hybrid mode.  It is only when Ramp-Up only mode is selected that there is any issue and it appears the original intent was that mode should not be used for normal usage/control.

Jason2866, 2022-12-07:
Closing, since the Ramp Up mode is working as designed and is not ment to be used as standard mode.

closed 2022-12-07 by Jason2866

Jason2866, 2022-12-07:
It is Open Source, everyone can contribute. And yes Docu needs change, since it describes not the state of the implememtation how it is done.
The docu is Open Source too, a PR is welcome.

Gubelat, 2022-12-08:
@arijav 
I try to describe my case here. Sorry for my bad english.
I starting to build the system as described in the documentation at august 22. i compiled tasmota with thermostat successfully and flashed my 4 shellys. At this time the central heating system was not powered on yet, because the temperature outside was plenty high . so i had to wait until oktober. Then in oktober the temperatures dropped under 15°C at night. The heating system turned on. So i powered the shellys on, and gave them temperaturetargetset 19 °C,  but in my appartment the temperatue was stll above this targettemperature for longer then 1440 minuts.  After this time the temperaturs was dropped below 19°C and the thermostat had stopped to work. 

The Problem is not only the ramuptime. It seems to be a global timeout. If there is no action on the relay for more then 16 h then the software stopps working in general. Valve control is not possible anymore. In worst case the appartment will freeze if nobody is at home or other thermostats take over.

maybe there is a simple way to turn off this timouts, so that we can use this nice software in a endless loop without the fear of frozen homes :-)

Thanks in advance. 
Marco

Frogmore42, 2022-12-09:
@Gubelat you should probably create a new issue (follow the template and include all the logs).  This issue is about the controller being forced to be in Ramp-Up mode.  If you enabled that, turn it off and use hybrid mode.  I looked at the code and it appeared to handle the case where the time in ramp-up mode exceeds the maximum, when the controller is in hybrid mode.  If you are seeing a case where the controller is in hybrid mode and is failing, logs are the only way to get help with it.

Gubelat, 2022-12-09:
@Frogmore42 

There only 2 Modes. Controllermodeset 1 for pure PID and Controllermodeset 2 for a Rampup and PID.
There is no pure rampup mode.  In Controllermodeset 2 the thermostat stays in Rampupmode shortly until the setpoint temperature is reached, then it switches to PID.

My case is direct relatet to this.

Jason2866, 2022-12-09:
@Gubelat Why not use Mode 1 ?

Gubelat, 2022-12-09:
@Jason2866 because my homassistant sets the target temperature to 18° C at night and 20°C @ daytime.
the pure pid control is unable to reach such deltas in hours. 
If the target temperature delta is more than -0,4 Kelvin then the rampup mode kicks in, and try to reach setpoint temperature as fast as possible. If Delta T in the range of 0,4 Kelvin around Setpoint, then PID Mode Kicks in again.

Jason2866, 2022-12-09:
@Gubelat  Seems you have to find an other solution. As the Author of this code mentioned it is as it is, no changes planned.

Frogmore42, 2022-12-09:
From the documentation:

> Set the control strategy[~](https://tasmota.github.io/docs/Thermostat/#set-the-control-strategy)
The control strategy by default is a Hybrid one. The hybrid control strategy mixes a so called "Ramp-Up" strategy (invention during the development of this driver, initially as a LUA script and ported to Tasmota later) and a PI one. The "Ramp-Up" strategy is typically used to reach as fast as possible the setpoint for big deltas between desired temperature and setpoint, the PI on the other hand for most part of the normal thermostat operation.

>The control strategy can be however modified, if the Hybrid one is not desired, the PI or "Ramp-Up" mode, can be forced. For that purpose the following command can be used:

>cmnd/Tasmota_Name/CONTROLLERMODESET 1
>The value 1 forces the PI operation and the value 2 the "Ramp-Up" one.

From the code:
`enum ControllerModes { CTR_HYBRID, CTR_PI, CTR_RAMP_UP, CTR_PI_AUTOTUNE, CTR_MODES_MAX };
`

That looks like mode 0 is the correct mode to use for your case.  As @arijav said many people have used the code successfully previously.

wchulze, 2022-12-09:
@[Gubelat](https://github.com/Gubelat)
At the moment I'm using the rule-based control from
https://tasmota.github.io/docs/Rules/#simple-thermostat-example
This is a simple 2 step controller. I have extended the rules to enable night setback / frost protection. The rules support the detection of a sensor failure.

Wolfgang

Gubelat, 2022-12-10:
if there really a controllermodeset 0 i give this a try now.

the documentation foold me then. because the Words ramp-up where in quotemarks. this leads me to think that "Ramp-up" means the Hybrid mode (controllermodeset 2). there is no mention about controllermode 0 in the docs.

wchulze, 2022-12-10:
[Gubelat](https://github.com/Gubelat)  You weren't wrong at all. The documentation of the thermostat in Tasmota is here :
https://tasmota.github.io/docs/Commands/#thermostat
All modes 0, 1, 2 are clearly described.
There is also no contradiction in the further description
https://tasmota.github.io/docs/Thermostat/

Wolfgang

dexterbot80, 2023-05-15:
Hello guys,
You killed me.
I read this forum and understood that it is a problem with the thermostat because it stops after a while.
Please tell me how to make it work permanently because otherwise I'm freezing here - the only heat source is the soldering station :)

I installed the Tasmota Thermostat on a Shelly Uni device and actuate a contactor that feeds an electrical resistance in the floor.
Everything works perfectly for a while, after which the thermostat freezes - although the set temperature is 22 degrees Celsius, it no longer engages the contactor to heat the resistance.

Can someone tell me how to set the thermostat so that it works permanently without stopping?
Thank you