arendst/Tasmota #16431: GPIO not responding correctly in timings critical sensors

tsctrl, 2022-09-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am reading gpio input which require critical timing for start bit and end bit. However GPIO are not responding correctly in the timings input that was initiated to the gpio pin. For the sketch with simple loop in other sketch i am getting the correct timings within the required tolerances. the timing was 3.7ms and tolerance for 0.2ms. by using new driver and feed the codes in the 'case FUNC_LOOP:' i am getting feedback pulse around 5ms which too slow to confirm the start bits. tried to use interrupt but inconsistence reading for 1ms-5ms.

added the codes in the main loop and main setup under support files also did not fix the issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ /] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [/ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ /] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [/ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [/ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ /] Device used (e.g., Sonoff Basic): _____ devkits esp8266
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [/ ] Self-compiled 
- [/ ] Flashing tools used: _____ pio
- [/ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:01:08.158 CMD: Backlog Template; Module; GPIO 255
22:01:08.251 RSL: RESULT = {"NAME":"Generic","GPIO":[224,0,8832,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":18}
22:01:08.458 RSL: RESULT = {"Module":{"0":"Generic"}}
22:01:08.716 RSL: RESULT = {"GPIO0":{"224":"Relay1"},"GPIO1":{"0":"None"},"GPIO2":{"8832":"SENSOR"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:02:29.560 CMD: Status 0
22:02:29.566 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_97E4D6","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:02:29.594 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:02:55","StartupUTC":"2022-09-03T20:59:34","Sleep":50,"CfgHolder":4617,"BootCount":33,"BCResetTime":"2022-09-03T18:48:20","SaveCount":64,"SaveAddress":"B9000"}}
22:02:29.626 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.1.1(tasmota)","BuildDateTime":"2022-09-04T04:36:02","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"367/699"}}
22:02:29.646 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
22:02:29.679 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":676,"Free":68,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004084","00201000","54000020","00000004"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,38,45,50,62","Sensors":"1,2,3,4,5,6"}}
22:02:29.710 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-97XXD6-1238","IPAddress":"192.168.0.142","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
22:02:29.736 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_97E4D6","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:02:29.756 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-09-03T21:02:29","Local":"2022-09-03T22:02:29","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:10","Sunset":"19:28"}}
22:02:29.775 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-09-03T22:02:29"}}
22:02:29.781 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-09-03T22:02:29","Uptime":"0T00:02:55","UptimeSec":175,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"","BSSId":"28:87:BA:81:D5:8F","Channel":1,"Mode":"11n","RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
hook an osci with 3.7ms intervals in gpio and observe the timings between change in loop

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
loop timings on rising and falling bits should consistence based on input timings

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-09-03:
How is that a bug in Tasmota?
You are reporting problems within your own code that you are not sharing.
Tasmota have some drivers with critical timings including DS18X20, DHTxx, Wd2812
Have you taken the time to have a look at those?

s-hadinger, 2022-09-03:
Such tight timing require an interrupt driver. Looking at TasmotaSerial can give you some hints. I applied some optimizations to make 115200 more reliable. 

What is your protocol?

tsctrl, 2022-09-03:
i am following those example too. DHT use sequence and keep to read pin until specific pulse read occur and read sequence until end (consistent timings). Where i am continuously reading input pin and get the difference to get the specific low duration which is 3.7ms for the low duration and go to high to 4.5ms for the start bit timing. reading pins status in the loops and deduct the timing from first bit does not give me consistence reading. the driver mentioned approach are blocking and cant be interrupt. expecting the loop timings will always consistence or at least faster than 0.5ms to get consistence result. my codes with empty sketch just work fine and i am getting consistent timing. will check again for any workaround

tsctrl, 2022-09-03:
@s-hadinger not sure what protocol, but i am reading gpio for high and lows timings

tsctrl, 2022-09-03:
i guess i would need to enter critical for the transmission duration for the loop to run for sometime until complete/timeout when found beginning of a start bit. not efficient and blocking but i guess this is the way for 8266.

thanks @barbudor @s-hadinger

closed 2022-09-03 by tsctrl

barbudor, 2022-09-04:
You are not sharing much on what you are trying to achieve like pointing to the device doumentation with protocols and timings.
What you can do on a simple Arduino sketch where you are King Of The World and have the power of the ESP for yourself cannot be applied on a real system where you have to share and collaborate with the rest of Tasmota.
Blocking for short time (like 10-20ms) might be acceptable. 50ms would be already quite extreme.
When using interrupts there are mainly 2 ways of doing that:
1. Use a timer interrupt for period interrupts at least 4x the bit rate and read the GPIO, detect start-bit and then sample each bits at the right time. You need to be sure that there is no drift between the interrupt frequency and the protocol bit rate. You can still try to resynchronize during the transmission if you detect an edge too early or too late.
2. Use interrupts based on the GPIO. Start with falling edge for example and in the interrupt read the timer and change the interrupt edge. In the interrupt just store the timer value in an circular buffer. Process the timestamps in the background to calculate the waveform and find out the bit values.

The 1st option is generally simpler but would require a much higher interrupt frequency. You also need a free timer to use at your own frequency.
The 2nd option is less CPU consuming as the interrupt is triggered only on edges but its more complicated to process timestamps than sampling bits.

s-hadinger, 2022-09-04:
TasmotaSerial uses a mix of both. The first trigger is based on interrupt from a GPIO then there is a spin loop that check the signal at 4x the baud rate until the bytes are received. The only caveat is that you don't know the timestamp of the initial interrupt. 

Interrupt driver take typically 500 cycles to enter your code so you need to deduct it from your time calculation. However sometimes the ESP is already in another interrupt driver so you may miss more cycles than the 500 mentioned. So it's good to re-synchronize if possible

tsctrl, 2022-09-05:
thank you @barbudor , @s-hadinger for the feedback and recommendation. 

i have notice the issue was causing by the power option to reduce the processing power on the main loop. puting the power option to 0 allow my codes to run properly and able to capture the toggling signals. understood this was not optimal as is sometime cause missed triggers/signal from the sensors. however this was the easiest solution i have temporarily as i am 2 days experienced on tasmota codes and home automation stuff.

thank you so much on the suggestion as thats what should be done properly. i will go thru the approach as the time to learn how to handle this kind of challenge properly as i am from rtos that cause me to think this interupt timings is too complex for me to do it.

thank you!

tsctrl, 2022-09-05:
the current approach is to sample on the main loop and process the edge change agains the last timestamp. i guess the 1st approach will be direct drop solution but require additional software timers if this i understand correctly.
the 2nd approach i mention before where i am using the gpio interupts does not work as per my skills. i just trigger the loop function as the approach 1 as per last timestamp but it does not pickup the timing correctly. i guess there is delay on interrupt triggers as @s-hadinger mention. i will try both options again and investigate further. 

thank you @barbudor , @s-hadinger