arendst/Tasmota #16446: Strange behavior of ESP_IR_TR with SerBr Tx or IRsend on GPIO4/GPIO14

TheChatty, 2022-09-05:
### PROBLEM DESCRIPTION
I own an [ESP_IR_TR](https://ae01.alicdn.com/kf/S7c37a1d55ab545909dca7446ff60997cZ/ESP8285-ESP-01M-IR-Transceiver-Wifi-Wireless-Module-Remote-Control-Switch-Development-Learning-Board-esp-8285.jpg_Q90.jpg_.webp) which connects IR emitter to GPIO4 and IR receiver to GPIO14.

When configured to **IRsend/IRrecv** it **works** like it should: emitter usually dark, receives commands from real TV remote, sends those commands to real TV.

When configured to **SerBr Tx/Rx** the emitter is **always lit**, nothing is received when other IR head tries to send something (both configured to e.g. 9600,8n1), steady "garbage" is received by other IR head.

FYI: [other IR head](https://i.ebayimg.com/images/g/WNUAAOSwYI9h0Loa/s-l1600.png) + FTDI (working flawlessly in other environments)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP_IR_TR
- [x] Tasmota binary firmware version number used: 12.1.1.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: esptool.py
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
at ESP_IR_TR
```
backlog SBaudrate 9600;SSerialConfig 8N1
SSerialSend5 1F 2E 3D 4C 5B 6A 79 88 97 A6 B5 C4 D3 E2 F1
```

at other IR head:
```
SerialReceived 70 C2 85 FF C6 00 FE C2 0D FF EF D1 FE E2 00 FE F7 59 FE FB 18 00
```

### EXPECTED BEHAVIOUR
at other IR head:
```
SerialReceived 1F 2E 3D 4C 5B 6A 79 88 97 A6 B5 C4 D3 E2 F1
```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-09-05:
If I understand well you are asking for inverted IrTx which is not supported by TasmotaSerial. I'm hesitant to change TasmotaSerial driver for esp8266 since it's a delicate piece of code.

TheChatty, 2022-09-05:
So it's normal that the IR emitter is lit on LOW?

s-hadinger, 2022-09-05:
Serial is always high when idle and low to mark start and end bits. This is the contrary of IR.

TheChatty, 2022-09-05:
But the IR emitter connected to my FTDI is not lit.