arendst/Tasmota #16454: PWMCT vs SetOption20

dgwynne, 2022-09-07:
### PROBLEM DESCRIPTION

When the colour temperature of a light using PWMCT is changed while the light is off (thanks to SetOption20), the CT change isn't applied until the bulb is powered on. This is particularly noticeable with `fade` enabled too. The following commands should demonstrate the issue.

- `so92 on` - enable the second PWM as CT (Color Temp) instead of Warm White
- `so20 on` - Update of Dimmer/Color/CT without turning power on
- `backlog power on; ct 153; white 100; fade on; speed 2` - force a known state: cool white powered on
- `power off`
- `ct 500` - the CT should be warm white now
- `power on`

After the last command you can see the bulb fade on from cool white to warm white. I was expecting it fade on as warm white from 0, like all my other lights that have a separate warm white channel instead of something that needs PWMCT configured.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Athom RGBCCT bulb (LB01-15W-E27-TAS-1)
- [x] Tasmota binary firmware version number used: 12.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: preinstalled, and then OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

See PROBLEM DESCRIPTION above

### EXPECTED BEHAVIOUR

A CT change while the light is off should apply immediately, not after the light is powered on.

### SCREENSHOTS

N/A

### ADDITIONAL CONTEXT

Should I patch this myself?

**(Please, remember to close the issue when the problem has been addressed)**

techman83, 2022-09-07:
Are you sure the white command is what you want here?

> 1..100 = set white channel brightness in single white channel lights (single W or RGBW lights)

https://templates.blakadder.com/athom_LB01-15W-E27.html

I'm guessing you probably want to use the Dimmer command to adjust the brightness

s-hadinger, 2022-09-07:
I can reproduce the bug. The fade wrongly takes CT 153 as a starting point for the fade, since it was the last effective value. The algorithm didn't take into account changes when SO92 is 1.

Let me see how I can fix this

closed 2022-09-08 by s-hadinger

s-hadinger, 2022-09-09:
It should be solved now, can you confirm?

dgwynne, 2022-09-09:
Yes, it's working great. Bulbs without so92 are still working fine too.

Thank you.

PR #16468 (2022-09-08): Fix fade pwmct #16454
## Description:

Fix fading wrong CT for devices with PWMCT, when the CT was changed as the bulb was off.

**Related issue (if applicable):** fixes #16454

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.4.1
  - [x] I accept the [CLA](https://github.com/ [... cut]