arendst/Tasmota #16457: Hiking DDS238-2 WIFI Blocked Loop in Tasmota 9.4.0 and newer

pablozg, 2022-09-07:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have a Hiking DDS238-2 WIFI, when I try to configure it with the template ```{"NAME":"Hiking DDS238-2 WIFI","GPIO":[0,2272,0,2304,0,0,0 , 0,0,0,320,0,32,0],"FLAG":0,"BASE":54}``` based on https://templates.blakadder.com/hiking_DDS238-2.html, after a few seconds , tasmota crashes and I lose access to the webpage, after some time tasmota reboots and I am able to access the webpage again, but in the Info tab I can see cause of reboot: Blocked loop.

[image]

The problem occurs from version 9.4.0, in version 9.3.1 it doesn't crash but I don't have any type of information on the main screen.

[image]

It doesn't matter if I activate the tuya commands or not, as soon as I apply the template, the problem happens, but If I select other template like "Sonoff Basic" no hang occurs, but I can see in the console:

```
11:41:29.404 CMD: 11:41:29.366 RSL: RESULT = {"Command":"Unknown"}
11:41:29.407 SRC: Serial
11:41:29.410 CMD: Group 0, Index 11, Command "", Data ":41:29.366 RSL: RESULT = {"Command":"Unknown"}"
11:41:29.414 RSL: RESULT = {"Command":"Unknown"}
11:41:29.421 CMD: Serial buffer overrun
11:41:29.427 CMD: Serial buffer overrun
11:41:29.432 CMD: Serial buffer overrun
11:41:29.438 CMD: Serial buffer overrun
11:41:29.445 CMD: Serial buffer overrun
11:41:29.452 CMD: 11:41:29.414 RSL: RESULT = {"Command":"Unknown"}
11:41:29.455 SRC: Serial
11:41:29.458 CMD: Group 0, Index 11, Command "", Data ":41:29.414 RSL: RESULT = {"Command":"Unknown"}"
11:41:29.462 RSL: RESULT = {"Command":"Unknown"}
11:41:29.469 CMD: Serial buffer overrun
11:41:29.474 CMD: Serial buffer overrun
11:41:29.480 CMD: Serial buffer overrun
11:41:29.486 CMD: Serial buffer overrun
11:41:29.492 CMD: Serial buffer overrun
11:41:29.499 CMD: 11:41:29.462 RSL: RESULT = {"Command":"Unknown"}
11:41:29.502 SRC: Serial
11:41:29.505 CMD: Group 0, Index 11, Command "", Data ":41:29.462 RSL: RESULT = {"Command":"Unknown"}"
11:41:29.511 RSL: RESULT = {"Command":"Unknown"}
11:41:29.518 CMD: Serial buffer overrun
11:41:29.523 CMD: Serial buffer overrun
11:41:29.528 CMD: Serial buffer overrun
11:41:29.534 CMD: Serial buffer overrun
11:41:29.541 CMD: Serial buffer overrun
11:41:29.548 CMD: 11:41:29.511 RSL: RESULT = {"Command":"Unknown"}
11:41:29.551 SRC: Serial
11:41:29.555 CMD: Group 0, Index 11, Command "", Data ":41:29.511 RSL: RESULT = {"Command":"Unknown"}"
11:41:29.559 RSL: RESULT = {"Command":"Unknown"}
11:41:29.565 CMD: Serial buffer overrun
11:41:29.571 CMD: Serial buffer overrun
11:41:29.576 CMD: Serial buffer overrun
11:41:29.582 CMD: Serial buffer overrun
11:41:29.589 CMD: Serial buffer overrun
11:41:29.596 CMD: 11:41:29.559 RSL: RESULT = {"Command":"Unknown"}
11:41:29.599 SRC: Serial
11:41:29.602 CMD: Group 0, Index 11, Command "", Data ":41:29.559 RSL: RESULT = {"Command":"Unknown"}"
11:41:29.608 RSL: RESULT = {"Command":"Unknown"}
11:41:29.614 CMD: Serial buffer overrun
11:41:29.620 CMD: Serial buffer overrun
11:41:29.626 CMD: Serial buffer overrun
11:41:29.631 CMD: Serial buffer overrun
11:41:29.638 CMD: Serial buffer overrun
11:41:29.645 CMD: 11:41:29.608 RSL: RESULT = {"Command":"Unknown"}
11:41:29.648 SRC: Serial
11:41:29.651 CMD: Group 0, Index 11, Command "", Data ":41:29.608 RSL: RESULT = {"Command":"Unknown"}"
11:41:29.658 RSL: RESULT = {"Command":"Unknown"}
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
[... 156 lines cut ...]
11:51:43.508 HTP: Console
11:51:44.420 TYA: Send "55aa00000000ff"
11:51:44.424 {"TuyaReceived":{"Data":"55AA00000000FF","Cmnd":0}}
11:51:44.427 TYA: Heartbeat
11:51:45.779 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Flash TYWE3S Board with Tasmota 9.4.0, and apply template `{"NAME":"Hiking DDS238-2 WIFI","GPIO":[0,2272,0,2304,0,0,0 , 0,0,0,320,0,32,0],"FLAG":0,"BASE":54}

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Have a working meter without hangs

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-09-07:
You said and newer. But your logs are from 9.4.0. Please use latest Tasmota 12.1.1 and post this logs. Only the latest release version is supported.

pablozg, 2022-09-07:
> You said and newer. But your logs are from 9.4.0. Please use latest Tasmota 12.1.1 and post this logs. Only the latest release version is supported.

I have updated the first post with the 12.1.1 console log.

> Have you applied all commands as described in the template site for the [device](https://templates.blakadder.com/hiking_DDS238-2.html) ?

Yes, and I can't see any difference with or without the commands applied. If I switch to the basic module, it doesn't crash, but with tuyamcu's configuration, it does.

pablozg, 2022-09-08:
Hi, I can confirm that hangs occurs only when the module selected is Tuya (54), Yesterday I leave the hiking meter configured as Generic (18) and it still working without issues, but when choose TuyaMCU it's hangs after exactly one minute from boot. Any Idea?

pablozg, 2022-09-09:
@Jason2866  Hi, I think I found the blocking function, after take a look to the source code I found a call to TuyaSetTime() every 60 seconds if we have rtp time. 

https://github.com/arendst/Tasmota/blob/f347a5b19c5bbc3cabf9c4b419822f4b8e9f1c2b/tasmota/tasmota_xdrv_driver/xdrv_16_tuyamcu.ino#L1395-L1419

So I have blocked the internet connection to the hiking ip and now still working without hangs, but still no having info in the main screen.

Log with level 4:

```
[19:56:49] cmnd/tasmota_F809C5/restart 1
[19:56:49] stat/tasmota_F809C5/RESULT {"Restart":"Restarting"}
[19:56:50] stat/tasmota_F809C5/LOGGING 00:03:09.079 SRC: MQTT
[19:56:50] stat/tasmota_F809C5/LOGGING 00:03:09.082 CMD: Grp 0, Cmd 'RESTART', Idx 1, Len 1, Pld 1, Data '1'
[19:56:50] stat/tasmota_F809C5/LOGGING 00:03:09.086 MQT: stat/tasmota_F809C5/RESULT = {"Restart":"Restarting"}
[19:56:52] tele/tasmota_F809C5/LWT Offline
[19:57:00] tele/tasmota_F809C5/LWT Online
[19:57:00] cmnd/tasmota_F809C5/POWER 
[19:57:00] tele/tasmota_F809C5/INFO1 {"Info1":{"Module":"Tuya MCU","Version":"12.1.1(tasmota)","FallbackTopic":"cmnd/DVES_F809C5_fb/","GroupTopic":"cmnd/tasmotas/"}}
[19:57:00] tele/tasmota_F809C5/INFO2 {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-F809C5-2501","IPAddress":"10.10.0.89"}}
[19:57:00] tele/tasmota_F809C5/INFO3 {"Info3":{"RestartReason":"Software/System restart","BootCount":53}}
[19:57:00] stat/tasmota_F809C5/RESULT {"POWER":"OFF"}
[19:57:00] stat/tasmota_F809C5/POWER OFF
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.001 HDW: ESP8266EX
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.050 CFG: Loaded from flash at F5, Count 144
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.055 QPC: Count 1
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.057 CFG: CR 364/699, Busy 0
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.061 ROT: Mode 1
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.107 SNS: Hardware Serial
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.107 TYA: Request MCU configuration at 9600 bps
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.108 SRC: Restart
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.122 TYA: Send "55aa0006000501010001000d"
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.143 Project tasmota - Tasmota Version 12.1.1(tasmota)-2_7_4_9(2022-08-25T11:33:55)
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.147 RSL: RESULT = {"TuyaReceived":{"Data":"55AA0006000501010001000D","Cmnd":6,"CmndData":"0101000100"}}
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.147 TYA: RX unknown command
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.154 TYA: Set WiFi LED 2 (-1)
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.163 TYA: Send "55aa000300010205"
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.166 RSL: RESULT = {"TuyaReceived":{"Data":"55AA000300010205","Cmnd":3,"CmndData":"02"}}
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.166 TYA: RX WiFi LED set ACK
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.215 WIF: Checking connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.216 WIF: Attempting connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:00.552 WIF: Connecting to AP1 Isabella Channel 4 BSSId C4:AD:34:B7:64:E7 in mode 11n as tasmota-F809C5-2501...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:01.754 WIF: Checking connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:01.755 WIF: Attempting connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:02.754 WIF: Checking connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:02.755 WIF: Attempting connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:03.755 WIF: Checking connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:03.756 WIF: Connected
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:04.009 HTP: Web server active on tasmota-F809C5-2501 with IP address 10.10.0.89
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:04.379 WIF: Sending Gratuitous ARP
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:04.380 NTP: Sync time...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:05.467 NTP: No reply from 162.159.200.1
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:05.469 TYA: Set WiFi LED 3 (0)
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:05.478 TYA: Send "55aa000300010306"
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:05.482 RSL: RESULT = {"TuyaReceived":{"Data":"55AA000300010306","Cmnd":3,"CmndData":"03"}}
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:05.483 TYA: RX WiFi LED set ACK
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:05.490 NTP: Sync time...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:06.578 NTP: No reply from 94.23.147.231
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:06.627 QPC: Reset
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:06.628 NTP: Sync time...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.698 NTP: No reply from 129.250.35.250
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.765 MQT: Attempting connection...
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.773 MQT: Connected
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.776 MQT: tele/tasmota_F809C5/LWT = Online (retained)
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.780 MQT: cmnd/tasmota_F809C5/POWER = 
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.782 MQT: Subscribe to cmnd/tasmota_F809C5/#
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.785 MQT: Subscribe to cmnd/tasmotas/#
[19:57:00] stat/tasmota_F809C5/LOGGING 00:00:07.789 MQT: Subscribe to cmnd/DVES_F809C5_fb/#
[... cut ...]
[20:02:13] stat/tasmota_F809C5/LOGGING 00:05:19.516 TYA: Heartbeat
[20:02:23] stat/tasmota_F809C5/LOGGING 00:05:29.791 WIF: Checking connection...
[20:02:23] tele/tasmota_F809C5/RESULT {"TuyaReceived":{"Data":"55AA00000000FF","Cmnd":0}}
[20:02:24] stat/tasmota_F809C5/LOGGING 00:05:30.414 TYA: Send "55aa00000000ff"
[20:02:24] stat/tasmota_F809C5/LOGGING 00:05:30.420 MQT: tele/tasmota_F809C5/RESULT = {"TuyaReceived":{"Data":"55AA00000000FF","Cmnd":0}}
[20:02:24] stat/tasmota_F809C5/LOGGING 00:05:30.422 TYA: Heartbeat
´´´

pablozg, 2022-09-22:
@Jason2866 @arendst Any suggestions or tests I can do? thanks.

Jason2866, 2022-09-22:
If i would have an idea i would have written. Maybe Theo

arendst, 2022-09-26:
Sorry can'y help. Have no Tuya stuff.

barbudor, 2022-09-26:
@pablozg Do I understand right that `if (!RtcTime.valid)` then it doesn't crash ?

Note that you probably don't get any information in the main screen because you seems to receive only the hearbeat (55AA00000000FF), no energy data.
May be we need to 1st solve that TuyaTime issue for the device to send you energy information

pablozg, 2022-09-26:
No, I think the problem is when the command ```TuyaSendCmd(TUYA_CMD_SET_TIME, payload_buffer, payload_len);``` is sent, because if I block the ntp server on the router, then it can't get the correct time and exits the function without sending the command.

When it is able to get the time correctly, it crashes exactly 60 seconds after boot (when the function is called).

The only command that works is the on/off relay.

barbudor, 2022-09-26:
I think your analysis is wrong
There should be nothing wrong with the command `TuyaSendCmd(TUYA_CMD_SET_TIME, payload_buffer, payload_len);`
TuyaTime is used by many devices, and the function `TuyaSendCmd(...)`is used for any mesasge sent to TuyaMCU including the relay on/off.

However I get the point that it gets wrong as soon as you get time, but not in this part of the code.
There are 2 other parts in the TuyaMCU code that are depending on having proper time (using `if (RtcTime.valid)`
Also as I pointed out before, when you don't have time, you don't seem to receive any Tuya messages regarding power consumption.

My current assumption is that as soon as you provide a proper time to TuyaMUC you start receiving power consumption messages and this is where the crash occurs.

As I don't own a Hijing device, I would suggest the following to validate my hypothesis:
- Set on Generic Module
- Configure Serial Bridge on TX and RX
- Configure `Baudrate 9600`
- use `SerialSend5 55AA00000000FF`
- you should get a `SerialReceived 55AA00000000FF` in response
- now send a TuyaTime message with `SerialSend5 55AA001C00080116091A0C0000026B`

And reports the logs
I don't expect a crash but some more messages to come in

Thanks

pablozg, 2022-09-27:
Thank you for your answer, after trying the commands you suggested and not getting response at all, I started thinking in a faulty module, so I desoldered the TYWE3S Board and replaced it with a standard esp8266, after that, I started getting voltage info, I haven't tested it under load yet, but I think it will be fine.

Sorry for all the inconvenience.

closed 2022-09-27 by pablozg