arendst/Tasmota #16461: Is Tasmota working properly on the new THR320D with DS18B20?

jmcucurella, 2022-09-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

We are using:
Equipment: THR320D
Tasmota version: 12.1.1.1
Flashing tools: embebed OTA URL firmware upgrade
Configuration as indicated on Github: {"NAME":"Sonoff THR320D","GPIO":[32,0,0,0,226,9280,0,0,0,321,0,576,320,9184,9216,9312,0,0,9313,9248,0,1,0,3840,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
Problem: Display is not working properly, shows apparently inner temperature, not from probe DS18B20.
Temperature using MQTT is the same as showed on display
MQTT apparently does not support Temperature Setpoint HIGH & LOW

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-09-08:
You need to configure the DS18B20 sensor on GPIO25 for it to work.

I'd much recommend you to not have that confusing and directly flawed ESP32 Temperature visble anywhere, this is already the default on latest development Tasmota, or you can disable it using `SetSensor127 0`.

jmcucurella, 2022-09-08:
> You need to configure the DS18B20 sensor on GPIO25 for it to work.
> 
> I'd much recommend you to not have that confusing and directly flawed ESP32 Temperature visble anywhere, this is already the default on latest development Tasmota, or you can disable it using `SetSensor127 0`.

Ok, now is working properly, temperature is ok, and ESP32 not apears.
You know somethig about setpoints ?

sfromis, 2022-09-08:
Tasmota does have an optional (with a custom build) thermostat feature, or you can have rules or Berry code for simple on/off control.