arendst/Tasmota #16462: Google Charts are not loading because of a commas!

sawacs, 2022-09-08:
Hello.
ESP8266 and ESP32 platform...
My Tasmota 12.1.0.1 compilation with Polish language set.
I added Google Chart to my simple script. Since it didn't appear on the website.
I started checking the script. But that's not the problem in the script! 
I have debug the html generated by Tasmot www server, in the place where it should be located:
    ..... https://www.gstatic.com/charts/loader.js .....  
I got:
   <script type="text/javascript" src="https://www,gstatic,com/charts/loader,js"></script>
   <script type="text/javascript">google,charts,load('current',{packages: ['corechart']}); </script>
Of course, all the required for compilation options were set. 
I suspect that it is the language-specific settings that are causing this problem. 
In Polish, the separator f/e for numbers is "comma" and not "dot" as in other languages (we use f/e 23,124).

[script_and_html_source.txt](https://github.com/arendst/Tasmota/files/9526571/script_and_html_source.txt)

closed 2022-09-08 by close-issue-app[bot]

barbudor, 2022-09-08:
@gemu2015

You shouldn't use `WSContentSend_PD(...)` to send those HTML strings but `WSContentSend_P(...)` to avoid conversion of `.` into `,`
Could you kindly fix it as you have a better view on the architecture ?
Thanks

gemu2015, 2022-09-08:
ok, ill fix that tomorrow

PR #16470 (2022-09-09): google chart fix
## Description:

**Related issue (if applicable):** fixes #16462
+ some other fixes

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [x] The code change is tested and works with Tasmota core ESP32 V.2.0.4.1
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreeme [... cut]