arendst/Tasmota #16469: Web UI slow response from Mijia Desk Lamp 1S (MJGJD02YL) Tasmota ESP32

outlying, 2022-09-08:
### PROBLEM DESCRIPTION
Response from web UI on my device is randomly slow

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

no idea what is it, never used those (I think)

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

22:02:56.581 CMD: Status 0
22:02:56.589 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"lightbar","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:02:56.598 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Vbat power on reset","Uptime":"2T22:50:29","StartupUTC":"2022-09-05T22:12:27","Sleep":50,"CfgHolder":4617,"BootCount":57,"BCResetTime":"2022-04-21T15:59:59","SaveCount":1823}}
22:02:56.611 RSL: STATUS2 = {"StatusFWR":{"Version":"12.0.1(tasmota)","BuildDateTime":"2022-06-17T14:40:29","Core":"2_0_3","SDK":"v4.4.1","CpuFrequency":80,"Hardware":"ESP32-S0WD-OEM rev.1","CR":"427/699"}}
22:02:56.617 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bananowo-2.4GHz","Recovery"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00006000","00000000"]}}
22:02:56.639 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1371,"Free":1856,"Heap":169,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9F9AC7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","000840D2","20200000","1434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,86","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
22:02:56.652 RSL: STATUS5 = {"StatusNET":{"Hostname":"lightbar-5825","IPAddress":"192.168.10.72","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.10.1","DNSServer2":"0.0.0.0","Mac":"7C:C2:94:30:16:C1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:02:56.660 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"LIGHTBAR_%06X","MqttClient":"LIGHTBAR_3016C1","MqttUser":"addon","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:02:56.671 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-09-08T21:02:56","Local":"2022-09-08T22:02:56","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:17","Sunset":"19:18"}}
22:02:56.684 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-09-08T22:02:56","ESP32":{"Temperature":71.1},"TempUnit":"C"}}
22:02:56.695 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-09-08T22:02:56","Uptime":"2T22:50:29","UptimeSec":255029,"Heap":172,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":101,"MqttCount":0,"Berry":{"HeapUsed":3,"Objects":40},"POWER":"ON","Dimmer":5,"Color":"000D","White":5,"CT":500,"Channel":[0,5],"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Bananowo-2.4GHz","BSSId":"3E:37:86:2A:CB:46","Channel":11,"Mode":"11n","RSSI":38,"Signal":-81,"LinkCount":4,"Downtime":"0T00:02:13"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

no idea how to 

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Install Tasmota on Mijia Desk Lamp 1S (MJGJD02YL) Tasmota ESP32, run WebUI 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Fast response, below at least 500ms

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Recording, the lamp is on top of the display
https://youtube.com/shorts/Bj13bas0mEk?feature=share

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

none

**(Please, remember to close the issue when the problem has been addressed)**

closed 2022-09-08 by outlying

outlying, 2022-09-09:
@sfromis my access point is less than 1,5m away from lamp, stright line, no walls

I used `SetOption127` the same moment and stability is superb now

s-hadinger, 2022-09-09:
As an alternative you can try `Sleep 5`

It comes the ESP less warm

sfromis, 2022-09-09:
Talking about a low physical distance is not relevant when you have **horrible** signal quality, more susceptible to temporary issues.