arendst/Tasmota #16472: Shelly 1PM power monitoring not working after first flash

AleXSR700, 2022-09-09:
### PROBLEM DESCRIPTION
Similar to issue https://github.com/arendst/Tasmota/issues/5592, after first flashing the current Tasmota 12.1.1 firmware to my Shelly 1PM, I see only incomplete power monitoring data.
[image]

I took the devices from the grid and waited a few minutes, but the situation did not change.

SerialLog is disabled.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Shelly 1PM
- [ ] Tasmota binary firmware version number used:  12.1.1
  - [ x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: https://github.com/yaourdt/mgos-to-tasmota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:00:43.085 MQT: stat/tasmota_E588CD/RESULT = {"NAME":"Shelly 1PM","GPIO":[320,0,0,0,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}
16:00:43.284 MQT: stat/tasmota_E588CD/RESULT = {"Module":{"0":"Shelly 1PM"}}
16:00:43.539 MQT: stat/tasmota_E588CD/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"192":"Switch_n1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"4736":"ADC Temp"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Everything working out of the box.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

AleXSR700, 2022-09-09:
?
Yours looks normal. You have Current and Voltage. That's what my 2.5 looks like also.

isabido, 2022-09-09:
I just realized that if I activate the relay output, I start to see measurements, but the device responds very slowly when moving through the WebUI menus, it must be a GPIO configuration problem or something similar.

[image]

AleXSR700, 2022-09-09:
Not sure these two issues are related.

barbudor, 2022-09-09:
Probably not
@isabido see https://tasmota.github.io/docs/Commands/#setoptions

@isabido no idea what is your issue yet. I would assume may be a wrong template as the Sh1PM is supposed to use serial to get power measurement but nothing in the template is about serial.
May be someone who owns one will pass by

isabido, 2022-09-09:
Thanks for the suggestion it worked! I have tried this option and it works correctly, even if I turn the relay on or off, I always have the measurement.

SetOption21 on

Another problem I see is that the independent energy counters do not appear for each of the two channels that ShellyEM has.

barbudor, 2022-09-09:
There's another setoption for that
https://tasmota.github.io/docs/Commands/#setoption129

Read the doc section regarding Power Monitoring everything is listed there
https://tasmota.github.io/docs/Commands/#power-monitoring

isabido, 2022-09-09:
Perfect!!! Now it's working, thank you very much for your help!

[image]

isabido, 2022-09-09:
If I activate the temperature, it marks a value that is not real and the RELAY is automatically deactivated when activated. It is not possible to calibrate the temperature more than 12º parameter TempOffset -12.0

[image]
[image]

isabido, 2022-09-09:
@barbudor Already a more in-depth topic, the ADE7953 has an important register to calibrate the Phase in the address 0x203 (PHCALx register)

This is very important to correct the phaseshift introduced by the CT (CLAMP) with very low PF loads.

In my case I have two air conditioning machines that consume P=20W in standby with a CosPhi of 0.1. The Shelly is not able to measure this active power correctly, it indicates -5W (negative) and a PF= -0.2

Would it be possible to access this register?

I enclose some images where you can compare values between the reading of my company's digital meter and the shellyEM

[image]
[image]

https://www.analog.com/media/en/technical-documentation/application-notes/AN-1118.pdf
[image]

https://www.analog.com/media/en/technical-documentation/application-notes/AN-639.pdf
[image]

isabido, 2022-09-10:
More information  PHCAL

https://www.analog.com/media/en/technical-documentation/data-sheets/ADE7953.pdf
[image]

barbudor, 2022-09-10:
@isabido it would be better that you move to a discussion related to the EM or open your own as your questions are nor related to original poster. Search on Discussions as all these have already been addressed as well as with calibration procedure documentation

closed 2022-10-11 by github-actions[bot]