arendst/Tasmota #16493: IR crashes on Sonoff Basic R2 v1.0

MyIgel, 2022-09-10:
### PROBLEM DESCRIPTION
When issuing the following command, the devices gets reset by a Soft WDT reset: `IRSend {"Protocol":"PANASONIC","Bits":16388,"Data": 0x05000401}`

I hope it's not a duplicate / config error in which case a hint on how to configure it properly would be nice :)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Basic R2 v1.0
- [x] Tasmota binary firmware version number used: 12.1.1(ir)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

23:42:11.881 CMD: Backlog Template; Module; GPIO 255
23:42:11.904 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
23:42:12.128 RSL: RESULT = {"Module":{"18":"Generic"}}
23:42:12.382 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"1056":"IRsend1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

23:43:05.334 CMD: Status 0
23:43:05.339 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_951A21","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
23:43:05.365 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T00:07:15","StartupUTC":"2022-09-10T22:35:50","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"2022-09-10T23:30:38","SaveCount":14,"SaveAddress":"F6000"}}
23:43:05.399 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.1(ir)","BuildDateTime":"2022-08-25T11:36:17","Boot":7,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"385/699"}}
23:43:05.417 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["InternetOfShit",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A00192800000000","00000080","00006000","00004000","00000000"]}}
23:43:05.446 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":627,"Free":376,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F1A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000","00000000"],"Drivers":"1,2,4,5,7,9,10,12,62","Sensors":"1"}}
23:43:05.475 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-951A21-6689","IPAddress":"192.168.16.199","Gateway":"192.168.16.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.16.1","DNSServer2":"0.0.0.0","Mac":"DC:4F:22:95:1A:21","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
23:43:05.500 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_951A21","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
23:43:05.520 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-09-10T22:43:05","Local":"2022-09-10T23:43:05","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:19","Sunset":"19:14"}}
23:43:05.537 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-09-10T23:43:05"}}
23:43:05.546 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-09-10T23:43:05","Uptime":"0T00:07:15","UptimeSec":435,"Vcc":3.595,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"InternetOfShit","BSSId":"04:18:D6:E2:29:04","Channel":13,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:03"}}}
23:43:05.582 RSL: STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["4000dd3b","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["40231924","40100a38","40100a5c","40230e7e","40230f4d","40230fa8","40230f4d","4023152c","402310f1","4023152c","402430e2","4020ddb4","40231820","4024c087","4020e945","40289427","40289427","40288d6f","4020efe2","4021d900","40288d60","40288d58","401009ce","4021db60","40211cf6","4021df2a","402583d0","402070e4","4024f4f3","4025835c","401010df"]}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here: (from serial)

23:35:48.682 CMD: IRSend {"Protocol":"PANASONIC","Bits":16388,"Data": 0x05000401}

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffff890 end: 3fffffc0 offset: 01a0
3ffffa30:  3ffffb40 000001b0 3ffffba8 40231924  
3ffffa40:  40100a38 000001b0 05000401 40100a5c  
3ffffa50:  007b0ef8 05000401 3ffffba8 40230e7e  
3ffffa60:  0000000c 0000000b 3ffffba8 40230f4d  
3ffffa70:  000001b0 00000d80 3ffffba8 000033be  
3ffffa80:  00000000 05000401 3ffffba8 40230fa8  
3ffffa90:  000001b0 000001b0 000001b0 00000510  
[... 93 lines cut ...]
~ld

00:00:00.001 HDW: ESP8285
00:00:00.050 CFG: Loaded from flash at F7, Count 13
00:00:00.055 FRC: Some settings have been reset (3)
00:00:00.061 Project tasmota - Tasmota Version 12.1.1(ir)-2_7_4_9(2022-08-25T11:36:17)
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
* Flash the Sonoff and connect to wifi
* Set the Type to Generic (18)
* Configure GPIO13 as IRsend 1
* Run `IRSend {"Protocol":"PANASONIC","Bits":16388,"Data": 0x05000401}`

### EXPECTED BEHAVIOUR
Sends the IR command but instead gets a software watchdog reset

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I connected the ir led before R16 which worked on 6.4.0.1 with a config that i obviously can't import to the current version,

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-09-11:
The sonoff Basic R2 v1.0 has a borderline power supply. Beside this the PCB heat design is bad too. Do not expect reliable function when added anything. Newer versions of Tasmota do have more features and draws a little bit more current, this is enough to bring this device in trouble.

s-hadinger, 2022-09-11:
I can reproduce the crash, it's not linked to the power supply

s-hadinger, 2022-09-11:
Hmmm. Your command tries to send `16388` bits or 2KB of IR data which is not supported. The erroneous value is passed on to IRremoteESP8266 and it triggers a WDT (taking too long I suppose).

Where did you get such a value?

s-hadinger, 2022-09-11:
Closing again as the value sent does not make sense

closed 2022-09-11 by s-hadinger