arendst/Tasmota #16505: Smart Meter OBIS ASCII creates false sensor values due to missing CRC calculations/checks

helander, 2022-09-13:
### PROBLEM DESCRIPTION
Sporadic erroneous metric values are created by the OBIS ASCII part of the Smart Meter interface. Every now and then characters are lost in transmission between the meter and the tasmota device. In these situations tasmota is likely to misinterpret the information and creates erroneous metric values. The line protocol includes a CRC that could (should) be used to make sure that no metrics are created from wrong input. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2022-09-16:
sml datagrams do not contain CRC for each value but only for a whole (sometimes very large) data set.
since the sml driver evaluates value per value this CRC is not supported

every protocol that has CRC per value like MODBUS  is checked and rejected if wrong.

wrong values are always due to poor hardware, either inappropriate or misplaced photo sensors, wrong pull-up values or unshielded cables.

serial transfer is extremely robust if signals are OK.

helander, 2022-09-16:
The CRC is there for a reason, pretending it is not required is not very helpful. In my case I get faulty messages approx a few times per hour. The root cause being the state of electric signals or hardware/ software errors in any of the peers does not matter, the CRC is able to help. Having to deal with the erroneus values upstream is far more complex.

gemu2015, 2022-09-16:
anyhow this driver is not prepared to deal with large blocks and then scan theses blocks for contained values.
it would require a complete rewrite of main parts to adapt for complete SML blocks.
if will definitely not do this.

helander, 2022-09-16:
I understand that it is a major rewrite and that it will not happen. I will have to find some other way to deal with this problem. 
I think there is a valid reason for why most SML protocols do not have a per metric CRC but a CRC on the whole bunch of metrics for a given meter sample. In case of error on a few of the metrics in a sample, the complete sample should most likely be discarded. Incomplete samples lead to problematic handling upstream.
My guess is that this was the understanding of the designers of the SML protocols and lead to the single CRC per sample message and not a CRC per metric.

closed 2022-09-16 by Jason2866