arendst/Tasmota #16510: Can't connect to Azure IoT Hub (ESP8266EX)

peterrre, 2022-09-14:
### PROBLEM DESCRIPTION
Hi,

i am trying to connect my device (ESP8266EX) to the Azure IoT Hub. I tried several compiling and configurations but still no success.

My current output from the cosole:

`06:15:52.088 MQT: Authenticating with an Azure IoT Hub Token
06:15:52.201 MQT: TLS connection error: 1
06:15:52.204 MQT: Connect failed to host1.azure-devices.net:8883, rc -2. Retry in 70 sec`

I followed the Azure IoT Hub guidline and enabled: 
`#ifndef USE_MQTT_TLS`
`#define USE_MQTT_TLS`
`#endif`
`#define USE_MQTT_AZURE_IOT`

`SetOption 132 1`

Inside MQTT configuration: CLIENT, USER, TOPIC and FULL TOPIC have the same value.

I also tried to create a second Azure IoT Hub with the preview features enabled (changes the ciphers) but still no success to connect.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): NOUS A1T
- [ ] Tasmota binary firmware version number used: 12.1.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: gitpod
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2022-09-14:
Have you seen #15763 ?

peterrre, 2022-09-14:
Yes, maybe I set the wrong TOPIC and FULLTOPIC details. I take the same details i use for Client and User 
For example:

`Host: xyz.azure-devices.net`
`Port: 8883`
`Client: DeviceID_XYZ`
`User: DeviceID_XYZ`
`TOPIC: DeviceID_XYZ`
`FULLTOPIC: DeviceID_XYZ`

is this correct?

sfromis, 2022-09-14:
Fulltopic is bad. You should leave it as default, or possibly *add* something to (or rearrange) the existing `%prefix%/%topic%/`, not replace it. Don't specifically know if this affects your situation.

Padu98, 2022-09-22:
Hi @peterrre , the error (1) means "invalid fingerprint". Try to reset it via the console: mqttfingerprint1 reset
I had the same issue when switching my iothub/device. The old fingerprint has to be reset. Hope that helps!

pexeum90, 2022-10-15:
Have the same issue. Seems like, if the TLS Fingerprint is not generated. Are there any updates or workarounds in this topic?

AaronAutomation, 2022-11-24:
Exact same issue as above for me also. I pinged the Tasmota Discord and didn't have any luck there either. 
[image]

s-hadinger, 2022-11-24:
It looks like TLS works, but credentials are rejected. Can you make sure they are not truncated? Also does Azure require a specific ALPN?

paulobriennz, 2022-11-25:
same issues here, SSL -1002

paulobriennz, 2022-11-26:
as per this thread https://github.com/arendst/Tasmota/issues/14624

If you create an IOT Hub in Central US, turn on preview mode, then you can connect as you would expect above.  Something to do with the SSL the Tasmota handles?

closed 2022-12-26 by github-actions[bot]

martinhenrysson, 2025-02-14:
I'm having the exact same issue as @AaronAutomation. Did you find a solution?

> MQT: TLS Connection Error: 0 
> MQT: Connect failed to *.azure-devices.net:8883, rc -2. Retry in n sec

I also followed the Tasmota Azure IoT Hub documentation https://tasmota.github.io/docs/Azure-IoT-Hub/ and built custom firmware (tasmota32solo1) with:
```
#ifndef USE_MQTT_TLS
#define USE_MQTT_TLS
#endif
#define USE_MQTT_AZURE_IOT
```

`SetOption132 ON`

Using a preview version of Azure IoT Hub is not an option as it is not usable for production.

martinhenrysson, 2025-02-15:
@arendst Should I open another issue?

s-hadinger, 2025-02-16:
Not necessary. You need to find someone interested in solving this or adding the necessary features