arendst/Tasmota #16556: Tasmota web server is not accesible until running a network scan

Android-X13, 2022-09-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have a custom Termux task on my phone that sends specific web requests to an IR controller which is flashed with Tasmota. However 1 out of 5 times it fails to execute because Tasmota's web server apparently is not accesible for some reason.

If I repeat the task, it keeps failing, but I have noticed that I can successfully "wake up" the web server by running an nmap scan on the local network. Even the network scan usually doesn't find the device at the first try, so I have created another Termux task which runs the nmap scan in a loop until the Tasmota device responds. When I get a hit and the task exits successfuly, I can then continue sending my web requests normally until the next time.

Can anyone explain why this happens (as well as why nmap is able to "wake up" the device) and how it can be mitigated?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): YTF IR Bridge
- [x] Tasmota binary firmware version number used: 11.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Tuya-Convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
8:20:10.223 RSL: RESULT = {"NAME":"YTF IR Controller","GPIO":[1,1,1,1,320,1088,0,0,0,32,1056,0,0,0],"FLAG":0,"BASE":62}
18:20:10.452 RSL: RESULT = {"Module":{"0":"YTF IR Controller"}}
18:20:10.706 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"320":"Led_i1"},"GPIO5":{"1088":"IRrecv"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"32":"Button1"},"GPIO14":{"1056":"IRsend"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:22:04.628 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E17B94","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
18:22:04.633 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-ir.bin.gz","RestartReason":"Power On","Uptime":"12T19:05:48","StartupUTC":"2022-09-04T20:16:16","Sleep":50,"CfgHolder":4617,"BootCount":37,"BCResetTime":"2022-04-24T21:42:38","SaveCount":2299,"SaveAddress":"F9000"}}
18:22:04.637 RSL: STATUS2 = {"StatusFWR":{"Version":"11.1.0(ir)","BuildDateTime":"2022-04-13T06:40:31","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"378/699"}}
18:22:04.640 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID_1","SSID_2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000180","00006000","00004000"]}}
18:22:04.648 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":631,"Free":372,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","0F5A0582","04000001","00000003","00000000","00000000","00000000","00000000","04000000"],"Drivers":"1,2,4,5,7,9,10,12,62","Sensors":"1"}}
18:22:04.652 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-E17B94-7060","IPAddress":"192.168.1.40","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"BC:DD:C2:E1:6B:54","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:22:04.656 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E17B94","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:22:04.661 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-09-17T15:22:04","Local":"2022-09-17T18:22:04","StartDST":"2022-03-27T03:00:00","EndDST":"2022-10-30T04:00:00","Timezone":99,"Sunrise":"08:29","Sunset":"20:59"}}
18:22:04.665 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-09-17T18:22:04"}}
18:22:04.671 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-09-17T18:22:04","Uptime":"12T19:05:48","UptimeSec":1105548,"Vcc":3.521,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":0,"Wifi":{"AP":1,"SSId":"SSID_1","BSSId":"08:AA:89:32:ED:40","Channel":5,"Mode":"11n","RSSI":92,"Signal":-54,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-09-17:
May be try to play with SetOption41

Android-X13, 2022-09-17:
Thanks @barbudor I'll try it. I see that the default is 60 but I'm sure the device is not accessible for more than that unless I run a network scan, so I still not get why nmap manages to wake up the device..

I'm closing this issue because it was moved to [Discussions](https://github.com/arendst/Tasmota/discussions/16555)

closed 2022-09-17 by Android-X13