arendst/Tasmota #16558: Zigbee Configure Reporting

Stefano2021, 2022-09-17:
### PROBLEM DESCRIPTION
Hi, I try to send a zigbee Configure Reporting Command but Tasmota don't translate the 
"Reportable change" field.

I using
Program Version      12.1.1.1(zbbridge)
Core/SDK Version     2_7_4_9/2.2.2-dev(38a443e)

00:07:04.595 CMD: **ZbSend {"Device":"0x8584","Config":{"Temperature":{"MinInterval":30,"MaxInterval":600,"ReportableChange":0.5}}}**
00:07:04.596 SRC: WebConsole from 192.185.159.100
00:07:04.598 CMD: Grp 0, Cmd 'ZBSEND', Idx 1, Len 104, Pld -99, Data '{"Device":"0x8584","Config":{"Temperature":{"MinInterval":30,"MaxInterval":600,"ReportableChange":0.5}}}'
00:07:04.601 ZIG: guessing endpoint 10
00:07:04.606 ZigbeeZCLSend device: 0x8584, endpoint:10, cluster:0x0402, cmd:0x06, send:"000000291E0058020000"
00:07:04.607 ZIG: ZbEZSPSend 340000848504010204010A4001000004010D100406000000291E005802**0000**
00:07:04.608 ZIG: adding packet to_send, to_ack:7, to_send:7, to_end:7
00:07:04.610 RSL: RESULT = {"ZbSend":"Done"}
00:07:04.700 ZIG: Something to_send, to_ack:7, to_send:7, to_end:0
00:07:04.701 ZIG: ZbEZSPSentRaw 752F0001340000848504010204010A4001000004010D100406000000291E0058020000

ZigBee Cluster Library Frame, Command: Configure Reporting, Seq: 4
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 4
    Command: Configure Reporting (0x06)
    Reporting Configuration Record
        Direction: Reported (0x00)
        Attribute: Measured Value (0x0000)
        Data Type: 16-Bit Signed Integer (0x29)
        Measured Value: 0,00 [°C]
    Minimum Interval: 30
    Maximum Interval: 600

This is the right command that should be send:
**ZbEZSPSend 340000848504010204010A4001000004010D100406000000291E0058023200**

Thanks for the great work
CIAO From Italy

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2022-09-17:
Thanks for the very detailed bug report. I will look at it shortly

Stefano2021, 2022-09-17:
Thank you for your quick answer.
Can I ask you if it is possible to implement the ability to disable the reporting of a attribute.

[image]

Thanks again
Stefano.

s-hadinger, 2022-09-18:
Thanks, it is now fixed in the `development` branch.

To disable the attribute reporting, simply set `MaxInterval` to `-1`:

`ZbSend {"Device":"0x8584","Config":{"Temperature":{"MinInterval":30,"MaxInterval":-1,"ReportableChange":0}}}`

sends:

`ZigbeeZCLSend device: 0x8584, endpoint:1, cluster:0x0402, cmd:0x06, send:"000000291E00FFFF0000"`

PR #16567 (2022-09-18): Fix zigbee attribute writes and configuration
## Description:

Fix attribute reporting and attribute write bug, that would invert the multiplier/divider. The bug was introduced in #16340

**Related issue (if applicable):** fixes #16558

This PR also allows to pass "zigbee invalid value" by passing `null` instead of a value.
Example :
```ZbSend {"Device":"0x6EC0","Config":{"Temperature":{"MinInterval":30,"MaxInterval":-1,"ReportableChange":null}}}```
sends 
```ZigbeeZCLSend device: 0x6EC0, endpoint:1, cluster:0x0402, cmd:0x06, send:"000000291E00FFFF0080"```

Also fixes a rounding error when mul [... cut]
closed 2022-09-18 by s-hadinger