arendst/Tasmota #16632: Shelly 1/1PM with Shelly temp Addon & DHT22/AM2302 or similar sensors

thwandi, 2022-09-23:
### PROBLEM DESCRIPTION
Trying to use a Shelly1 with shelly Temperature Addon (Optical isolator circuit) & connect DHT22/AM2302 to it.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 1 with Temperature Addon
- [x] Tasmota binary firmware version number used: V11.0.0
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESPhome flasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:10:29.791 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
12:10:30.008 RSL: RESULT = {"Module":{"0":"Generic"}}
12:10:30.262 RSL: RESULT = {"GPIO0":{"1216":"AM2301"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"192":"Switch_n1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: none

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:12:01.097 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_6A3B83","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
12:12:01.102 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:10:46","StartupUTC":"2022-09-23T11:01:15","Sleep":50,"CfgHolder":4617,"BootCount":26,"BCResetTime":"2022-09-23T10:35:59","SaveCount":68,"SaveAddress":"F8000"}}
12:12:01.107 RSL: STATUS2 = {"StatusFWR":{"Version":"11.0.0(TasmoCompiler-esp8266generic)","BuildDateTime":"2022-05-03T14:24:46","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"335/699"}}
12:12:01.112 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["smarthome",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A190000000000","00000080","00006000","00004000"]}}
12:12:01.124 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":703,"Free":300,"Heap":22,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"15605E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9FD065D7","04260001","600207C7","180C0000","D032F981","0009200C","00001000","00000040"],"Drivers":"1,2,3,4,6,7,8,9,10,11,12,18,19,22,24,26,27,29,30,35,37,45","Sensors":"1,2,3,4,5,6,7,8,9,14,26,27,38,39,46,47,55,58,59,63,65,72,73,74,88"}}
12:12:01.131 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-6A3B83-7043","IPAddress":"10.90.1.116","Gateway":"10.90.0.1","Subnetmask":"255.255.240.0","DNSServer1":"10.90.0.30","DNSServer2":"0.0.0.0","Mac":"34:94:54:6A:3B:83","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
12:12:01.134 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6A3B83","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
12:12:01.140 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-09-23T11:12:01","Local":"2022-09-23T12:12:01","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:38","Sunset":"18:46"}}
12:12:01.147 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-09-23T12:12:01","Switch1":"OFF","AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}}
12:12:01.152 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-09-23T12:12:01","Uptime":"0T00:10:46","UptimeSec":646,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"smarthome","BSSId":"06:EC:DA:40:9D:FB","Channel":5,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Possibility to use the shelly temp addon within tasmota to connect DHT22 or similar sensors.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I read that DS18B20 needs to be selected with both Input and Output in Tasmota while using it on a shelly 1/1PM with shelly temp addon, AM2301/AM2302/DHT22 is not available in the dropdown with both In & Out.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-09-23:
Not sure that this will work, but as support for DHT11 and DHT22 is shared in the same driver, it appears that configuring a `DHT11_o` second pin may have effect also for DHT22.

thwandi, 2022-09-23:
> Not sure that this will work, but as support for DHT11 and DHT22 is shared in the same driver, it appears that configuring a `DHT11_o` second pin may have effect also for DHT22.

That seems to work, i dropped the DHT11_o on the second data pin and it is now showing something. I will check the readings for plausability soon, but it looked not too bad. Thanks a lot for the quick hint! :)

closed 2022-09-23 by thwandi

chivas03, 2022-11-20:
Could you get plausible values? I have similar challanges. Can you share your last configuration/template of the Shelly?

notacoderjustauser, 2022-11-26:
I got some values building this template:

`
{"NAME":"Shelly Zuluft","GPIO":[1280,0,0,1216,82,134,0,0,0,0,0,0,21,0],"FLAG":2,"BASE":18}`

This post lead me to the right direction:
[https://github.com/arendst/Tasmota/issues/7469#issuecomment-575664033](url)

unfortunately i have no switch, no internal temperature, no power measurement now and dont know how to configure this.

chivas03, 2022-11-27:
> I got some values building this template:
> 
> `{"NAME":"Shelly Zuluft","GPIO":[1280,0,0,1216,192,2720,0,0,0,0,0,224,0,4736],"FLAG":0,"BASE":18}`
> 
> This post lead me to the right direction: [#7469 (comment)](https://github.com/arendst/Tasmota/issues/7469#issuecomment-575664033)
> 
> edit: updated template after some try and errors

Thanks for sharing! This was very helpful to me :-) It works...that configuration I never tried before.
I just changed the GPIO for my project as I need the relay.
[image]

feuh, 2023-03-11:
Hello,

I tried it with your Template but I only get "null" as value.
Do I have to change anything at the FW before?