arendst/Tasmota #16663: ELSE stagment not work

zygfrydos, 2022-09-26:
Hi, I have tasmota version 2_7_4_9/2.2.2-dev(38a443e) with compiled USE_EXPRESSION and SUPPORT_IF_STATEMENT.
I first time try If stagment in tasmota so my examples was tested only in console
This work without problem:
`IF(POWER1#STATE==OFF) POWER1 ON ENDIF`
Also this:
`IF(POWER1#STATE==ON) POWER1 OFF ENDIF`
But this not work:
`IF(POWER1#STATE==ON) POWER1 OFF ELSE POWER1 ON ENDIF`
I also try with space between:
`IF (POWER1#STATE==ON) POWER1 OFF ELSE POWER1 ON ENDIF`
But not work

closed 2022-09-26 by close-issue-app[bot]

sfromis, 2022-09-26:
What do you mean by "work fine"? Did a small reductio ad absurdum test:
```
on event#test do IF(POWER1#STATE==OFF) var1 match off ENDIF endon
on event#test do IF(POWER1#STATE==ON) var1 match on ENDIF endon

13:38:10.893 CMD: power
13:38:10.899 MQT: xxxxxxxx/stat/POWER = {"POWER":"OFF"}
13:38:10.901 MQT: xxxxxxxx/stat/POWER = OFF
13:38:21.406 CMD: event test
13:38:21.413 MQT: xxxxxxxx/stat/EVENT = {"Event":"Done"}
13:38:21.452 RUL: EVENT#TEST performs "IF(POWER1#STATE==OFF) var1 match off ENDIF"
13:38:21.462 MQT: xxxxxxxx/stat/IF = {"If":"Done"}
13:38:21.475 RUL: EVENT#TEST performs "IF(POWER1#STATE==ON) var1 match on ENDIF"
13:38:21.484 MQT: xxxxxxxx/stat/IF = {"If":"Done"}
13:38:21.695 MQT: xxxxxxxx/stat/VAR = {"Var1":"match on"}
13:38:21.945 MQT: xxxxxxxx/stat/VAR = {"Var1":"match off"}
```
When the condition does not work, it is obvious that the ELSE cannot work.

sfromis, 2022-09-26:
Trigger expressions with `#` only work in the trigger part of a rule, between on/do, not between do/endon. For your case, you might want to use something like:
`on Power1#State do if (%value%==0) Power1 1 else Power1 0 endif endon`
Obviously, this assumes that the if/else/endif were the whole rules.

sfromis, 2022-09-26:
Trigger expressions with `#` only work in the trigger part of a rule, between on/do, not between do/endon. For your case, you might want to use something like:
`on Power1#State do if (%value%==0) Power1 1 else Power1 0 endif endon`
Obviously, this assumes that the if/else/endif were the whole rules. And yes, ELSE works ;-)