arendst/Tasmota #16668: Smart Meter Interface - SML binary mode  Incorrect values ​​from time to time

tomy42, 2022-09-26:
### PROBLEM DESCRIPTION
The Smartmeter interface transmits incorrect values at irregular intervals.

I connected two reading heads for smart meters to a WeMOS D1. A SmartMeter uses the OBIS ASCII protocol and works correctly. The second smart meter uses the SML binary protocol. Incorrect values ​​are transmitted here at irregular intervals.

-- Example 1
Sep 26 13:14:56 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:14:55","WP":{"Energie_Tarif2":**2532259.76940**}}
Sep 26 13:14:57 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:14:56","WP":{"Energie_Tarif2":**2531735.48140**}}
Sep 26 13:14:58 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:14:57","WP":{"Energie_Tarif2":**2532259.76940**}}
-- Example 2
Sep 25 09:21:11 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T08:21:10","WP":{"Energie_Tarif2":**2515185.25170**}}
Sep 25 09:21:12 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T08:21:11","WP":{"Energie_Tarif2":**2514910.00050**}}
Sep 25 09:21:13 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T08:21:12","WP":{"Energie_Tarif2":**2515185.25170**}}

The difference between the values ​​is always exactly 524.288 which corresponds to HEX 80000.
It could indicate a problem with signed numbers?

I could also observe incorrect values ​​for other measured values.
-- Example 3
Sep 25 14:28:12 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T13:28:10","HA":{"Spannung_L2":**232.50**}}
Sep 25 14:28:12 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T13:28:11","WP":{"Spannung_L2":**999999.00**}}
Sep 25 14:28:13 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T13:28:11","HA":{"Spannung_L2":**231.50**}}

-- Example 4
Sep 26 13:13:26 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:13:25","WP":{"Leistung_L1":**507.52**}}
Sep 26 13:13:27 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:13:26","WP":{"Leistung_L1":**0.00**}}
Sep 26 13:13:28 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:13:27","WP":{"Leistung_L1":**507.90**}}

There is always only one incorrect value and then the correct values ​​follow again. The problem usually occurs after a few days

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WeMOS D1 Mini
- [ ] Tasmota binary firmware version number used: 12.1.1
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
stat/Power_Meter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
The problem usually reoccurs randomly a few days later.

I could also observe this behavior with older versions of Tasmota 8.x,  9.x

### EXPECTED BEHAVIOUR

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Script file:
>D
>B
=>sensor53 r
>M 2
1,=h<h1>Haus</h1>
; Device: eBZ DD3 2R06 ODZ1
; protocol is D0 OBIS ASCII
; 9600@7E1 for OP-type devices, 9600@8N1 for SM-type devices
+1,3,o,0,9600,HA
; Zählerstand zu +A, tariflos, 
; Zählerstände Auflösung 10 µW*h (6 Vorkomma- und 8 Nachkommastellen)
1,1-0:1.8.0*255(@0.001,Verbrauch Summe,Wh,Energie_Summe,21
; Summe der Momentan-Leistungen in allen Phasen, Auflösung 0,01W (5 Vorkomma- und 2 Nachkommastellen)
1,=h<hr/>
1,1-0:16.7.0*255(@1,Leistung,W,Leistung_Sum,18
; Momentane Leistung in Phase Lx, Auflösung 0,01W (5 Vorkomma- und 2 Nachkommastellen)
1,1-0:36.7.0*255(@1,Leistung L1,W,Leistung_L1,18
1,1-0:56.7.0*255(@1,Leistung L2,W,Leistung_L2,18
1,1-0:76.7.0*255(@1,Leistung L3,W,Leistung_L3,18
1,=h<hr/>
; Spannung in Phase Lx, Auflösung 0,1V (nur über MSB)
1,1-0:32.7.0*255(@1,Spannung L1,V,Spannung_L1,18
1,1-0:52.7.0*255(@1,Spannung L2,V,Spannung_L2,18
1,1-0:72.7.0*255(@1,Spannung L3,V,Spannung_L3,18
1,=h<hr/>
; Geräte-Identifikation, Nach DIN 43863-5 
1,1-0:96.1.0*255(@#),Identifikation,,96_1_0,0
; Heizung
2,=h<h1>Heizung</h1>
+2,14,s,0,9600,WP
2,77070100010801ff@1,Verbrauch Tarif_1,Wh,Energie_Tarif1,21
2,77070100010802ff@1,Verbrauch Tarif_2,Wh,Energie_Tarif2,21
2,77070100010800ff@1,Verbrauch Summe,Wh,Energie_Summe,21
2,=h<hr/>
2,77070100200700ff@1,Spannung L1,V,Spannung_L1,18
2,77070100340700ff@1,Spannung L2,V,Spannung_L2,18
2,77070100480700ff@1,Spannung L3,V,Spannung_L3,18
2,=h<hr/>
2,77070100100700ff@1,Leistung,W,Leistung_Sum,18
2,77070100240700ff@1,Leistung L1,W,Leistung_L1,18
2,77070100380700ff@1,Leistung L2,W,Leistung_L2,18
2,770701004c0700ff@1,Leistung L3,W,Leistung_L3,18
#

This was noticed because the values ​​are displayed using Grafana and the consumption values ​​show the consumption per hour. So it happens that from time to time consumption of several MW per hour is displayed.

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2022-09-28:
try with a median filter (16 means use 5 point median filter)

+2,14,s,16,9600,WP

tomy42, 2022-09-28:
Thanks, that might really solve the problem. Since the difference in several cases is exactly 0x80000, I would assume that it could be an error in the code and not an error in transmission. As far as I know, the SML protocol should have a checksum.

I'll test the suggested settings first, but since the error occurs several days later, I have to wait a while to see if the problem still occurs.

thanks for the tip

deesnow, 2022-10-04:
> ### PROBLEM DESCRIPTION
> The Smartmeter interface transmits incorrect values at irregular intervals.
> 
> I connected two reading heads for smart meters to a WeMOS D1. A SmartMeter uses the OBIS ASCII protocol and works correctly. The second smart meter uses the SML binary protocol. Incorrect values ​​are transmitted here at irregular intervals.
> 
> -- Example 1 Sep 26 13:14:56 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:14:55","WP":{"Energie_Tarif2":**2532259.76940**}} Sep 26 13:14:57 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:14:56","WP":{"Energie_Tarif2":**2531735.48140**}} Sep 26 13:14:58 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:14:57","WP":{"Energie_Tarif2":**2532259.76940**}} -- Example 2 Sep 25 09:21:11 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T08:21:10","WP":{"Energie_Tarif2":**2515185.25170**}} Sep 25 09:21:12 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T08:21:11","WP":{"Energie_Tarif2":**2514910.00050**}} Sep 25 09:21:13 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T08:21:12","WP":{"Energie_Tarif2":**2515185.25170**}}
> 
> The difference between the values ​​is always exactly 524.288 which corresponds to HEX 80000. It could indicate a problem with signed numbers?
> 
> I could also observe incorrect values ​​for other measured values. -- Example 3 Sep 25 14:28:12 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T13:28:10","HA":{"Spannung_L2":**232.50**}} Sep 25 14:28:12 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T13:28:11","WP":{"Spannung_L2":**999999.00**}} Sep 25 14:28:13 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-25T13:28:11","HA":{"Spannung_L2":**231.50**}}
> 
> -- Example 4 Sep 26 13:13:26 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:13:25","WP":{"Leistung_L1":**507.52**}} Sep 26 13:13:27 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:13:26","WP":{"Leistung_L1":**0.00**}} Sep 26 13:13:28 Power-Meter-01 ESP-MQT: tele/Power_Meter/SENSOR = {"Time":"2022-09-26T12:13:27","WP":{"Leistung_L1":**507.90**}}
> 
> There is always only one incorrect value and then the correct values ​​follow again. The problem usually occurs after a few days
> 
> ### REQUESTED INFORMATION
> _Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_
> 
> * [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
> * [x]  Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
> * [x]  Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
> * [x]  Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
> * [ ]  Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
> * [x]  Device used (e.g., Sonoff Basic): WeMOS D1 Mini
> * [ ]  Tasmota binary firmware version number used: 12.1.1
>   
>   * [ ]  Pre-compiled
>   * [x]  Self-compiled
> * [x]  Flashing tools used: OTA
> * [x]  Provide the output of command: `Backlog Template; Module; GPIO 255`:
> 
> ```lua
>   Configuration output here:
> stat/Power_Meter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
> ```
> 
> * [ ]  If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
> 
> ```lua
>   Rules output here:
> ```
> 
> * [ ]  Provide the output of this command: `Status 0`:
> 
> ```lua
>   STATUS 0 output here:
> ```
> 
> * [ ]  Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
> 
> ```lua
>   Console output here:
> ```
> 
> ### TO REPRODUCE
> The problem usually reoccurs randomly a few days later.
> 
> I could also observe this behavior with older versions of Tasmota 8.x, 9.x
> 
> ### EXPECTED BEHAVIOUR
> ### SCREENSHOTS
> _If applicable, add screenshots to help explain your problem._
> 
> ### ADDITIONAL CONTEXT
> Script file:
> 
> > D
> > B
> > =>sensor53 r
> > M 2
[... 25 lines cut ...]
> > +2,14,s,0,9600,WP
> > 2,77070100010801ff@1,Verbrauch Tarif_1,Wh,Energie_Tarif1,21
> > 2,77070100010802ff@1,Verbrauch Tarif_2,Wh,Energie_Tarif2,21
> > 2,77070100010800ff@1,Verbrauch Summe,Wh,Energie_Summe,21
> > 2,=h
> > 2,77070100200700ff@1,Spannung L1,V,Spannung_L1,18
> > 2,77070100340700ff@1,Spannung L2,V,Spannung_L2,18
> > 2,77070100480700ff@1,Spannung L3,V,Spannung_L3,18
> > 2,=h
> > 2,77070100100700ff@1,Leistung,W,Leistung_Sum,18
> > 2,77070100240700ff@1,Leistung L1,W,Leistung_L1,18
> > 2,77070100380700ff@1,Leistung L2,W,Leistung_L2,18
> > 2,770701004c0700ff@1,Leistung L3,W,Leistung_L3,18
> 
> This was noticed because the values ​​are displayed using Grafana and the consumption values ​​show the consumption per hour. So it happens that from time to time consumption of several MW per hour is displayed.
> 
> **(Please, remember to close the issue when the problem has been addressed)**

@tomy42
what type of smart meter you have? I tried to use script published for [Sanxing 631](https://tasmota.github.io/docs/Smart-Meter-Interface/#sanxing-sx6x1-sxxu1x-ascii-obis) but got always just {"Command":"Unknown"} for every line received on serial. 

````
17:27:08.723 CMD: 1-0:8.7.0(00.142*kvar)
17:27:08.727 MQT: stat/tasmota_5A9AE8/RESULT = {"Command":"Unknown"}
```

tomy42, 2022-10-07:
> I have the EasyMeter Q3A, the dump mode helped me to see which values are being sent.

tomy42, 2022-10-07:
> 
It looks like the median filter helped, since I activated it (9 days ago) no more wrong values have been sent via mqtt.

The solution would be good for me, should it be pursued further and is further debug information required?

closed 2022-11-06 by github-actions[bot]

OLFDB, 2023-04-23:
I'm facing the same issue with a Logarex LK13BE606739 using the script from the Tasmota documentation.
I have now enabled the median filter as supposed above. 

My questions is, why are wrong values delivered as SML has a checksum? Is the meter sending those values with correct checksum, or is there an issue in Tasmota verifying the checksum?

gemu2015, 2023-04-24:
SML has a checksum only over the complete block of data only, which may be very large for some meters.

the implemented driver decodes one value after the other by pattern matching, thus no checksum checked

other protocols like modbus have checksum for each value and that is verified by the driver.

serial errors indicate poor data quality.  normally if IR receiver hardware is properly designed and sensor is positioned  well there are nearly no errors.