arendst/Tasmota #16669: Tasmota crashing on Shelly when asked to power on

r0bfult, 2022-09-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have a Shelly 1, it is configured with SwitchMode 15 such that a power change event only generates a MQQT event.
Home assistant is configured to action the MQQT event, then after a specific delay switch the relay.
When the Switch goes on, following a delay, the relay is turned on by Home Assistant
When the Switch goes off, following a delay, Tasmota crashes when it should turn off the relay

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Shelly1
- [ ] Tasmota binary firmware version number used: 12.1.1
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

22:19:44.566 MQT: stat/tasmota_7777DA/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
22:19:44.780 MQT: stat/tasmota_7777DA/RESULT = {"Module":{"46":"Shelly 1"}}
22:19:45.033 MQT: stat/tasmota_7777DA/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"192":"Switch_n1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
NA

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

22:20:16.897 MQT: stat/tasmota_7777DA/STATUS = {"Status":{"Module":46,"DeviceName":"Nest","FriendlyName":["Nest"],"Topic":"tasmota_7777DA","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[15,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
22:20:16.907 MQT: stat/tasmota_7777DA/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T07:58:24","StartupUTC":"2022-09-26T13:21:52","Sleep":50,"CfgHolder":4617,"BootCount":39,"BCResetTime":"2022-01-17T21:04:04","SaveCount":946,"SaveAddress":"FB000"}}
22:20:16.912 MQT: stat/tasmota_7777DA/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:33:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"335/699"}}
22:20:16.920 MQT: stat/tasmota_7777DA/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":4,"LogHost":"192.168.2.1","LogPort":514,"SSId":["ha",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
22:20:16.936 MQT: stat/tasmota_7777DA/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":24,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"15701C","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
22:20:16.946 MQT: stat/tasmota_7777DA/STATUS5 = {"StatusNET":{"Hostname":"nest","IPAddress":"192.168.2.232","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"C4:5B:BE:77:77:DA","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:20:16.953 MQT: stat/tasmota_7777DA/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_7777DA","MqttUser":"mqqt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:20:16.962 MQT: stat/tasmota_7777DA/STATUS7 = {"StatusTIM":{"UTC":"2022-09-26T21:20:16","Local":"2022-09-26T22:20:16","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:42","Sunset":"18:39"}}
22:20:16.973 MQT: stat/tasmota_7777DA/STATUS10 = {"StatusSNS":{"Time":"2022-09-26T22:20:16","Switch1":"OFF"}}
22:20:16.980 MQT: stat/tasmota_7777DA/STATUS11 = {"StatusSTS":{"Time":"2022-09-26T22:20:16","Uptime":"0T07:58:24","UptimeSec":28704,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"ha","BSSId":"FA:92:BF:61:A9:85","Channel":6,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:03"}}}
22:20:16.996 MQT: stat/tasmota_7777DA/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["4027294b","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["401026f0","40102263","402701e9","4026fb3b","40000f49","40000f49","40000e19","40105aa9","40105aaf","4010000d","4026ea18","4026e9c9","40100a7c","40102b21","401026f0","4010598b","40104b66","40104fd7","40104898","401037fb","40104a9f","4000050c","40103964","4000e27d","4000d53a","4000050c","4010364c","402548d0","402548b5","402542c8","402542c8"]}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Delay here was 60 seconds, you can see crash 63 seconds after power off

Sep 26 14:20:52 nest ESP-MQT: tele/tasmota_7777DA/SENSOR = {"Time":"2022-09-26T14:20:51","Switch1":"OFF"}
Sep 26 14:20:53 nest ESP-WIF: Sending Gratuitous ARP
Sep 26 14:21:03 nest ESP-WIF: Checking connection...
Sep 26 14:21:23 nest ESP-WIF: Checking connection...
Sep 26 14:21:43 nest ESP-WIF: Checking connection...
Sep 26 14:21:56 nest ESP-CFG: Loaded from flash at F5, Count 944
Sep 26 14:21:56 nest ESP-QPC: Count 1
Sep 26 14:21:56 nest ESP-ROT: Mode 1
Sep 26 14:21:56 nest ESP-SRC: Restart
Sep 26 14:21:56 nest ESP-Project tasmota - Nest Version 12.1.1(tasmota)-2_7_4_9(2022-08-25T11:33:55)
Sep 26 14:21:56 nest ESP-WIF: Checking connection...
Sep 26 14:21:56 nest ESP-WIF: Attempting connection...
Sep 26 14:21:56 nest ESP-WIF: Connecting to AP1 ha Channel 6 BSSId FA:92:BF:61:A9:85 in mode 11n as nest...
[... 14 lines cut ...]
Sep 26 14:21:57 nest ESP-MQT: tele/tasmota_7777DA/INFO1 = {"Info1":{"Module":"Shelly 1","Version":"12.1.1(tasmota)","FallbackTopic":"cmnd/DVES_7777DA_fb/","GroupTopic":"cmnd/tasmotas/"}}
Sep 26 14:21:57 nest ESP-MQT: tele/tasmota_7777DA/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"nest","IPAddress":"192.168.2.232"}}
Sep 26 14:21:57 nest ESP-MQT: tele/tasmota_7777DA/INFO3 = {"Info3":{"RestartReason":{"Exception":0,"Reason":"Exception","EPC":["4027294b","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["401026f0","40102263","402701e9","4026fb3b","40000f49","40000f49","40000e19","40105aa9","40105aaf","4010000d","4026ea18","4026e9c9","40100a7c","40102b21","401026f0","4010598b","40104b66","40104fd7","40104898","401037fb","40104a9f","4000050c","40103964","4000e27d","4000d53a","4000050c","4010364c","402548d0","402548b5","402542c8","402542c8"]},"BootCount":39}}
Sep 26 14:21:57 nest ESP-MQT: stat/tasmota_7777DA/RESULT = {"POWER":"ON"}
Sep 26 14:21:57 nest ESP-MQT: stat/tasmota_7777DA/POWER = ON

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Configure switchmode to 15

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Device should not crash

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-09-27:
I cannot replicate using a Sonoff PowR320D. Considering the exception 0 it may mean a power issue.

Try to downgrade to an aerlier version of tasmota and see what happens.

Jason2866, 2022-10-29:
@r0bfult Can we close?

Jason2866, 2022-11-17:
Closing since no feedback. Feel free to reopen if the problem is still there.

closed 2022-11-17 by Jason2866