arendst/Tasmota #16691: Solax does not work in compiled tasmota.bin

amuelkx, 2022-09-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The lines 
#ifndef USE_SOLAX_X1                          // Enable Solax X1 series log info support (+4k1 code)
#define USE_SOLAX_X1   
#endif
in the user_config_override- does not work 

When trying to allocate to a GPIO the SOLAX_RX or TX they are not available ... 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ____ESP32_
- [X] Tasmota binary firmware version number used: _____12.1.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: _ESP Download tool 3.9.2____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The SOLAX_RX and SOLAX_TX values are availabe in the list when configuring the device

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-09-29:
There is template that is required to be filled for good reasons
Please edit your issue and fill up EVERY section as required

Jason2866, 2022-09-30:
Closing since issue template not filled.

closed 2022-09-30 by Jason2866

amuelkx, 2022-09-30:
It is filled - some Fields cannot be filled Wohle data are Not available What is missing Von meinem iPhone gesendet

Jason2866, 2022-09-30:
Just tested does work
[Download.zip](https://github.com/arendst/Tasmota/files/9686239/Download.zip)

barbudor, 2022-09-30:
Most of the fields could be filled including
- Provide the output of command: Backlog Template; Module; GPIO 255:
- Provide the output of this command: Status 0:

Without information we can't help

Jason2866, 2022-09-30:
Probably the changes are copied in the comment section of `user_config_override.h`
The firmware bin i uploaded is compiled via gitpod with the change you wrote.
Solaxx can be selected...