arendst/Tasmota #16701: flashing and configuring Sonoff TH16 using Tasmotizer but getting null values for temp, humidity and dew point

roblatour, 2022-09-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I am flashing and configuring a Sonoff TH16 using the Windows Tasmotizer software but getting null values for temp, humidity and dew point.

It is configured as follows as shown in the screenshots section below.

Prior to flashing the SONOFF temp etc. were reporting fine.

The AMC2301 is firmly plugged into the device.

I see others had a similar problem which was resolved with a code change:
https://github.com/arendst/Tasmota/issues/14203

However, as I am using the Tasmotizer software I can not change the code as described in the link above.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

21:15:01.665 CMD: status 0
21:15:01.673 MQT: stat/tasmota_328ED1/STATUS = {"Status":{"Module":4,"DeviceName":"Tasmota","FriendlyName":["MQTTUser"],"Topic":"tasmota_328ED1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
21:15:01.678 MQT: stat/tasmota_328ED1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T01:22:22","StartupUTC":"2022-09-30T18:52:39","Sleep":50,"CfgHolder":4617,"BootCount":19,"BCResetTime":"2022-09-30T18:21:00","SaveCount":55,"SaveAddress":"FA000"}}
21:15:01.682 MQT: stat/tasmota_328ED1/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:33:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"377/699"}}
21:15:01.686 MQT: stat/tasmota_328ED1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Latour-24G",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:15:01.698 MQT: stat/tasmota_328ED1/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1420C2","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
21:15:01.705 MQT: stat/tasmota_328ED1/STATUS5 = {"StatusNET":{"Hostname":"tasmota-328ED1-3793","IPAddress":"192.168.1.170","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"80:7D:3A:32:8E:D1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
21:15:01.712 MQT: stat/tasmota_328ED1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.173","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_328ED1","MqttUser":"MQTTUser","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:15:01.719 MQT: stat/tasmota_328ED1/STATUS7 = {"StatusTIM":{"UTC":"2022-09-30T20:15:01","Local":"2022-09-30T21:15:01","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:48","Sunset":"18:31"}}
21:15:01.725 MQT: stat/tasmota_328ED1/STATUS10 = {"StatusSNS":{"Time":"2022-09-30T21:15:01","AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}}
21:15:01.730 MQT: stat/tasmota_328ED1/STATUS11 = {"StatusSTS":{"Time":"2022-09-30T21:15:01","Uptime":"0T01:22:22","UptimeSec":4942,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Latour-24G","BSSId":"50:D4:F7:4C:46:FC","Channel":8,"Mode":"11n","RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

21:07:49.425 MQT: tele/tasmota_328ED1/STATE = {"Time":"2022-09-30T21:07:49","Uptime":"0T01:15:10","UptimeSec":4510,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xRedactedx","BSSId":"xRedactedx","Channel":8,"Mode":"11n","RSSI":100,"Signal":-36,"LinkCount":1,"Downtime":"0T00:00:03"}}
21:07:49.433 MQT: tele/tasmota_328ED1/SENSOR = {"Time":"2022-09-30T21:07:49","AM2301":{"Temperature":null,"Humidity":null,"DewPoint":null},"TempUnit":"C"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Plug in the TH16 and problem manifests itself

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Should be reporting temp, humidity, and dew point.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-09-30:
Getting nulls is typical for bad config, bad connections or faulty sensor. Bad config is the rather likely in this case, try configuring the pin as SI7021 instead. If Sonoff characterizes the sensor as "AM2301", it is usually a lie.

roblatour, 2022-09-30:
Amazing, that fixed it!  Thanks for the super quick response.

closed 2022-10-01 by Jason2866