arendst/Tasmota #16707: problem issue on compiling knx tasmota with ITA language

platini76, 2022-10-01:
### PROBLEM DESCRIPTION
HELLO,
I made my personal compile of tasmota with knx and italian language.
there is a bug
when you try to use, save and update something in knx page it exit without save.
with english no issue
thanks for help.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-10-02:
May be you could provide more details?
- your compile details
- log in Weblog 3 when trying to save

platini76, 2022-10-02:
[image]

these are options for compile... in italian...

sorry but I just recompile in english and program it... I cannot weblog

platini76, 2022-10-02:
I am sure that with only knx and italian It goes in this error

sfromis, 2022-10-02:
Better provide the console log (after using the command `Weblog 3` for details), from when trying to save. And details of exactly what you do when saving, like what button with what text on what page. Wondering if it could be a translation issue of some sort.

barbudor, 2022-10-02:
Do you really really need all these options ?

When you say it works in English, do you mean built from TasmoCompiler with exactly the same options ?
Translation itself should break the code (unless a bad translation is pointing you to the wrong button, so compare the button location on both languages)

sfromis, 2022-10-02:
More generally, it saving or not *might* depend on having a sensible configuration to save. Details can matter a lot, instead of telling us nearly nothing by just saying "exit without save".

Jason2866, 2022-10-02:
Just tried with Tasmo Compiler. Building in Italian with KNX is successfull.
Do only include "stuff" you need.

platini76, 2022-10-02:
> Do you really really need all these options ?
>
> When you say it works in English, do you mean built from TasmoCompiler with exactly the same options ? Translation itself should break the code (unless [...]

yes in english with same options works well

> More generally, it saving or not _might_ depend on having a sensible configuration to save. Details can matter a lot, instead of telling us nearly nothing by just saying "exit without save".

when I save knx active page exit and when I reenter inside... it has no active knx function. 
sorry for my english

> Just tried with Tasmo Compiler. Building in Italian with KNX is successfull. Do only include "stuff" you need.

building is successfull but when I try to activate knx it wont' save option.
also with zero "stuff" only knx

barbudor, 2022-10-02:
ok
So I can confirm something is going wrong with your configuration and IT language, while EN works
Language should not impact behavior
This is weird

platini76, 2022-10-02:
> ok
> So I can confirm something is going wrong with your configuration and IT language, while EN works
> Language should not impact behavior
> This is weird

did you try it?

barbudor, 2022-10-02:
yes, I build the EN and the IT versions with your exact sttings and I can see that the ITalian version is behaving weird

platini76, 2022-10-02:
I was sure about this...
who can solve it? transator?

barbudor, 2022-10-02:
Not a translation problem
As I said it is weird
I'm looking at it now

arendst, 2022-10-02:
Italian need more words than non-italian. File size is probably larger

barbudor, 2022-10-02:
@platini76 
Problem is fixed
Please wait for https://github.com/arendst/Tasmota/pull/16711 to be merged, then recompile using the Development branch (not release)

closed 2022-10-02 by ascillato2