arendst/Tasmota #16708: ESP32 Switch discovery bug(MQTT)- Binary sensor

nitinuae, 2022-10-02:
### PROBLEM DESCRIPTION
On ESP32 plateform, discovery protocol of MQTT is wrongly sending the number of switches. Its sending only one switch, though I have 8 (or more) present.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
I am using Astra r4G8, which has four relays and 8 inputs (they are defined correctly, using this template.
{"NAME":"ASTRA R4G8","GPIO":[1,1,1,1,576,1,1,1,480,160,161,162,3232,3200,1,1,0,640,608,1,0,224,225,163,0,0,0,0,227,226,164,165,166,0,0,167],"FLAG":0,"BASE":1}

The discovery MQTT capture is 

{"ip":"192.168.39.225","dn":"Pool","fn":["FilterPump","UnderwaterLights","Spa","MainPool",null,null,null,null],"hn":"tasmota-C541FC-0508","mac":"1097BDC541FC","md":"ASTRA R4G8","ty":0,"if":0,"ofln":"Offline","onln":"Online","state":["OFF","ON","TOGGLE","HOLD"],"sw":"12.1.1","t":"tasmota_C541FC","ft":"%prefix%/%topic%/","tp":["cmnd","stat","tele"],"rl":[1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"swc":[1,0,0,0,0,0,0,0,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1,-1],"swn":["Switch1","Switch2","Switch3","Switch4","Switch5","Switch6","Switch7","Switch8",null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null],"btn":[0,0,0,0,0,0,0,0],"so":{"4":0,"11":0,"13":0,"17":0,"20":0,"30":0,"68":0,"73":0,"82":0,"114":1,"117":0},"lk":0,"lt_st":0,"sho":[0,0,0,0],"sht":[[0,0,0],[0,0,0],[0,0,0],[0,0,0]],"ver":1}

After changing swc to 1,1,1,1,1,1,1,1 my HA instance properly recognizes 8 switches (as binary sensors)
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-10-02:
You haven't configured switchmode for switches 2 to 8

Please refrain opening an issue when obviously you are asking for help. Discussions as there for that.
Issues are for qualified bug reports
Thanks

nitinuae, 2022-10-02:
My bad - I set each and every switch with switch topic and it sent the mqtt discovery. I did try to search and was convinced that it was a bug.

Please document it (we need to set switch topic of each switch to be automtically shown in the HA)

closed 2022-10-02 by nitinuae

barbudor, 2022-10-02:
If you feel something need to be added in the doc, you are welcome to contribute with the pencil button on the top right of the page

Thanks

sfromis, 2022-10-02:
I don't think that you need to configure Switch**Topic**, only Switch**Mode**, as switches are a local-only affair until you configure for them to be published instead of local-only action.