arendst/Tasmota #16717: DNS unable to resolve NTP hostnames

blacknell, 2022-10-04:
### PROBLEM DESCRIPTION
NTP  is not successful on Sonoff DUALR3-v2 tracked down to DNS unable to resolve

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff DUALR3v2
- [x] Tasmota binary firmware version number used: 12.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:08:25.717 CMD: Backlog Template; Module; GPIO 255
00:08:25.765 MQT: stat/tasmota_AA791C/RESULT = {"NAME":"Sonoff Dual R3 v2","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
00:08:26.013 MQT: stat/tasmota_AA791C/RESULT = {"Module":{"0":"Sonoff Dual R3 v2"}}
00:08:26.275 MQT: stat/tasmota_AA791C/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"3200":"Serial Tx"},"GPIO26":{"8128":"BL0939 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
0
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
```
- [ ] Provide the output of this command: `Status 0`:
```lua
00:09:45.990 CMD: status 0
00:09:46.001 MQT: stat/tasmota_AA791C/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota",""],"Topic":"tasmota_AA791C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":4,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
00:09:46.044 MQT: stat/tasmota_AA791C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:08:31","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":5,"BCResetTime":"1970-01-01T00:00:00","SaveCount":19}}
00:09:46.077 MQT: stat/tasmota_AA791C/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:34:17","Core":"2_0_4_1","SDK":"v4.4.3","CpuFrequency":80,"Hardware":"ESP32-D0WD-V3 rev.3","CR":"402/699"}}
00:09:46.097 MQT: stat/tasmota_AA791C/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sandalwood",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
00:09:46.149 MQT: stat/tasmota_AA791C/STATUS4 = {"StatusMEM":{"ProgramSize":1414,"Free":1465,"Heap":143,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","9F9AD7CF","0015A001","B7F7BFCF","05DA9BC0","E0360DC7","400840D2","20200000","D434482D","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,14,16,17,20,21,24,26,27,29,34,35,38,50,52,59,62,63,82,86,87","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,127"}}
00:09:46.216 MQT: stat/tasmota_AA791C/STATUS5 = {"StatusNET":{"Hostname":"tasmota-AA791C-6428","IPAddress":"192.168.100.85","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"8.8.4.4","Mac":"B8:D6:1A:AA:79:1C","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00"},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:09:46.259 MQT: stat/tasmota_AA791C/STATUS6 = {"StatusMQT":{"MqttHost":"hatest.local","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AA791C","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:09:46.291 MQT: stat/tasmota_AA791C/STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:09:46","Local":"1970-01-01T00:09:46","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:09:46.313 MQT: stat/tasmota_AA791C/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
00:09:46.347 MQT: stat/tasmota_AA791C/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:09:46","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"1970-01-01T00:00:00","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":[ 0, 0],"Power":[ 0, 0],"ApparentPower":[ 0, 0],"ReactivePower":[ 0, 0],"Factor":[0.00,0.00],"Voltage": 0,"Current":[0.000,0.000],"BL09XX":{"Temperature":26.3}},"ESP32":{"Temperature":45.0},"TempUnit":"C"}}
00:09:46.394 MQT: stat/tasmota_AA791C/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:09:46","Uptime":"0T00:08:31","UptimeSec":511,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":45,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"sandalwood","BSSId":"BC:A5:11:23:44:D0","Channel":6,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
00:07:38.731 BL9: Rx 55EB020000000000000048393FB50200000000000000000000000000B60100FE030039
00:07:39.542 NTP: Sync time...
00:07:42.612 DNS: Unable to resolve '2.pool.ntp.org'
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flashed a fresh binary on the device and booted into my network. DHCP provides the IP address of DNS servers and NTP settings left untouched.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I expected DNS to resolve NTP so it could set the time

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
- I have other devices on my network which work fine (all ATHOM PG04-UK16A-TAS). However, these never worked with DHCP so I had to manually set IPaddress4/5 (and 1,2,3). NTP then worked immediately.
- On another Sonoff DUALR3v2 which has same problem (I have 3) I manually set the IPaddress4/5 and shortly after DNS resolution started working. I tried this on this device and no luck - I can't figure out the sequencing I must have done on the other device.

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2022-10-04:
Are you sure your device has access to the Internet?

blacknell, 2022-10-04:
> 

Absolutely certain. I was able to upgrade the firmware over the internet and also connect mqtt to test.mosquitto.org (I had to use the IP address because the DNS failed on resolution.

PS. Not sure this latter was a good idea - someone could have put a nasty retained message in there

sfromis, 2022-10-04:
If you "had to use the IP", you have DNS trouble outside of Tasmota, just like with NTP.

The bad idea about connecting to the unsecured public broker test.mosquitto.org is mainly because someone could have a client dynamically reacting to "fresh meat" popping up.

blacknell, 2022-10-04:
I'm not convinced how we can be certain this is outside of Tasmota. Either a setting (relating to DNS) is wrong or something about the way Tasmota has saved the DHCP assigned values isn't quite right. Like I said, several other Tasmotas are working fine which aren't using DHCP - it's something about the way DHCP is interpreted on this device type.

sfromis, 2022-10-04:
If there was a general issue with Tasmota and DNS or DHCP, we should expect a stampede of users having problems. Assuming that you are using standard precompiled binaries, it is just the same firmware as lots of other users are on.

A simple test to check DNS lookup is the Tasmota command `WebQuery http://example.org GET` giving the response `{"WebQuery":"Done"}` vs `{"WebQuery":"Connect failed"}`. The command `IPAddress` shows IP of device, gateway, subnet mask, DNS server and secondary DNS server, typically coming from DHCP.

blacknell, 2022-10-04:
Agree with the principle of you reply. Something's weird as this worked

```
00:02:29.863 CMD: WebQuery http://example.org GET
00:02:29.868 SRC: WebConsole from 192.168.100.13
00:02:29.871 CMD: Grp 0, Cmd 'WEBQUERY', Idx 1, Len 22, Pld -99, Data 'http://example.org GET'
00:02:30.482 MQT: stat/tasmota_AA791C/RESULT = {"WebQuery":"Done"}
```

sfromis, 2022-10-04:
A common way of shooting yourself in the foot is to have some sort of device or settings messing up traffic while performing security theater. Including but not limited to Pi-hole.

blacknell, 2022-10-04:
I have only my router, no other 'clever' stuff. Is there a way to dump out DHCP data as retrieved during start up

barbudor, 2022-10-04:
Your router may offer DNS service
Could you please changed the IP of DnsServer1 to your router and see if it changes the issue?
If yes, it might be a setting in the router letting http and mqtt flowing out but not dns

blacknell, 2022-10-04:
Yes. That worked and DNS is working (and of course NTP). I don't know why this device isn't allowed (by my router) to use 8.8.8.8 when assigned by DHCP so I'll log a ticket with Draytek. Thanks for you help - I'll close this now

closed 2022-10-04 by blacknell