arendst/Tasmota #16720: Incorrect CO2 Sensor Selected

neumann13, 2022-10-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have an air quality sensor that uses a Senseair S8 CO2 sensor. The sensor is detected correctly when the device it initially flashed with the Tasmota-Sensors build. However, after some time, Tasmota changes the device to a Kx0 CO2 sensor. I'm unable to get it to switch back without re-flashing the device entirely

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Lolin D1 Pro, Senseair S8, Plantower PMS5003, SHT30, SGP30
- [ x] Tasmota binary firmware version number used: 12.1.1
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: Tasmotizer
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
RESULT = {"GPIO0":{"1600":"SAir Tx"},"GPIO1":{"0":"None"},"GPIO2":{"1632":"SAir Rx"},"GPIO3":{"0":"None"},"GPIO4":{"640":"I2C SDA"},"GPIO5":{"608":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"1664":"PMS5003 Tx"},"GPIO13":{"0":"None"},"GPIO14":{"1696":"PMS5003 Rx"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:28:56.703 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_61A910","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
14:28:56.708 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Hardware Watchdog","Uptime":"1T05:35:21","StartupUTC":"2022-10-03T07:53:35","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-10-02T16:50:41","SaveCount":19,"SaveAddress":"F9000"}}
14:28:56.712 RSL: STATUS2 = {"StatusFWR":{"Version":"12.1.1(sensors)","BuildDateTime":"2022-10-02T10:47:14","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"329/699"}}
14:28:56.716 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Osaka",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
14:28:56.729 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":684,"Free":316,"Heap":20,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"1840D8","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00184052","20000000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,20,21,24,29,34,62","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,28,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,75"}}
14:28:56.734 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-61A910-2320","IPAddress":"192.168.0.17","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.2","DNSServer2":"0.0.0.0","Mac":"34:94:54:61:A9:10","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:28:56.737 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_61A910","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:28:56.743 RSL: STATUS7 = {"StatusTIM":{"UTC":"2022-10-04T13:28:56","Local":"2022-10-04T14:28:56","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:54","Sunset":"18:23"}}
14:28:56.784 RSL: STATUS10 = {"StatusSNS":{"Time":"2022-10-04T14:28:56","SHT3X":{"Temperature":25.3,"Humidity":40.0,"DewPoint":10.7},"Kx0":{"CarbonDioxide":665,"Temperature":0.0,"Humidity":0.0,"DewPoint":null},"PMS5003":{"CF1":8,"CF2.5":11,"CF10":11,"PM1":8,"PM2.5":11,"PM10":11,"PB0.3":1521,"PB0.5":431,"PB1":76,"PB2.5":3,"PB5":0,"PB10":0},"SGP30":{"eCO2":1211,"TVOC":253,"aHumidity":9.0366},"TempUnit":"C"}}
14:28:56.789 RSL: STATUS11 = {"StatusSTS":{"Time":"2022-10-04T14:28:56","Uptime":"1T05:35:21","UptimeSec":106521,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Osaka","BSSId":"E4:C3:2A:C6:0C:A3","Channel":7,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Compile Tasmota using the Tasmota-Sensors template and the USE_PROMETHEUS option. 
Flash device.
Wait a day or two.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expected it to continue detecting the Senseair S8 and not switch devices to the Kx0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Incorrect Device:
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I compiled this myself using the Tasmota-Sensors as a base and included USE_PROMETHEUS. 

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2022-10-04:
My experience with the S8 is that seeing "Kx0" means trouble communicating with the sensor, circumstances including things like bad connections or not enough power available. Easy way of getting "Kx0" is to disconnect the sensor.....

barbudor, 2022-10-04:
{"GPIO0":{"1600":"SAir Tx"}, "GPIO2":{"1632":"SAir Rx"}
"GPIO12":{"1664":"PMS5003 Tx"},"GPIO14":{"1696":"PMS5003 Rx"}

2 software serial on the same ESP8266 is calling for problems
You'd better move to ESP32(c3) to have at least 2 hardware UART

neumann13, 2022-10-04:
> {"GPIO0":{"1600":"SAir Tx"}, "GPIO2":{"1632":"SAir Rx"} "GPIO12":{"1664":"PMS5003 Tx"},"GPIO14":{"1696":"PMS5003 Rx"}
> 
> 2 software serial on the same ESP8266 is calling for problems You'd better move to ESP32(c3) to have at least 2 hardware UART

I'm really new to this whole ESP thing. What about this configuration do you think is causing this problem? Is the problem something that can alleviated by tweaking some settings? Idk, maybe polling the sensors less frequently?

sfromis, 2022-10-05:
Using software serial means bit banging, which is a high load on the processor to monitor pins to measure timing. That's why I made the suggestion of using hardware serial, probably on the one with most traffic. Hardware serial is when the MCU chip has electronics to do it very efficiently.

Using an ESP32 would mean that you have at least two hardware serial available, meaning no bit banging at all, which is optimal.

I'd suggest first trying if it works out on ESP8266 using hardware serial for one of them. That would be TX on gpio1 and RX on gpio3.

neumann13, 2022-10-05:
Thanks for the assistance. I'll look into the hardware configuration further.

closed 2022-10-05 by neumann13

markruys, 2024-11-15:
The detection of Senseair sensors doesn't handle error situations always correctly. Although a more stable sensor connection is preferable, PR #22487 will won't have the issue that a S8 suddenly is regarded as a Kx0 anymore. Still WIP.