arendst/Tasmota #16722: Nous A1 web energy update reloads from wrong device

joba-1, 2022-10-04:
### PROBLEM DESCRIPTION
A power plug with latest tasmota dev (and older versions) does energy sensor updates on the main web page from a wrong IP.
In my case it partially loads the menu of a shelly 2.5 (which runs original shelly firmware)

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Nous A1
- [x] Tasmota binary firmware version number used: Tasmota 12.1.1.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:13:52.194 CMD: Backlog Template; Module; GPIO 255
15:13:52.242 MQT: stat/nous3/RESULT = {"NAME":"NOUS A1","GPIO":[320,0,576,0,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
15:13:52.454 MQT: stat/nous3/RESULT = {"Module":{"0":"NOUS A1"}}
15:13:52.709 MQT: stat/nous3/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"2656":"HLWBL CF1"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"32":"Button1"},"GPIO14":{"0":"None"},"GPIO15":{"224":"Relay1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
15:14:23.345 CMD: status 0
15:14:23.353 MQT: stat/nous3/STATUS = {"Status":{"Module":0,"DeviceName":"PV Lüfter","FriendlyName":["PV Lüfter"],"Topic":"nous3","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:14:23.385 MQT: stat/nous3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:06:05","StartupUTC":"2022-10-04T14:08:18","Sleep":50,"CfgHolder":4617,"BootCount":23,"BCResetTime":"2021-11-26T01:27:53","SaveCount":809,"SaveAddress":"FA000"}}
15:14:23.421 MQT: stat/nous3/STATUS2 = {"StatusFWR":{"Version":"12.1.1.4(tasmota)","BuildDateTime":"2022-10-04T11:30:50","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"358/699"}}
15:14:23.438 MQT: stat/nous3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"job4","LogPort":514,"SSId":["job.fritz.ssid","job.iot.ssid"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["0000A009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
15:14:23.473 MQT: stat/nous3/STATUS4 = {"StatusMEM":{"ProgramSize":626,"Free":376,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
15:14:23.508 MQT: stat/nous3/STATUS5 = {"StatusNET":{"Hostname":"nous3","IPAddress":"192.168.1.144","Gateway":"192.168.1.221","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.236","DNSServer2":"0.0.0.0","Mac":"C4:4F:33:F7:B3:8F","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:14:23.531 MQT: stat/nous3/STATUS6 = {"StatusMQT":{"MqttHost":"job4","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F7B38F","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:14:23.554 MQT: stat/nous3/STATUS7 = {"StatusTIM":{"UTC":"2022-10-04T14:14:23","Local":"2022-10-04T15:14:23","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:54","Sunset":"18:23"}}
15:14:23.571 MQT: stat/nous3/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:14:23.591 MQT: stat/nous3/STATUS10 = {"StatusSNS":{"Time":"2022-10-04T15:14:23","ENERGY":{"TotalStartTime":"2021-11-26T01:27:53","Total":8.634,"Yesterday":0.007,"Today":0.001,"Power":1,"ApparentPower":8,"ReactivePower":8,"Factor":0.08,"Voltage":237,"Current":0.035}}}
15:14:23.614 MQT: stat/nous3/STATUS11 = {"StatusSTS":{"Time":"2022-10-04T15:14:23","Uptime":"0T00:06:05","UptimeSec":365,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"job.fritz.ssid","BSSId":"28:D1:27:4E:6B:9D","Channel":9,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:15:13.332 CMD: weblog 4
15:15:13.338 MQT: stat/nous3/RESULT = {"WebLog":4}
15:15:13.868 CFG: Saved to flash at F9, Count 810, Bytes 4096
15:15:14.201 HLW: EPC 1, CFlen 1308651 usec
15:15:15.201 HLW: EPC 1, CFlen 1140559 usec
15:15:17.201 HLW: EPC 1, CFlen 1965776 usec
15:15:18.200 HLW: EPC 1, CFlen 830221 usec
15:15:19.200 HLW: EPC 1, CFlen 1528920 usec
15:15:19.438 WIF: Sending Gratuitous ARP
15:15:20.200 HLW: EPC 1, CFlen 1397410 usec
15:15:21.200 HLW: EPC 1, CFlen 972463 usec
15:15:21.395 HTP: Main Menu
15:15:23.200 HLW: EPC 1, CFlen 1180218 usec
15:15:24.014 WIF: Checking connection...
15:15:24.200 HLW: EPC 1, CFlen 1536215 usec
15:15:26.200 HLW: EPC 1, CFlen 1351052 usec
15:15:27.200 HLW: EPC 1, CFlen 1514457 usec
15:15:29.200 HLW: EPC 1, CFlen 1731386 usec
15:15:29.471 MQT: tele/nous3/STATE = {"Time":"2022-10-04T15:15:29","Uptime":"0T00:07:11","UptimeSec":431,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"job.fritz.ssid","BSSId":"28:D1:27:4E:6B:9D","Channel":9,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:05"}}
15:15:29.502 MQT: tele/nous3/SENSOR = {"Time":"2022-10-04T15:15:29","ENERGY":{"TotalStartTime":"2021-11-26T01:27:53","Total":8.634,"Yesterday":0.007,"Today":0.001,"Period":0,"Power":1,"ApparentPower":6,"ReactivePower":6,"Factor":0.09,"Voltage":237,"Current":0.027}}
15:15:30.200 HLW: EPC 1, CFlen 1360795 usec
15:15:32.200 HLW: EPC 1, CFlen 1892209 usec
15:15:33.200 HLW: EPC 1, CFlen 1034310 usec
15:15:34.200 HLW: EPC 1, CFlen 1116747 usec
15:15:36.200 HLW: EPC 1, CFlen 1719576 usec
15:15:37.200 HLW: EPC 1, CFlen 1177258 usec
15:15:38.201 HLW: EPC 1, CFlen 1087178 usec
15:15:39.200 HLW: EPC 1, CFlen 1320052 usec
15:15:41.200 HLW: EPC 1, CFlen 1120310 usec
15:15:42.200 HLW: EPC 1, CFlen 1070686 usec
15:15:43.200 HLW: EPC 1, CFlen 1737428 usec
15:15:44.032 WIF: Checking connection...
15:15:45.200 HLW: EPC 1, CFlen 1654532 usec
15:15:47.200 HLW: EPC 1, CFlen 1482886 usec
15:15:49.200 HLW: EPC 1, CFlen 2110087 usec

```

### TO REPRODUCE
just open main web page and wait for one sensor update

### EXPECTED BEHAVIOUR
Update sensor values

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT
Ip of tasmota device is next to shelly:

```
job4:~ > ping -c3 nous3
PING nous3.job (192.168.1.144) 56(84) bytes of data.
64 bytes from nous3.job (192.168.1.144): icmp_seq=1 ttl=255 time=3.95 ms
64 bytes from nous3.job (192.168.1.144): icmp_seq=2 ttl=255 time=43.3 ms
64 bytes from nous3.job (192.168.1.144): icmp_seq=3 ttl=255 time=3.56 ms

--- nous3.job ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2001ms
rtt min/avg/max/mdev = 3.566/16.952/43.340/18.659 ms

job4:~ > ping -c3 shellyswitch25-0083A7
PING shellyswitch25-0083A7.job (192.168.1.143) 56(84) bytes of data.
64 bytes from shellyswitch25-0083A7.job (192.168.1.143): icmp_seq=1 ttl=128 time=65.8 ms
64 bytes from shellyswitch25-0083A7.job (192.168.1.143): icmp_seq=2 ttl=128 time=88.6 ms
64 bytes from shellyswitch25-0083A7.job (192.168.1.143): icmp_seq=3 ttl=128 time=1.73 ms

--- shellyswitch25-0083A7.job ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2002ms
rtt min/avg/max/mdev = 1.731/52.077/88.699/36.806 ms

```

both devices get IP config from same DHCP server (an openwrt router)

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2022-10-04:
I see no reason to expect a Tasmota issue here.

As can be seen from your screenshot the browser requests ``nous3/?`` and resolves the wrong address. No tasmota involved here. Check your ARP tables.

joba-1, 2022-10-04:
The strange thing for me is that the main page is always displayed correctly -> resolve ok.
Then comes the sensor values update (which is ajax?) and that resolves wrong every time.
If I reload the page a second later it resolves fine. Doesn't sound plausible to me that arp tables switch back and forth all the time. But I will check.

I have other nous with the same version and no problem there. So that indicates a non-tasmota network problem. 

Maybe some combination.