arendst/Tasmota #16723: Serialsend via mqtt skips leading space characters

tjareson, 2022-10-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Serialsend via MQTT skipps leading space characters in a string. 
If the string is "     Test" and I check the TX-pin on the tasmota device with a logic analyzer, I only get "Test". (see screenshot below) The MQTT-message is sent correctly as I can see in a MQTT client subscribed to the same MQTT server the string with leading space characters. Space characters within other characters are not affected.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): sonoff Basic R2
- [x] Tasmota binary firmware version number used: 12.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:24:28.580 CMD: Backlog Template; Module; GPIO 255
15:24:28.614 MQT: stat/test/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
15:24:28.829 MQT: stat/test/RESULT = {"Module":{"1":"Sonoff Basic"}}
15:24:29.083 MQT: stat/test/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
stat/test/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"test","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0}}
15:25:48.088 MQT: stat/test/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:01:40","StartupUTC":"2022-10-04T14:24:08","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2022-01-16T22:27:48","SaveCount":22,"SaveAddress":"F9000"}}
15:25:48.093 MQT: stat/test/STATUS2 = {"StatusFWR":{"Version":"12.1.1(tasmota)","BuildDateTime":"2022-08-25T11:33:55","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285","CR":"366/699"}}
15:25:48.097 MQT: stat/test/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["itacemainfra",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00408009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
15:25:48.108 MQT: stat/test/STATUS4 = {"StatusMEM":{"ProgramSize":624,"Free":376,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":3,"Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62","Sensors":"1,2,3,4,5,6"}}
15:25:48.115 MQT: stat/test/STATUS5 = {"StatusNET":{"Hostname":"test-3719","IPAddress":"xxxxxxxxxxxx","Gateway":"xxxxxxx","Subnetmask":"255.255.255.0","DNSServer1":"xxxxxxxxx","DNSServer2":"xxxxxxxxxx","Mac":"9C:9C:1F:88:0E:87","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:25:48.121 MQT: stat/test/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxxxxxxxx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_880E87","MqttUser":"xxxxxxxxxxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:25:48.128 MQT: stat/test/STATUS7 = {"StatusTIM":{"UTC":"2022-10-04T14:25:48","Local":"2022-10-04T15:25:48","StartDST":"2022-03-27T02:00:00","EndDST":"2022-10-30T03:00:00","Timezone":"+01:00","Sunrise":"06:54","Sunset":"18:23"}}
15:25:48.134 MQT: stat/test/STATUS10 = {"StatusSNS":{"Time":"2022-10-04T15:25:48"}}
15:25:48.139 MQT: stat/test/STATUS11 = {"StatusSTS":{"Time":"2022-10-04T15:25:48","Uptime":"0T00:01:40","UptimeSec":100,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxxxx","BSSId":"22:E8:29:C7:4F:02","Channel":6,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:27:18.177 MQT: stat/test/RESULT = {"WebLog":4}
15:27:27.672 SRC: MQTT
15:27:27.675 CMD: Grp 0, Cmd 'SERIALSEND', Idx 2, Len 4, Pld -99, Data 'Test'
15:27:27.680 MQT: stat/test/RESULT = {"SerialSend":"Done"}
```

### TO REPRODUCE
echo -ne "    Test" | mosquitto_pub -h 192.168.0.12 -t cmnd/test/serialsend2 -s

### EXPECTED BEHAVIOUR
TX-pin should send space characters as well

### SCREENSHOTS
[image]
first terminal is mqtt send, second terminal shows mqtt client subscribed to same mqtt server. logic analyer shows Test only.
### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

tjareson, 2022-10-04:
Oh ok. This was really a headache, as I'm trying to use tasmota to switch on a matrix printer and transfer the printing data to it's serial port via tasmota as well. I've searched almost everywhere until I've noticed the issue can only be with Tasmota somehow.
(bash, mqtt, escape codes of the printer, etc.)
I guess if data gets manipulated while doing something called serialsend it should definitely be mentioned somewhere... 
(plus being configurable) I wonder what is meant by after processing here - writing it to the TX-port?

tjareson, 2022-10-04:
I'm not sure what the exact difference between SendSerial2 (send) and SendSerial4 (send binary) is (as they look the same in the logic analyzer). But I think it would make sense then to have something like a "plain mode", where the output is exactly according to what was sent. 
Like SerialSend7 or so. So it is at the discretion of the sender then if it is with/without CR, in hex or whatever is required.

arendst, 2022-10-04:
Try latest dev release which won't remove leading spaces on any ``SerialSend`` or SSerialSend`` command.

tjareson, 2022-10-04:
Last version worked on the spot for the printer setup here, thanks!
But will it stay implemented that way for the next release then? I'm asking, because that might break backward compability I think.

barbudor, 2022-10-04:
@tjareson I wonder if Tcpbridge would be an interresting alternative
You would use mqtt to turn the printer on but sends the data to the serial through a 'telnet-like' connection, perfectly byte exact

tjareson, 2022-10-04:
Yes, most likely that would work as well. Still I think there should be a way to do a serialsend without data getting corrupted, independent of my indeed a bit special use case.
Possibly mqtt is a bit more robust, in case the wifi connection alone is not completely stable, as MQTT QoS 2 would make sure the data is arriving in one piece just once.

barbudor, 2022-10-04:
Tasmota only support Qos 0. A limitation of the MQTT library used AFAIK.

Alternatively you can use SerialSend5 with payload in hex to guarantee what you send
I wouldn't expect something based on ASCII to respect heading or trailing spaces

tjareson, 2022-10-04:
I'm not even sure what is meant by transfer in hex in this case in detail. (likewise I observe the outcome of ASCII and binary sending to be the same, if I have a look at the serial data on hardware level on the TX pin. Considerung the term binary as such I would expect the outcome to be exactly the same than the input data.)
But if there is a data translation necessary to get the data unchanged accross via tasmota and that would require the same intelligence then on the receiving end to get to the original format back, I would find that technically a difficult position.
Ideal would be just another serialsend variant which is just not touching the data. From user perspective I would also understand, if the binary variant is used for that.
But in the end that's a tech. design decision I guess. No matter what the outcome is, mentioning that blanks are filtered in serialsend would be really important I think. Would have saved me an afternoon. ;-) Not sure if I can do something about it myself?

barbudor, 2022-10-04:
Rolling back on your question about SerialSedn2 vs SerialSend4: there is no difference in the sending but that impact the way the reponse is handled
With SerialSend1/2/3, a delimiter is needed to mark the end of the serial message like \n or ETX
With SerialSend4/5/6, the response is handled without delimiter, as it comes

SerialSend5 use HEX which means that is you want to send `   Hello!` with 3 leading spaces you would have to use `SerialSend5 20202048656C6C6F21` 
Tasmota is then doing the HEX to Binary conversion so nothing needed on the other side
If the other side is sending anything, the SerialReceived message will also be encoded in HEX

Now as you listed bash in your common tools you may have noticed that 
```
echo                    hello
```
just echoes `hello`
If you want to have heading/trailing spaces on a bash command, you generally have to enclose it in quotes such as
```
echo                  "   Hello!"
```
would echo `  Hello!`

So would this be an solution for you ?

It could be either 
1. a change on existing `SerialSend1/2/3/4` : if the 1st character is a `"` then the `"` is ignored and the sent string is everything after the `"` until the last character. If the last character is also a `"` it would also be ignored. If it is not, it would be part of the sent string. However this could be a breaking change for users who are currently expecting `SerialSend "Hello!"` to actually send the `"`. So not really a good idea.
2. such changes conditionned by a setoption (oh, another setoption !)
3. totally "new" `SerialSend` such as `SerialSend11`, `SerialSend21`, `SerialSend31` and `SerialSend41` that would behave the same of there counter part but with the `"` handling

@tjareson @arendst 
What would be your position on that ?
I believe that could be done with very ilttle addition of code

tjareson, 2022-10-04:
@barbudor Not sure I understand you comments on echo. 
As you can see in the screenshot I've used echo -ne "    Test" | ... to pipe it to mqtt. So the approach with having a different behaviour if the string itself start with " or not would require then string constuctions in bash looking like echo "\"   Test\"", which is even more surprising that an escaped string delimiert changes how that data is handled. (The tasmota command itself is the topic in mqtt in that case. So from an mqtt perspective on bash, most likely only the msg part would be in quotes anyway.)

Out of the 3 options you suggest I would still go for option 4: just introduce one more Sendserial variant, for those who want to use it as a bare data transport. For that variant it would be only logical that if I don't want to have leading spaces filtered, I also don't want to have any other changes applied.

barbudor, 2022-10-04:
In you echo command, the `"`are NOT sent
They are just string delimiter that allow the space in front of `Test` to be taken into account in the sending
This is exactly what I'm proposing

In Tasmota `SerialSend2 " Test"` actually sends the `"`

SerialSend1 is echo with `-n`
SerialSend2 is echo with `-n`
SerialSend3 is echo with `-ne`
SerialSend4 is like SerialSend2 but expect no delimiter for hte serial message in response
SerielSend5 as no echo equivalent

And just to clarify : Your option 4 is not possible at all
The spaces between the command and the 1st non space character are filtered before it even reach the SerialSend command code
Just like in bash ani character after the `o` until the 1st non space character is filtered out by bash before reading the echo command

tjareson, 2022-10-04:
Of course they are not send. And if I want to send quotes I have to escape them as I wrote in the second example. 

Option 4 is absolutely possible and exactly what is currently implemented by arendst in the development version. Just for all Serialsend variants. It is running here on a sonoff device and I'm printing successfully with it with leading spaces.

What you probably don't realize is that the mqtt message is split into two parts for the mqtt client I'm using. First one is the topic, second one is the data. I don't send the complete command in one string, instead of that I pipe the string to the mosquitto_pub client which has the topic already as parameter:
So the command 
`echo -ne "    Test" | mosquitto_pub -h xxx.yyy.zzz.xxx -t cmnd/test/serialsend2 -s`
is exactly sending the spaces first followed by Test on the TX-pin. Option 4 here we go.

tjareson, 2022-10-04:
Maybe I should add: if I talk about option 4, I mean a fourth option compared to the three you had suggested, not SerialSend4.

barbudor, 2022-10-04:
I didn't saw that Theo already made that
If you are both happy with that...
I just find weird to impact the command processor specifically for SerialSend instead of working on a proper syntax

And by the way I perfectly knows how MQTT is working

tjareson, 2022-10-04:
I mean, of course I understand your point. If you see it from a parsing perspective - if there is nothing catering for string delimiters, hence quotes would be send by SerialSend along with the rest of the data. Which, I can only guess, is maybe the trade-off with limited resources as it probably makes parsing tokens etc easier. 
On the other hand I surprisingly do not see bigger impacts yet:
setting for instance other alphanumeric values, such as switchtopic, leading blanks are still eliminated:
[image]

But then again sendserial works "space sensitive":
[image]
leading now to:
[image]

I'm not sure if [d6b045e](https://github.com/arendst/Tasmota/commit/d6b045e3d423a3182e8d1b43937fb43251ff2fee) applies  only to serialsend or to all kind of commands. At least with the switchtopic example it doesn't seem to have any other side effect.

barbudor, 2022-10-04:
Pretty clear : applies only to SerialSend
[image]

tjareson, 2022-10-04:
An ok, so then no surprise that the rest is still working like before. I didn't really look at the change. 
What hinders then to just make this case one level more specific for a new SerialSend7 for instance?
So the "other" SerialSends are kept the way they were for backward compability.

arendst, 2022-10-05:
There are six different SerialSend implemntations. I see at least three which can benefit from the "non-leading space removal" option. This would mean three additional commands. One command is removing the leading spaces anyway (``SendSerial5``).

For now I settle with the current solution. Most users do not add leading spaces anyway and if they do they just need to do the pre-processing themselves.

Jason2866, 2022-10-06:
Closing since the original wanted feature is implemented.

closed 2022-10-06 by Jason2866

avp-avp, 2022-12-04:
This logic broke web interface (S)SerialSend command
Now this command send additional space to serial port

Jason2866, 2022-12-04:
So the space was there before and removed in older version, now it is not removed anymore.

avp-avp, 2022-12-04:
Now you cannot send any string to serial port without space in begin of message!

arendst, 2022-12-04:
I'll revert the change and add new commands (S)SerialSend11 to 16 performing like (S)SerialSend1 to 6 but keeping the leading space(s).

Hold on.

arendst, 2022-12-04:
@avp-avp try lates dev release. Pre-compiled available for download in an hour from http://ota.tasmota.com/tasmota and/or http://ota.tasmota.com/tasmota32

avp-avp, 2022-12-05:
@arendst Thanks!

But may be it can be a general problem because payload from mqtt a different from other sources that parse '<command> <parameter>' like rules, web console etc